evcc-io/evcc #6684: Goodwe ET - Connection Refused

Sandmanyz, 2023-03-07:
Beim Hinzufügen meines Wechselrichters Goddwe ET erhalte ich die Meldung Connection Refused.
Port: 507
Modbus ID: 247

Aktuell habe ich den Wechselrichter über WLAN und einem ioBroker Adapter jedoch lokal eingebunden. Der ioBroker-Adapter nutzt meines Wissens den Port 8899 (UDP). Dazu ist hier aus etwas zu finden....https://community.openems.io/t/kommunikation-mit-goodwe-et-im-lokalen-netz/541/3

Kann das Protokoll (TCP) die Ursache sein, dass es bei mir nicht funktioniert? Kann die Verbindung über UDP implementiert werden?

andig, 2023-03-08:
Das bräuchte https://github.com/grid-x/modbus/issues/50. Aktuell mangels Support leider wontfix.

closed 2023-03-08 by andig

premultiply, 2023-03-08:
Ansonsten bitte einfach die offiziellen Schnittstellen verwenden. Dafür ist das Template gemacht und sollte auch funktionieren.

premultiply, 2023-03-08:
Möglicherweise.
Ich kenne das Gerät ja nicht persönlich.

Wenn es per RS485 kommuniziert brauchst du halt für zwei Euro ein RS485-USB-Interface oder eine vergleichbare Lösung.

maatinh, 2023-03-08:
Hatte das gleiche Problem. Hab anfangs ein custom-meter verwendet um evcc mit den korrekten Daten zu versorgen (bei mir hat evcc die Daten bei fhem abgeholt; ins fhem kamen sie über die goodwe-python lib über udp).
Inzwischen habe ich den lan/wlan Adapter von goodwe. Der macht modbus über lan und tcp.

andig, 2023-03-09:
Weiß jemand, ob sich UDP per Portweiterleitung in beiden Richtungen sprechen ließe? Falls ja gerne Portweiterleitung an info@evcc.io, da müsste ich mal experimentieren wie schwierig es wäre, Modbus UDP umzusetzen.

premultiply, 2023-03-09:
Ja, geht genauso wie bei TCP.

maatinh, 2023-03-09:
Ich habe inzwischen den anderen GoodWe-Adapter. Soweit ich bis jetzt gesehen habe, macht der über LAN ausschließlich Modbus/TCP. Und der kann leider nicht WLAN/LAN gleichzeitig. Das heißt, ich kann aktuell keine UDP-Portweiterleitung anbieten ohne dabei auch mein ganzes Monitoring und EVCC-Setup zu deaktivieren.
   
  
   Ich denke aber darüber nach, wie wir das hinkriegen können. Da ich aber aktuell beruflich unterwegs bin, wird das noch einige Tage dauern.
   
  
    
   
  
    
   
  
   Gruß,
   
  
   Martin

maatinh, 2023-03-21:
Hallo,
   
  
    
   
  
   sieht so aus, als hätte ich mich geirrt. ModbusTCP und UDP funktionieren gleichzeitig. Ich könnte also beide Ports öffnen.
   
  
    
   
  
    
   
  
   Gruß,
   
  
   Martin

andig, 2024-02-18:
@Sandmanyz @maatinh unsere Modbus Library bietet jetzt auch Modbus UDP an. Gibts den Bedarf noch? Handelt es sich um Modbus TCP oder RTU? Könntet ihr info@evcc.io Zugang gegen?

maatinh, 2024-02-19:
@andig Grundsätzlich bin ich der Meinung, dass der Bedarf schon besteht. Neuere Goodwe WRs werden ab Werk mit Wifi ausgeliefert, so dass ich annehmen würde, dass Modbus-UDP der Standard für Goodwe WRs wäre. Modbus-TCP benötigt ein separat zu erwerbendes USB-Modul.
Da ich inzwischen auf das externe Modul gewechelt habe, kann ich leider keinen Zugriff mehr geben.