evcc-io/evcc #6701: Virtueller Zählpunkt bei Kaskadenschaltung (z.B. Differenz zwischen zwei Discovergy-Zählern)

maf-soft, 2023-03-08:
I'm always frustrated when [...], äh, ich auf die hübsche Stromfluss-Visualisierung gucke und sie meine PV-Produktion nicht kennt, weil ich keinen Produktionszähler habe, sondern die sich bei mir aus der Differenz zwischen 2 Discovergy-Zählern in Kaskadenschaltung ergibt.

Es braucht also die Möglichkeit, in der Konfiguration einen virtuellen Zähler anlegen zu können, der seine Werte über eine benutzerdefinierte Formel mit Variablen von anderen Zählern erhält.

Für meinen Zweck würde es natürlich reichen, wenn genau 2 Discovergy-Zähler angegeben werden könnten, zwischen denen dann die Differenz gebildet wird. Aber eine universellere Lösung wäre schöner.

Zusatzgedanken zu Live-Daten: bestimmt wird es auch noch nötig, optional eine Glättung dazwischen zu schalten und/oder den Timestamp zu berücksichtigen, weil die beiden Zähler womöglich nicht genau in der gleichen Sekunde senden, und es dann in der Differenz Ausschläge gibt, obwohl sich die Produktion gar nicht geändert hat, sondern nur der Verbrauch.

andig, 2023-03-08:
Technischer Lösungsansatz: Meßwerte aus per Namen übergebenen Zählern extrahieren.

maf-soft, 2023-03-09:
Ich denke noch über einen anderen Ansatz nach, siehe auch [Slack-Thread](https://evccgroup.slack.com/archives/C01321PUJAD/p1678372765665509?thread_ts=1678112986.974499&cid=C01321PUJAD).

Im Grunde ist es ja schon so, dass evcc alle bekannten Verbraucher und Erzeuger mit dem Hauptzähler verrechnet, und wenn da was übrig bleibt, sich den entsprechenden Verbrauch oder Erzeugung dazudenkt und auch korrekt anzeigt, ohne dass der selbst irgendwo gezählt wird.

Also ließe sich mein Problem auch lösen, und vielleicht sogar eleganter, wenn man unter site/meters einfach weitere Zähler angeben könnte, die in die Gesamtbilanz einfließen, vermutlich (?) genau wie "pv" es tut, nur mit umgekehrtem Vorzeichen. Vielleicht möchte jemand z.B. seine Wärmepumpe als Verbraucher visualisiert haben und ich eben meinen Hausverbrauch und bei mir kann das "Haus" eben nicht nur verbrauchen, sondern auch einspeisen - nebenbei bemerkt werden Wallboxen in Zukunft auch einspeisen können. Negative Werte sind also nichts ungewöhnliches. Also eigentlich ist mein Hauszähler mathematisch wie pv, charger/loadpoint und auch battery zu behandeln, eben nur ein leicht anderer Typ, vielleicht ist das gar keine große Erweiterung?

Ich finde das klingt für das evcc-Produkt allgemein wie eine sehr flexible und transparente Möglichkeit.

Lediglich in meinem speziellen Fall käme das am Ende aufs gleiche Problem hinaus, denn ich müsste dann die Differenz zwischen Zähler 2 und dem Wallbox-Verbrauch bilden, weil die Wallbox dummerweise dahinter sitzt. Ich frage mich dann, ob man bei dem Loadpoint einfach ein Flag einbauen könnte, das diesen Verbrauchswert für die Gesamtbilanz ignoriert, aber ansonsten normal für die Steuerung dieses Loadpoints nutzen kann.

andig, 2023-03-09:
Keine Ahnung ob das funktioniert, aber schau Dir gerne mal die `aux` meters an.

maf-soft, 2023-03-09:
Habe ich zufällig auch gerade so laufen, aber jetzt im Dunkeln zeigt es in der GUI nichts anderes an als ohne `aux`. Lediglich im -l debug sieht man was davon:

```
charge power: 0W
grid power: 459W
pv power: 0W
aux power 0: 450W
site power: 9W
charge currents: [0.01 0.01 0.009]A
charger status: B
pv charge current: 0A = 0A + -0.04A (9W @ 1p)
```
Die 9W sind vermutlich der Eigenverbrauch der WR zwischen den beiden Zählern.

Die Beschreibung und der Sinn von `aux` passen allerdings nicht ganz, theoretisch könnte der Hybrid-WR zwar fast genau das aus dem Beispiel mit dem Heizstab machen, also mit der Batterie seine Produktion so ausregeln, dass der Zähler steht (wie wenn der Heizstab sich mangels Überschuss abschaltet). Aber das wäre nicht erwünscht. Es bräuchte also zumindest noch Konfigurationsmöglichkeiten für das Verhalten von `aux`.

andig, 2023-03-09:
...dann ist aux wohl nicht die Lösung.

mikeungers, 2023-03-13:
Auch noch ein nice to have wäre die Berücksichtigung der unterschiedlichen Einspeisevergütungen bei der Statistik.
Sprich man müsste die Einspeisevergütung pro Zähler konfigurieren können.

andig, 2023-03-19:
Aktuell kein Fokus das umzusetzen. Der Workaround wäre, die richtigen Werte extern zu ermitteln und evcc per Plugin zur Verfügung zu stellen. Tut mir leid...

> Auch noch ein nice to have wäre die Berücksichtigung der unterschiedlichen Einspeisevergütungen bei der Statistik.
Sprich man müsste die Einspeisevergütung pro Zähler konfigurieren können.

Hat hiermit nichts zu tun.

closed 2023-03-19 by andig