evcc-io/evcc #6720: show contactor delay timer in UI

VolkerK62, 2023-03-09:
Es verwirrt mich immer, wenn die Ladung nicht startet, wenn ich z.b. `minpv` aktiviere. Es wäre schön, wenn der contactor delay timer ebenfalls im UI angezeigt würde. So wie bei enable/disable,

andig, 2023-03-09:
Warum passiert das bei Dir? Schaltest Du soviel hin und her oder hast Du den so niedrig gesetzt? Mir ist das noch nie aufgefallen.

VolkerK62, 2023-03-09:
Das ist passiert, weil meine Wallbox nicht so richtig das gemacht hat, was ich erwartet habe. Hängt evtl mit #6728 zusammen

VolkerK62, 2023-03-09:
Wieso? Bei wechselnden Bedingungen. Disable ist abgelaufen, Ladung wird beendet. Dann kommt wieder Sonne, man schaut aufs UI und fragt sich, warum wird jetzt nicht eingeschaltet?

PR #6743 (2023-03-19): Better vehicle status messages (delays, timer)
Expose guard/contactor timer in UI to explain why charging has not started yet. This can occure if the pv enable timer is configured smaller than the contactor delay. This behavior can also be observed in the demo.yaml.

I decided to present this status in a neutral way ("Starting in 1:20m...") to avoid explaining the concept of fast switching protection to the UI. 

[image]

**Possible next step:** The current situation is not perfect from a users perspective because the user might first see the pv enable timer (e.g. 2m). When it's done the rest  [... cut]
closed 2023-03-19 by andig