evcc-io/evcc #6725: Evcc Screen on iPhone

Rudi3222, 2023-03-09:
### Describe the bug

Wenn ich http://localhost:7070 bei safari beim iPhone eingebe, erscheint eine Fehlermeldung, dass Safari keine Verbindung zum Server aufbauen konnte...
Über den Mac komme ich problemlos auf die Seite...
Was mache ich falsch?
Danke

### Steps to reproduce

1. siehe oben
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: 0d0a335ab98e22a2afe7d01bfe77b68c483321b63ce95eb3657835473e26b8cd

interval: 10s # control cycle interval

meters:
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.177  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.177  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.177  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery3

chargers:
- type: template
  template: fronius-wattpilot 
  host: 192.168.178.111  
  password: Katrin2505  
  name: wallbox5

vehicles:
- type: template
  template: id 
  title: ID3  
  user: ralfblohmann@gmx.de  
  password: doqwy4-xecwec-jornUs  
  vin: WVWZZZE1ZMP084202  
  capacity: 58  
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox5
  vehicle: ev4
  mode: now
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
[main  ] INFO 2023/03/09 15:44:28 evcc 0.114.1
[main  ] INFO 2023/03/09 15:44:28 using config file: /Users/ralf/evcc.yaml
[main  ] INFO 2023/03/09 15:44:28 starting ui and api at :7070
[db    ] INFO 2023/03/09 15:44:28 using sqlite database: /Users/ralf/.evcc/evcc.db
[lp-1  ] WARN 2023/03/09 15:44:33 locking phase config to 3p for switchable charger
[site  ] INFO 2023/03/09 15:44:33 site config:
[site  ] INFO 2023/03/09 15:44:33   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/03/09 15:44:33     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/03/09 15:44:33     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/03/09 15:44:33     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/03/09 15:44:33   vehicles:
[site  ] INFO 2023/03/09 15:44:33     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[lp-1  ] INFO 2023/03/09 15:44:33 loadpoint 1:
[lp-1  ] INFO 2023/03/09 15:44:33   mode:        now
[lp-1  ] INFO 2023/03/09 15:44:33   charger:     power ✓ energy ✗ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/03/09 15:44:33   meters:      charge ✓
[lp-1  ] INFO 2023/03/09 15:44:33     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2023/03/09 15:44:33 phase timer inactive
[lp-1  ] DEBUG 2023/03/09 15:44:33 pv timer inactive
[lp-1  ] INFO 2023/03/09 15:44:33 vehicle updated: unknown -> ID3
[site  ] DEBUG 2023/03/09 15:44:33 ----
[lp-1  ] DEBUG 2023/03/09 15:44:33 charge power: 0W
[main  ] FATAL 2023/03/09 15:44:33 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
```

### What type of operating system are you running?

Linux

maf-soft, 2023-03-09:
Schnell das Passwort rauslöschen!

goebelmeier, 2023-03-09:
> Schnell das Passwort rauslöschen!

Rauslöschen hilft auf Github nichts, die sind jetzt verbrannt und müssen geändert werden. Ich bekomme z.B. eine Mail über jeden Kommentar und da stehen die Passworte drin.

goebelmeier, 2023-03-09:
> ### Describe the bug
> Wenn ich http://localhost:7070 bei safari beim iPhone eingebe, erscheint eine Fehlermeldung, dass Safari keine Verbindung zum Server aufbauen konnte... Über den Mac komme ich problemlos auf die Seite... Was mache ich falsch? 

Du musst nicht `http://localhost:7070` eingeben sondern `http://IP-ADRESSE-DEINES-MACS:7070`

maf-soft, 2023-03-09:
Ich habe auch ganz viele Probleme (aber unter Windows), dass der Browser sagt, es könne keine Verbindung hergestellt werden. Im Nächsten Browser oder auf dem nächsten PC im gleichen Netzwerk geht es dann aber und dann plötzlich auch wieder auf dem ersten. Wollte das noch eine Weile beobachten, bevor ich nach Hilfe suche. Ich habe erstmal in der config das "evcc.local" durch den echten Hostname ersetzt, seitdem kommt es mir besser vor. Aber verstehen tue ich es nicht. Selbst mit "localhost:7070" auf dem PC auf dem der Server läuft, funktioniert es manchmal nicht.

maf-soft, 2023-03-09:
Ach jetzt kapiere ich erst, dass meine Probleme keine Ähnlichkeit mit dem Problem hier haben und ich hier falsch bin :)

Ja klar, localhost bezieht sich auf den Rechner auf dem man gerade ist, und da evcc nicht auf dem iPhone läuft, kann das so nicht gehen.

maf-soft, 2023-03-09:
Mit dem Passwort könnte man womöglich sogar das Auto aufschließen? @Rudi3222 sollte handeln und wenn der nicht reagiert, vielleicht ein Admin hier?