evcc-io/evcc #6728: Abfrageintervall abhängig von Anzahl der Loadpoints?

VolkerK62, 2023-03-09:
### Describe the bug

Bei der Suche nach der Ursache für ein kleines Problem mit meinem DaheimLader Modbus ist mir etwas aufgefallen.
Ich habe drei Loadpoints. In jedem Intervall wird nur immer ein Loadpoint detailliert abgefragt.
ChargePower in jedem Intervall, aber weitere Details (Status, socPoll, chargeCurrents, usw) nur in jedem dritten Intervall.

Siehe Log (im Ruhezustand, keine Ladung):
17:22:14 = LP-1
17:22:29 = LP-2
17:22:44 = LP-3
usw. 

### Steps to reproduce

.

### Configuration details

```yaml
Konfiguration (/etc/evcc.yaml):

network:
  schema: http
  host: evcc.local
  port: 7070

interval: 15s  

sponsortoken: *****

telemetry: true
  
log: debug
levels:
  #site: trace
  #lp-1: info
  #lp-2: info
  #lp-3: info
  #daheimladen-mb: trace
  #hyundai: fatal #trace
  #main: trace
  #cache: trace
  #db: trace
  #influx: trace
  #mqtt: trace
  proxy-5200: trace
  
########################################################################
meters:
- name: grid
  type: custom
  power:
    source: http
    uri: http://192.168.178.28/status
    jq: .emeters | map(.power) | add
  energy:
    source: http
    uri: http://192.168.178.28/status
    jq: .emeters | map(.total) | add
    scale: 0.001
  currents:
  - source: http
    uri: http://192.168.178.28/emeter/0
    jq: if .power < 0 then - .current else .current end
  - source: http
    uri: http://192.168.178.28/emeter/1
    jq: if .power < 0 then - .current else .current end
  - source: http
    uri: http://192.168.178.28/emeter/2
    jq: if .power < 0 then - .current else .current end
[... 337 lines cut ...]
Mär 09 17:24:00 raspberrypi evcc[20821]: [site  ] DEBUG 2023/03/09 17:24:00 grid currents: [-1.33 1.18 1.12]A
Mär 09 17:24:00 raspberrypi evcc[20821]: [site  ] DEBUG 2023/03/09 17:24:00 site power: 439W
Mär 09 17:24:00 raspberrypi evcc[20821]: [lp-2  ] DEBUG 2023/03/09 17:24:00 charger status: B
Mär 09 17:24:14 raspberrypi evcc[20821]: [site  ] DEBUG 2023/03/09 17:24:14 ----
Mär 09 17:24:14 raspberrypi evcc[20821]: [lp-1  ] DEBUG 2023/03/09 17:24:14 charge power: 0W
Mär 09 17:24:14 raspberrypi evcc[20821]: [lp-2  ] DEBUG 2023/03/09 17:24:14 charge power: 0W
Mär 09 17:24:14 raspberrypi evcc[20821]: [lp-3  ] DEBUG 2023/03/09 17:24:14 charge power: 0W
Mär 09 17:24:14 raspberrypi evcc[20821]: [site  ] DEBUG 2023/03/09 17:24:14 pv power: 200W
Mär 09 17:24:14 raspberrypi evcc[20821]: [site  ] DEBUG 2023/03/09 17:24:14 battery soc: 46%
Mär 09 17:24:14 raspberrypi evcc[20821]: [site  ] DEBUG 2023/03/09 17:24:14 battery power: 128W
Mär 09 17:24:14 raspberrypi evcc[20821]: [site  ] DEBUG 2023/03/09 17:24:14 grid power: 61W
Mär 09 17:24:14 raspberrypi evcc[20821]: [site  ] DEBUG 2023/03/09 17:24:14 grid powers: [-182 75 167]W
Mär 09 17:24:15 raspberrypi evcc[20821]: [site  ] DEBUG 2023/03/09 17:24:15 grid currents: [-1.17 1.19 1.18]A
Mär 09 17:24:15 raspberrypi evcc[20821]: [site  ] DEBUG 2023/03/09 17:24:15 site power: 439W
Mär 09 17:24:15 raspberrypi evcc[20821]: [lp-3  ] DEBUG 2023/03/09 17:24:15 charger status: B
Mär 09 17:24:15 raspberrypi evcc[20821]: [lp-3  ] DEBUG 2023/03/09 17:24:15 next soc poll remaining time: 2m15s
```

### What type of operating system are you running?

Linux

### Version

0.114.1 (1f4b7ab4)

closed 2023-03-09 by andig

VolkerK62, 2023-03-09:
Ok. Dann werde ich wohl 2 Loadpoints löschen müssen und schauen, ob sich das Problem damit erledigt hat.
Meine Wallbox hat unerwartet reagiert. Im Log ist mir dann aufgefallen, dass `charger enable` kam und erst 3 Intervalle später `max charge Current` gemeldet wurde. Zwischendurch kam dann `charger out of sync: expected enabled, got disabled`.

andig, 2023-03-09:
Ich verstehe nicht welches "Problem"? Die Frage des Issues ist mit "ja" beantwortet ;)