evcc-io/evcc #6730: anonymisieren des topic von ntfy

VolkerK62, 2023-03-09:
Wenn man mit `evcc dump --cfg` die Config ausliest, wird das topic der ntfy-Adresse beim Messaging nicht anonymisiert.
Das wäre mMn aber sinnvoll.

```
messaging:
  services:
  - type: ntfy
    uri: https://ntfy.sh/**hier steht das Topic in Klartext**
```

andig, 2023-03-09:
Closed in https://github.com/evcc-io/evcc/commit/1ae60cd823375db102da0712e8010c8fcc1aa789

andig, 2023-03-25:
Daran sollte sich auch nichts geändert haben- zumindest war das nicht die Absicht. Im Logfile müsste es aber weg sein.

VolkerK62, 2023-03-25:
Meine Wunsch wäre, dass das genauso anonymisiert wird wie Passworte und token

andig, 2023-03-25:
Nochmal: im Logfile sollte es anonymisiert sein, genau darüber hatten wir gesprochen. Passt das? Von evcc dump war bisher nicht die Rede, oder bin ich da falsch unterwegs?

VolkerK62, 2023-03-25:
Ich würde sagen, da bist du falsch unterwegs. ;-)
Mein Eingangspost bezog sich auf "dump". Da ist es mir aufgefallen.

andig, 2023-03-25:
Ausch. Leider geht das nicht. Alternative wäre, das topic als eigenen Wert zu konfigurieren, dann kann aber niemand mehr eine komplette uri angeben, die wär immer auf den Dienst fix.

VolkerK62, 2023-03-25:
Ok. Dann weiß ich Bescheid. Ich schau mal, dass ich dazu einen Hinweis in die Doku aufnehme.