evcc-io/evcc #6731: ELinta and Go-e wallboxes can't connect to EVCC via OCPP

thomas-hutterer-tik, 2023-03-09:
### Describe the bug

I have configured EVCC to host an OCPP server with the OCPP template and have configured a ELinta and a Go-e Charger to connect to the EVCC OCPP server on port 8887

### Steps to reproduce

1. configured EVCC to host an OCPP server with the OCPP template 
2. configure an ELinta and a Go-e Charger to connect to the EVCC OCPP server on port 8887
4. restart each wallbox
5. observer the errors messages on the wall boxes
...

### Configuration details

```yaml
---
network:
  schema: "http"
  host: "0.0.0.0"
  port: 7070
interval: "2s"
sponsortoken: "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3MDQwMzQwMDIsImlhdCI6MTY3MTIwMjAwMiwiaXNzIjoiZXZjYy5pbyIsInN1YiI6IldhdHQgQW5hbHl0aWNzIiwic3BlIjp0cnVlfQ.1BPAwegvKSsGWvvyNFFVvvLL3q_Fdi7Le5DRJ6_zkcQ"
log: "trace"
meters:
- name: "eMallorca Lab"
  type: "custom"
  power:
    source: "mqtt"
    topic: "wa-pi/grid"
  currents:
  - source: "mqtt"
    topic: "wa-pi/grid/p1"
  - source: "mqtt"
    topic: "wa-pi/grid/p2"
  - source: "mqtt"
    topic: "wa-pi/grid/p3"
chargers:
- name: "OCPP-1"
  type: "template"
  template: "ocpp"
circuits:
- name: "eMallorca"
  maxCurrent: 8
  meter: "eMallorca Lab"
site:
  title: "eMallorca"
  meters:
    grid: "eMallorca Lab"
    pvs: []
    batteries: []
loadpoints:
- title: "OCPP-1"
  charger: "OCPP-1"
  phases: 3
  mode: "now"
  circuit: "eMallorca"
mqtt:
  broker: "192.168.1.127"
  topic: "evcc"
  user: "wm-1aq0cpgFIoJZ"
  password: "_00mDC.wYJoH"
```

### Log details

```text
EVCC Log ====================================
Mar 09 17:50:22 iwattcontroller systemd[1]: Started evcc.
Mar 09 17:50:23 iwattcontroller evcc[8381]: [main  ] INFO 2023/03/09 17:50:23 evcc 0.109.2 (3ec2d1b70)
Mar 09 17:50:23 iwattcontroller evcc[8381]: [main  ] INFO 2023/03/09 17:50:23 using config file: /etc/evcc.yaml
Mar 09 17:50:23 iwattcontroller evcc[8381]: [main  ] INFO 2023/03/09 17:50:23 starting ui and api at :7070
Mar 09 17:50:24 iwattcontroller evcc[8381]: [db    ] TRACE 2023/03/09 17:50:24 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
Mar 09 17:50:24 iwattcontroller evcc[8381]: [db    ] TRACE 2023/03/09 17:50:24 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
Mar 09 17:50:24 iwattcontroller evcc[8381]: [db    ] TRACE 2023/03/09 17:50:24 SELECT * FROM `settings` LIMIT 1 -1 <nil>
Mar 09 17:50:24 iwattcontroller evcc[8381]: [db    ] TRACE 2023/03/09 17:50:24 SELECT * FROM `settings` 6 <nil>
Mar 09 17:50:24 iwattcontroller evcc[8381]: [mqtt  ] INFO 2023/03/09 17:50:24 connecting evcc-57260903 at tcp://192.168.1.127:1883
Mar 09 17:50:24 iwattcontroller evcc[8381]: [mqtt  ] DEBUG 2023/03/09 17:50:24 tcp://192.168.1.127:1883 connected
Mar 09 17:50:25 iwattcontroller evcc[8381]: [ocpp  ] DEBUG 2023/03/09 17:50:25 waiting for chargepoint: 10m0s
Mar 09 17:51:16 iwattcontroller evcc[8381]: [mqtt  ] TRACE 2023/03/09 17:51:16 recv wa-pi/grid: '461.81'
Mar 09 17:51:16 iwattcontroller evcc[8381]: [mqtt  ] TRACE 2023/03/09 17:51:16 recv wa-pi/grid/p1: '0.8553214'
Mar 09 17:51:16 iwattcontroller evcc[8381]: [mqtt  ] TRACE 2023/03/09 17:51:16 recv wa-pi/grid/p2: '0.5554507'
Mar 09 17:51:16 iwattcontroller evcc[8381]: [mqtt  ] TRACE 2023/03/09 17:51:16 recv wa-pi/grid/p3: '0.5037828'

ELinta logs from from elios.cloud: =========================
[03-09 18:42:54]: Settings have been changed (edit)! (by "wattanalytics")
[03-09 18:43:02]: Switched to Elios control mode
[03-09 18:43:02]: OCPP status changed from AUTHORIZED to UNAVAILABLE
[03-09 18:43:33]: Switched to OCPP control mode(OCPPJ1_6_CCM2, "ws://192.168.1.127:8887/")
[03-09 18:43:33]: OCPP status changed from UNAVAILABLE to DISCONNECTED
[03-09 18:43:33]: OCPP status changed from DISCONNECTED to CONNECTING
[03-09 18:43:37]: OCPP status changed from CONNECTING to DISCONNECTED
[03-09 18:44:33]: OCPP status changed from DISCONNECTED to CONNECTING
[03-09 18:44:36]: OCPP status changed from CONNECTING to DISCONNECTED
[03-09 18:45:33]: OCPP status changed from DISCONNECTED to CONNECTING
[03-09 18:45:36]: OCPP status changed from CONNECTING to DISCONNECTED
[03-09 18:46:33]: OCPP status changed from DISCONNECTED to CONNECTING
[03-09 18:46:37]: OCPP status changed from CONNECTING to DISCONNECTED
[03-09 18:47:33]: OCPP status changed from DISCONNECTED to CONNECTING
[03-09 18:47:36]: OCPP status changed from CONNECTING to DISCONNECTED
[03-09 18:48:33]: OCPP status changed from DISCONNECTED to CONNECTING
[03-09 18:48:37]: OCPP status changed from CONNECTING to DISCONNECTED
[03-09 18:49:33]: OCPP status changed from DISCONNECTED to CONNECTING
[03-09 18:49:37]: OCPP status changed from CONNECTING to DISCONNECTED
[03-09 18:50:33]: OCPP status changed from DISCONNECTED to CONNECTING
[03-09 18:50:37]: OCPP status changed from CONNECTING to DISCONNECTED
[03-09 18:51:33]: OCPP status changed from DISCONNECTED to CONNECTING
[03-09 18:51:37]: OCPP status changed from CONNECTING to DISCONNECTED
[03-09 18:52:33]: OCPP status changed from DISCONNECTED to CONNECTING
[03-09 18:52:37]: OCPP status changed from CONNECTING to DISCONNECTED
[03-09 18:53:31]: Connector 1 status changed from READY to CABLE_CONNECTED
[03-09 18:53:35]: Connector 1 status changed from CABLE_CONNECTED to CAR_WAITING
[03-09 18:54:33]: OCPP status changed from DISCONNECTED to CONNECTING
[03-09 18:54:37]: OCPP status changed from CONNECTING to DISCONNECTED
[03-09 18:56:34]: OCPP status changed from DISCONNECTED to CONNECTING
[03-09 18:56:37]: OCPP status changed from CONNECTING to DISCONNECTED
[03-09 18:56:57]: Connector 1 status changed from CAR_WAITING to CAR_SWITCHING
[03-09 18:56:59]: Connector 1 status changed from CAR_SWITCHING to CHARGING
[03-09 18:58:34]: OCPP status changed from DISCONNECTED to CONNECTING
[03-09 18:58:37]: OCPP status changed from CONNECTING to DISCONNECTED
[03-09 19:00:34]: OCPP status changed from DISCONNECTED to CONNECTING
[03-09 19:00:38]: OCPP status changed from CONNECTING to DISCONNECTED
[03-09 19:02:34]: OCPP status changed from DISCONNECTED to CONNECTING
[03-09 19:02:37]: OCPP status changed from CONNECTING to DISCONNECTED
[03-09 19:05:34]: OCPP status changed from DISCONNECTED to CONNECTING
[03-09 19:05:37]: OCPP status changed from CONNECTING to DISCONNECTED

Go-E error message from mobile app: ======================
esp_transport_connect() faled with -1, last_error=ESP_ERR:ESP_TLS_FAILED_CONNECT_TO_HOST, tls_error_code=0, tls_flags=0, errno=119
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.109.2 (3ec2d1b70)

andig, 2023-03-09:
@thomas-hutterer-tik please enable ocpp trace log on evcc side.

thomas-hutterer-tik, 2023-03-11:
Go-e connect issue is resolved by setting the OCPP url to contain the exact configured name/id of the Go-e Charger as explained in some other OCPP related ticket / many thanks. URL: ws://192.168.1.132:8887/go-e-Charger-224285

Unfortunately the reading meter value and switching the Go-e charger do not work. Will create dedicated 
tickets for this.

**The log shows the successful connection:**
Mar 11 09:50:20 iwattcontroller systemd[1]: evcc.service: Consumed 5.819s CPU time.
Mar 11 09:50:20 iwattcontroller systemd[1]: Started evcc.
Mar 11 09:50:20 iwattcontroller evcc[3395]: [main  ] INFO 2023/03/11 09:50:20 evcc 0.114.1
Mar 11 09:50:20 iwattcontroller evcc[3395]: [main  ] INFO 2023/03/11 09:50:20 using config file: /etc/evcc.yaml
Mar 11 09:50:20 iwattcontroller evcc[3395]: [main  ] INFO 2023/03/11 09:50:20 starting ui and api at :7070
Mar 11 09:50:22 iwattcontroller evcc[3395]: [db    ] INFO 2023/03/11 09:50:22 using sqlite database: /var/lib/evcc/evcc.db
Mar 11 09:50:22 iwattcontroller evcc[3395]: [mqtt  ] INFO 2023/03/11 09:50:22 connecting evcc-2055429189 at tcp://192.168.1.132:1883
Mar 11 09:50:22 iwattcontroller evcc[3395]: [mqtt  ] DEBUG 2023/03/11 09:50:22 tcp://192.168.1.132:1883 connected
Mar 11 09:50:23 iwattcontroller evcc[3395]: [ocpp  ] DEBUG 2023/03/11 09:50:23 waiting for chargepoint: 3m0s
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] INFO 2023/03/11 09:51:46 chargepoint connected, registering: go-e-Charger-224285
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 enqueued CALL [2506341193, GetConfiguration] for go-e-Charger-224285
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 dispatched request 2506341193 for go-e-Charger-224285
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 sent JSON message to go-e-Charger-224285: [2,"2506341193","GetConfiguration",{}]
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 started timeout timer for go-e-Charger-224285
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 received JSON message from go-e-Charger-224285: [3,"2506341193",{"configurationKey":[{"key":"GetConfigurationMaxKeys","readonly":true,"value":"99"},{"key":"ClockAlignedDataInterval","readonly":false,"value":"0"},{"key":"HeartbeatInterval","readonly":false,"value":"60"},{"key":"LightIntensity","readonly":false,"value":"100"},{"key":"MeterValuesSampledData","readonly":true,"value":"Current.Import,Current.Offered,Energy.Active.Import.Register,Frequency,Power.Active.Import,Power.Factor,Power.Offered,Temperature,Voltage"},{"key":"MeterValuesSampledDataMaxLength","readonly":true,"value":"10"},{"key":"MeterValueSampleInterval","readonly":false,"value":"0"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"ConnectorPhaseRotation","readonly":true,"value":"0.RST,1.RST"},{"key":"ConnectorPhaseRotationMaxLength","readonly":true,"value":"3"},{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,SmartCharging,RemoteTrigger"},{"key":"SupportedFeatureProfilesMaxLength","readonly":true,"value":"6"},{"key":"WebSocketPingInterval","readonly":false,"value":"10"},{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"20"},{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current"},{"key":"ChargingScheduleMaxPeriods","readonly":true,"value":"20"},{"key":"ConnectorSwitch3to1PhaseSupported","readonly":true,"value":"true"},{"key":"MaxChargingProfilesInstalled","readonly":true,"value":"20"}]}]
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 handling incoming CALL RESULT [2506341193] from go-e-Charger-224285
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 completed request 2506341193 for go-e-Charger-224285
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 go-e-Charger-224285 ready to transmit again
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 timeout canceled for go-e-Charger-224285
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 ChargeProfileMaxStackLevel (r/o): 20
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 ChargingScheduleAllowedChargingRateUnit (r/o): Current
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 ChargingScheduleMaxPeriods (r/o): 20
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 ClockAlignedDataInterval (r/w): 0
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 ConnectorPhaseRotation (r/o): 0.RST,1.RST
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 ConnectorPhaseRotationMaxLength (r/o): 3
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 ConnectorSwitch3to1PhaseSupported (r/o): true
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 GetConfigurationMaxKeys (r/o): 99
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 HeartbeatInterval (r/w): 60
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 LightIntensity (r/w): 100
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 MaxChargingProfilesInstalled (r/o): 20
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 MeterValueSampleInterval (r/w): 0
[... cut ...]
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 handling incoming CALL [531531533, StatusNotification] from go-e-Charger-224285
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 sent CALL RESULT [531531533] for go-e-Charger-224285
Mar 11 09:51:46 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 09:51:46 sent JSON message to go-e-Charger-224285: [3,"531531533",{}]
Mar 11 09:51:46 iwattcontroller evcc[3395]: [lp-1  ] DEBUG 2023/03/11 09:51:46 pv timer inactive
Mar 11 09:51:46 iwattcontroller evcc[3395]: [site  ] DEBUG 2023/03/11 09:51:46 ----

closed 2023-03-11 by thomas-hutterer-tik