evcc-io/evcc #6735: Ocpp: potentical improvements

andig, 2023-03-09:
https://github.com/evcc-io/evcc/discussions/3806#discussioncomment-5257599:

> Phasenumschaltung komplett unterdrücken oder bei Stromvorgaben keine Phasen mit schicken.

https://github.com/evcc-io/evcc/discussions/3806#discussioncomment-5257629:

> einige Boxen scheinen gar nicht zu antworten wenn man eine BootNotification haben will, andere scheinen die Abfrage zu erwarten. Lösungsansatz: BootNotification optional erzwingen.

/cc @tarelda

stumbaumr, 2023-03-13:
Mir kommt es so vor, als ob der Go-e sich überhaupt nicht mehr steuern lässt - das liegt nicht nur an der Übermittlung der Phasenumschaltung:
```
Mar 13 09:57:46 raspberrypi evcc[26316]: [lp-1  ] WARN 2023/03/13 09:57:46 charger logic error: disabled but charging
Mar 13 09:57:46 raspberrypi evcc[26316]: [ocpp  ] TRACE 2023/03/13 09:57:46 enqueued CALL [2017959155, RemoteStopTransaction] for go-e
Mar 13 09:57:46 raspberrypi evcc[26316]: [ocpp  ] TRACE 2023/03/13 09:57:46 dispatched request 2017959155 for go-e
Mar 13 09:57:46 raspberrypi evcc[26316]: [ocpp  ] TRACE 2023/03/13 09:57:46 sent JSON message to go-e: [2,"2017959155","RemoteStopTransaction",{"transactionId":0}]
Mar 13 09:57:46 raspberrypi evcc[26316]: [ocpp  ] TRACE 2023/03/13 09:57:46 started timeout timer for go-e
Mar 13 09:57:46 raspberrypi evcc[26316]: [ocpp  ] TRACE 2023/03/13 09:57:46 received JSON message from go-e: [3,"2017959155",{"status":"Rejected"}]
Mar 13 09:57:46 raspberrypi evcc[26316]: [ocpp  ] TRACE 2023/03/13 09:57:46 handling incoming CALL RESULT [2017959155] from go-e
Mar 13 09:57:46 raspberrypi evcc[26316]: [ocpp  ] TRACE 2023/03/13 09:57:46 completed request 2017959155 for go-e
Mar 13 09:57:46 raspberrypi evcc[26316]: [lp-1  ] ERROR 2023/03/13 09:57:46 charger: Rejected
Mar 13 09:57:46 raspberrypi evcc[26316]: [ocpp  ] TRACE 2023/03/13 09:57:46 go-e ready to transmit again
Mar 13 09:57:46 raspberrypi evcc[26316]: [ocpp  ] TRACE 2023/03/13 09:57:46 timeout canceled for go-e
```

andig, 2023-03-13:
> "RemoteStopTransaction",{"transactionId":0}

Es scheint als wäre das ein "fremder" Ladevorgang den evcc nicht kennt. Ohne Txn-ID ist der vmtl. nicht zu stoppen. Bitte neues Thema. Wenn das auch bei Start *durch* evcc auftritt.

stumbaumr, 2023-03-14:
Hab da jetzt nochmal alles "aus und wieder an" gemacht. evcc läuft zuerst, dann Sicherung rein vom Go-e und dann Auto angesteckt. Es scheint keine Sonne, das Auto fängt zum Laden an und dann sagt evcc, er soll ausmachen.
In der Konfig des chargers steht nur noch `type: ocpp`.

Auch beim Stoppen der transactionId 1 funktioniert das nicht.
Und dann probiert evcc das mit transactionId 0.

```
Mar 14 10:52:28 raspberrypi evcc[27292]: [main  ] INFO 2023/03/14 10:52:28 evcc 0.114.1 (14d2ba9b)
Mar 14 10:52:28 raspberrypi evcc[27292]: [main  ] INFO 2023/03/14 10:52:28 using config file: /etc/evcc.yaml
Mar 14 10:52:28 raspberrypi evcc[27292]: [main  ] INFO 2023/03/14 10:52:28 starting ui and api at :7070
Mar 14 10:52:29 raspberrypi evcc[27292]: [db    ] INFO 2023/03/14 10:52:29 using sqlite database: /var/lib/evcc/evcc.db
Mar 14 10:52:30 raspberrypi evcc[27292]: [ocpp  ] DEBUG 2023/03/14 10:52:30 waiting for chargepoint: 1m0s
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] INFO 2023/03/14 10:52:43 chargepoint connected, registering: go-e
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 enqueued CALL [3596933275, GetConfiguration] for go-e
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 dispatched request 3596933275 for go-e
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 sent JSON message to go-e: [2,"3596933275","GetConfiguration",{}]
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 started timeout timer for go-e
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 received JSON message from go-e: [3,"3596933275",{"configurationKey":[{"key":"AllowOfflineTxForUnknownId","readonly":false,"value":"0"},{"key":"GetConfigurationMaxKeys","readonly":true,"value":"99"},{"key":"ClockAlignedDataInterval","readonly":false,"value":"60"},{"key":"LocalAuthorizeOffline","readonly":false,"value":"0"},{"key":"HeartbeatInterval","readonly":false,"value":"60"},{"key":"LightIntensity","readonly":false,"value":"49"},{"key":"MeterValuesSampledData","readonly":true,"value":"Current.Import,Current.Offered,Energy.Active.Import.Register,Frequency,Power.Active.Import,Power.Factor,Power.Offered,Temperature,Voltage"},{"key":"MeterValuesSampledDataMaxLength","readonly":true,"value":"10"},{"key":"MeterValueSampleInterval","readonly":false,"value":"60"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"ConnectorPhaseRotation","readonly":true,"value":"0.RST,1.RST"},{"key":"ConnectorPhaseRotationMaxLength","readonly":true,"value":"3"},{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,SmartCharging,RemoteTrigger"},{"key":"SupportedFeatureProfilesMaxLength","readonly":true,"value":"6"},{"key":"WebSocketPingInterval","readonly":false,"value":"10"},{"key":"LocalAuthListEnabled","readonly":false,"value":"0"},{"key":"LocalAuthListMaxLength","readonly":true,"value":"10"},{"key":"SendLocalListMaxLength","readonly":true,"value":"10"},{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"20"},{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current"},{"key":"ChargingScheduleMaxPeriods","readonly":true,"value":"20"},{"key":"ConnectorSwitch3to1PhaseSupported","readonly":true,"value":"true"},{"key":"MaxChargingProfilesInstalled","readonly":true,"value":"20"}]}]
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 handling incoming CALL RESULT [3596933275] from go-e
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 completed request 3596933275 for go-e
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 go-e ready to transmit again
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 timeout canceled for go-e
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 AllowOfflineTxForUnknownId (r/w): 0
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 ChargeProfileMaxStackLevel (r/o): 20
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 ChargingScheduleAllowedChargingRateUnit (r/o): Current
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 ChargingScheduleMaxPeriods (r/o): 20
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 ClockAlignedDataInterval (r/w): 60
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 ConnectorPhaseRotation (r/o): 0.RST,1.RST
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 ConnectorPhaseRotationMaxLength (r/o): 3
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 ConnectorSwitch3to1PhaseSupported (r/o): true
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 GetConfigurationMaxKeys (r/o): 99
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 HeartbeatInterval (r/w): 60
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 LightIntensity (r/w): 49
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 LocalAuthListEnabled (r/w): 0
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 LocalAuthListMaxLength (r/o): 10
Mar 14 10:52:43 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:52:43 LocalAuthorizeOffline (r/w): 0
[... cut ...]
Mar 14 10:54:02 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:54:02 handling incoming CALL RESULT [2299380979] from go-e
Mar 14 10:54:02 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:54:02 completed request 2299380979 for go-e
Mar 14 10:54:02 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:54:02 go-e ready to transmit again
Mar 14 10:54:02 raspberrypi evcc[27292]: [ocpp  ] TRACE 2023/03/14 10:54:02 timeout canceled for go-e
Mar 14 10:54:02 raspberrypi evcc[27292]: [lp-1  ] ERROR 2023/03/14 10:54:02 charger: Rejected
```

andig, 2023-03-14:
> Mir kommt es so vor, als ob der Go-e sich überhaupt nicht mehr steuern lässt

Mach bitte ein neues Thema, ist hier off-topic. Ich denke hier wäre es sinnvoll die Ursache einzugrenzen (Go-E FW vs. evcc Version). Dafür gerne nochmal alte evcc Versionen die nachweislich funktionierten prüfen. Oder einfach die direkte Go-E Integration nutzen ;)

stumbaumr, 2023-03-14:
Ich benutze im "Produktivbetrieb" die direkte Go-E Integration. Das OCPP ist rumspielen, wenn das Auto mal da steht und kaum Sonne scheint. Ich will halt verstehen, wie OCPP funktioniert, wenn das Laden der Firmenwagen zuhause eben nicht mehr einfach pauschal oder nach selbst gemeldeten Zählerständen bezahlt wird. Das wird wohl dann auf OCPP hinauslaufen und ein Dienstleister oder die Firma hat irgendwo ein OCPP Backend stehen für die Abrechnung.

closed 2023-03-18 by andig