evcc-io/evcc #6738: Shelly EM missing

spikey-de, 2023-03-09:
Folks,

as I have switched from Shelly 1 PM plus to Shelly EM as the measurement is more exact, I'd appreciate if the Shelly EM could be integrated into EVCC. I am using it as power meter for my mini-PV.

Thanks Spikey

JanDragon, 2023-03-09:
As far as I know there is already a Shelly EM integration: 

https://docs.evcc.io/docs/devices/meters/#shelly-1pm-em

The Shelly integration was updated in a recent Evcc update.

spikey-de, 2023-03-09:
Arghhh...painfull....I deleted the 1PM config to and tried to update it to EM, but that was it...

closed 2023-03-09 by spikey-de

JanDragon, 2023-03-09:
> Arghhh...painfull....I deleted the 1PM config to and tried to update it to EM, but that was it...

Yeah the EM integration template should be named: shelly-pro-3em

spikey-de, 2023-03-09:
Exactly, I looked to "EM" only and in the YAML the template is called "1PM"....

Nevertheless, thanks!