evcc-io/evcc #6739: Error when running "make ui"

jheinitz, 2023-03-09:
Hello!

I'm trying to build the `evcc`binary locally on my linux system. I'm running `Debian 11`. I cloned the repo and installed `go 1.20` as well as `node 18` as stated as requirements in the `README.md`. I followed the instructions in the `README.md` and started to run `make ui`. Here is what I got:

```
jens@server2:~/git/evcc$ make ui
npm run build

> build
> vite build

qt.qpa.xcb: X server does not support XInput 2
failed to get the current screen resources
(./src/core/Core.cpp l.710): Opening the file: build
qt.qpa.xcb: QXcbConnection: XCB error: 1 (BadRequest), sequence: 165, resource id: 90, major code: 130 (Unknown), minor code: 47
qt.qpa.xcb: QXcbConnection: XCB error: 170 (Unknown), sequence: 180, resource id: 90, major code: 146 (Unknown), minor code: 20
Cannot open file build
Aborted
make: *** [Makefile:44: ui] Error 134
jens@server2:~/git/evcc$
```

Do I have to create the file **build** manually or is it simply missing in the repo?

Thanks and best regards

Jens

PS: I also tried to run `make install` which installed the go libraries, but did not solve the problem.

naltatis, 2023-03-10:
Ok, thats a strange error. Do you get the same of you run `npm run build` directly?

jheinitz, 2023-03-10:
Yes, I get the same error: 
```
jens@server2:~/git/evcc$ npm run build

> build
> vite build

qt.qpa.xcb: X server does not support XInput 2
failed to get the current screen resources
(./src/core/Core.cpp l.710): Opening the file: build
qt.qpa.xcb: QXcbConnection: XCB error: 1 (BadRequest), sequence: 165, resource id: 90, major code: 130 (Unknown), minor code: 47
qt.qpa.xcb: QXcbConnection: XCB error: 170 (Unknown), sequence: 180, resource id: 90, major code: 146 (Unknown), minor code: 20
Cannot open file build
Aborted
jens@server2:~/git/evcc$
```
I try to locate any file with **build** in it's name, but there is nothing:
```
jens@server2:~/git/evcc$ find . -name "*build*"
jens@server2:~/git/evcc$
```
Strange.

naltatis, 2023-03-10:
I`m not sure what `build` file he's looking for. Can you try these as well?

```
npm run dev
```

```
nix vite build
```

jheinitz, 2023-03-10:
> I'm not sure what `build` file he's looking for. Can you try these as well?
> 
> ```
> npm run dev
> ```
This gives me an empty  `Vite`GUI.
> 
> ```
> npx vite build
> ```
This gives  me this:
```
jens@server2:~/git/evcc$ npx vite build
qt.qpa.xcb: X server does not support XInput 2
failed to get the current screen resources
(./src/core/Core.cpp l.710): Opening the file: build
qt.qpa.xcb: QXcbConnection: XCB error: 1 (BadRequest), sequence: 165, resource id: 90, major code: 130 (Unknown), minor code: 47
qt.qpa.xcb: QXcbConnection: XCB error: 170 (Unknown), sequence: 180, resource id: 90, major code: 146 (Unknown), minor code: 20
Cannot open file build
Aborted
jens@server2:~/git/evcc$```

jheinitz, 2023-03-10:
Just added my versions in the original post. Is everything fine with my `npm`?

naltatis, 2023-03-10:
What's an `This gives me an empty ViteGUI`? 

You should see something like this:
```
$ npm run dev

> dev
> vite

  VITE v3.2.5  ready in 166 ms

  ➜  Local:   http://127.0.0.1:7071/
  ➜  Network: use --host to expose
```

Do you have some other program named `vite` installed on your systems that conflicts with the npm package?

jheinitz, 2023-03-10:
I have a package `vite` installed on my system:
```
jens@server2:~/git/evcc$ dpkg -l | grep vite
ii  vite                                  1.2+svn+git4.c6c0ce7-3           amd64        Efficient visual trace explorer
jens@server2:~/git/evcc$
```
At least it is an older version. Is that the right tool? After removing the package, I get (which is what I expect):
```
jens@server2:~/git/evcc$ npm run dev

> dev
> vite

sh: 1: vite: not found
jens@server2:~/git/evcc$
```
I will try to get `v3.2.5` of `vite` and try again .....

Thanks

andig, 2023-03-10:
Nodejs zu alt?

jheinitz, 2023-03-10:
So, ich glaube ich hab's: 
1. vite initialisiert mit `npm init vite`
2. Installation mit `npm install`
3. Build Prozeß gestartet mit `make`

Nun läuft der Build Prozeß ......

closed 2023-03-10 by jheinitz