evcc-io/evcc #6745: Fragen zum Ladestart bei DaheimLader Modbus

VolkerK62, 2023-03-10:
Der Ladestart ist etwas "holprig".
Siehe Log-File.
10:18:32 enable mit charge current 6A

eine Sekunde später
10:18:33 charge current 20A

ein Intervall später
10:18:50 start charging mit 6A

dann bleibt es bei 6A und erst eine Minute später kommt 
10:19:50 charge current 18A
und ab da funktioniert die Regelung wunderbar.

Warum bleibt es am Anfang eine Minute lang bei 6A?
Mag kleinlich sein ;-) , aber evtl lässt sich das ja ganz einfach verbessern.
Oder hat das evtl mit dem Umschalten auf `minpv` (10:19:05) zu tun?

Und wo kommen die 7,5A her?
`DEBUG 2023/03/10 10:18:50 pv charge current: 24.3A = 7.5A + 16.8A (-3868W @ 1p)`

```
Mär 10 10:18:04 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:04 ----
Mär 10 10:18:04 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:04 charge power: 0W
Mär 10 10:18:04 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:04 pv power: 4976W
Mär 10 10:18:04 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:04 battery soc: 20%
Mär 10 10:18:04 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:04 battery power: -1165W
Mär 10 10:18:04 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:04 grid power: -5102W
Mär 10 10:18:05 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:05 grid powers: [-1324 -1432 -2346]W
Mär 10 10:18:05 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:05 grid currents: [-5.84 -6.29 -10.2]A
Mär 10 10:18:05 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:05 site power: -6017W
Mär 10 10:18:05 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:05 charge voltages: [232 0 0]V
Mär 10 10:18:05 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:05 detected connected phases: 1p
Mär 10 10:18:05 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:05 charge currents: [0 0 0]A
Mär 10 10:18:05 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:05 charger status: B
Mär 10 10:18:05 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:05 next soc poll remaining time: 3h33m15s
Mär 10 10:18:15 raspberrypi evcc[21251]: [lp-1  ] INFO 2023/03/10 10:18:15 vehicle updated: ohne Bluelink -> (T)Raumschiff
Mär 10 10:18:15 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:15 set charge mode: pv
Mär 10 10:18:15 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:15 set target soc: 100
Mär 10 10:18:15 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:15 ----
Mär 10 10:18:15 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:15 charge power: 0W
Mär 10 10:18:15 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:15 pv power: 7206W
Mär 10 10:18:15 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:15 battery soc: 20%
Mär 10 10:18:15 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:15 battery power: -1143W
Mär 10 10:18:15 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:15 grid power: -3890W
Mär 10 10:18:15 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:15 grid powers: [-922 -1030 -1983]W
Mär 10 10:18:15 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:15 grid currents: [-4.29 -4.71 -8.57]A
Mär 10 10:18:15 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:15 site power: -4783W
Mär 10 10:18:15 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:15 vehicle odometer: 13248km
Mär 10 10:18:15 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:15 charge voltages: [232 0 0]V
Mär 10 10:18:15 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:15 detected connected phases: 1p
Mär 10 10:18:15 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:15 charge currents: [0 0 0]A
Mär 10 10:18:15 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:15 charger status: B
Mär 10 10:18:32 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:32 vehicle status: A
Mär 10 10:18:32 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:32 vehicle soc: 67%
Mär 10 10:18:32 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:32 vehicle soc limit: 100%
Mär 10 10:18:32 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:32 vehicle range: 260km
Mär 10 10:18:32 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:32 pv charge current: 20.8A = 0A + 20.8A (-4783W @ 1p)
Mär 10 10:18:32 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:32 site power -4783W <= 0W enable threshold
Mär 10 10:18:32 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:32 pv enable in 0s
Mär 10 10:18:32 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:32 pv enable timer elapsed
Mär 10 10:18:32 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:32 max charge current: 6A
Mär 10 10:18:32 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:32 charger enable
Mär 10 10:18:32 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:32 wake-up timer: start
Mär 10 10:18:32 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:32 ----
Mär 10 10:18:32 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:32 charge power: 0W
Mär 10 10:18:32 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:32 pv power: 7677W
Mär 10 10:18:32 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:32 battery soc: 20%
Mär 10 10:18:32 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:32 battery power: -1155W
Mär 10 10:18:32 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:18:32 grid power: -4151W
[... 169 lines cut ...]
Mär 10 10:20:05 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:20:05 pv charge current: 10.5A = 18A + -7.45A (1714W @ 1p)
Mär 10 10:20:05 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:20:05 max charge current: 10A
Mär 10 10:20:19 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:20:19 ----
Mär 10 10:20:19 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:20:19 charge power: 2000W
Mär 10 10:20:19 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:20:19 pv power: 3305W
Mär 10 10:20:19 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:20:19 battery soc: 20%
Mär 10 10:20:19 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:20:19 battery power: 66W
Mär 10 10:20:20 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:20:20 grid power: 999W
Mär 10 10:20:20 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:20:20 grid powers: [-809 267 1541]W
Mär 10 10:20:20 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:20:20 grid currents: [-3.58 1.52 7.13]A
Mär 10 10:20:20 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:20:20 site power: 1314W
Mär 10 10:20:20 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:20:20 charge voltages: [229 0 0]V
Mär 10 10:20:20 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:20:20 detected connected phases: 1p
Mär 10 10:20:20 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:20:20 charge currents: [9.4 0 0]A
Mär 10 10:20:20 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:20:20 detected active phases: 1p
Mär 10 10:20:20 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:20:20 charger status: C
Mär 10 10:20:20 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:20:20 soc estimated: 67.00% (vehicle: 67.00%)
Mär 10 10:20:20 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:20:20 vehicle soc: 67%
Mär 10 10:20:20 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:20:20 vehicle soc limit: 100%
Mär 10 10:20:20 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:20:20 vehicle range: 260km
Mär 10 10:20:20 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:20:20 pv charge current: 4.28A = 10A + -5.72A (1314W @ 1p)
Mär 10 10:20:20 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:20:20 max charge current: 6A
Mär 10 10:20:20 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:20:20 telemetry: charge: Δ25/29Wh @ 2000W
Mär 10 10:20:34 raspberrypi evcc[21251]: [site  ] DEBUG 2023/03/10 10:20:34 ----
```

andig, 2023-03-10:
Entweder ignoriert die WB die Stromvorgabe oder das Auto konsumiert nicht oder Deine Strommesswerte sind falsch.

VolkerK62, 2023-03-10:
Auto würde ich ausschließen, wenn ich mit "schnell" starte, wird sofort mit 20A geladen.
Strommesswerte würde ich auch ausschließen, weil ansonsten alles funktioniert.
Ich vermute, dass die 20A (eine Sekunde nach enable) nicht "ankommen".
Wenn es nach einer Minute nicht diese Änderung auf 18A gegeben hätte, sondern weiterhin 20A und mehr möglich gewesen wären, dann wäre der 6A Zeitraum wahrscheinlich noch länger gewesen.

VolkerK62, 2023-03-10:
Als workaround?
Du schickst am Anfang 6A und dann direkt den "richtigen" Ladestrom hinterher. 
Könnte man nicht sofort den richtigen Ladestrom schicken?

Oder alternativ, die 6A ganz weglassen (also 0A) und dann eine 5-Sekunden-Pause bis der "richtige" Ladestrom geschickt wird.
Das habe ich vorhin mal extern (Node Red) so gebaut, das hat perfekt funktioniert.

andig, 2023-03-10:
> Ich vermute, dass die 20A (eine Sekunde nach enable) nicht "ankommen".

Kann ich nicht beurteilen, wäre aber ein Bug der Wallbox. Ich halte es auch für unwahrscheinlich, da evcc ja mit jedem Intervall wieder eine neue Stromvorgabe schickt. Warum sollte die dann nicht greifen? Es sei denn die WB ignoriert eine ganze Weile...

Ich sehe jdfls erstmal nix was auf einen Fehler in evcc hindeuten würde?

VolkerK62, 2023-03-10:
In der Spezifikation von DaheimLaden heißt es zu Register 91 (LadestromLimit):
```
Anpassen der Limitierung des Ladestroms
Kann während des Ladevorgangs angepasst werden. Die Anpassung des
maximalen Ladestroms erfolgt innerhalb von 4 Sekunden. Befindet sich die
Ladestation nicht im Ladevorgang so kommt die Einstellung zum nächsten
Ladevorgang zum Einsatz.
```
Evtl sind hier zwei sehr kurz aufeinanderfolgende Änderungen zu viel.

> Ich halte es auch für unwahrscheinlich, da evcc ja mit jedem Intervall wieder eine neue Stromvorgabe schickt.

Ich trau mich kaum, dir zu widersprechen. Oben im Log sieht man, dass nach dem Start mit 6A einmal 20A geschickt werden.
Danach ist der Wert von `pv charge current` immer über 20A. Da wird keine neue Stromvorgabe geschickt.
Erst als dieser wert unter 20A fällt, kommt eine neue Stromvorgabe mit 18A, die dann auch befolgt wird.

Das Problem mMn ist die doppelte Stromvorgabe beim Ladestart.
```
Mär 10 10:18:32 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:32 max charge current: 6A
Mär 10 10:18:33 raspberrypi evcc[21251]: [lp-1  ] DEBUG 2023/03/10 10:18:33 max charge current: 20A
```

andig, 2023-03-10:
> Ich trau mich kaum, dir zu widersprechen. 

Du hast recht. Wenn das dein MAX Wert ist dann schreiben wir den nicht erneut, er ist ja schon gesetzt. Die Box muss den bitte befolgen.

premultiply, 2023-03-10:
Das Erste ist das Resultat von Enable(true), das Zweite das Einstellen der Ladestromlimitierung via MaxCurrent(20)

VolkerK62, 2023-03-10:
Könnte man den Zweiten um 5 Sek verzögern?
oder
Könnte man das Resultat von `Enable(true)` gleich mit der richtigen Limitierung füttern? 
oder
Könnte man die Limitierung vorher schicken?
```
 Befindet sich die
Ladestation nicht im Ladevorgang so kommt die Einstellung zum nächsten
Ladevorgang zum Einsatz.
```

premultiply, 2023-03-10:
1) nein, das würde den ganzen Prozess blockieren und ob das Kommando dann nach 5 Sek zuverlässig verarbeitet wird ist damit auch noch nicht gesagt.
2&3) Schwierig. Muss ich mir nochmal genau ansehen. Das kommt aus den Tiefen der Loadpoint-Implementierung und basiert auf ein paar bisherigen evcc-Grundprinzipien.

Ursächlich bleibt aber das Problem dass die Firmware das zweite Kommando letztendlich einfach ignoriert.
Man könnte natürlich da noch weiter drumherum bauen und auch den Ist-Stromzustand abfragen, und mit dem Soll-Zustand vergleichen und ggf. immer wieder schreiben bis die Box tatsächlich das macht was gewollt ist.
Dann werden es immer mehr Workarounds.

VolkerK62, 2023-03-10:
Ok. Das hab ich mir dann zu einfach vorgestellt. An Grundprinzipien will ich nicht rütteln.
Dann hoffe ich auf eine Firmwareanpassung. @stephanschwebe
Solange werde ich mir etwas externes bauen.

VolkerK62, 2023-03-11:
@premultiply gibt es in der Implementierung eine Bedingung, damit zusammen mit `charger enable` auch `max charge current 6A` geschickt wird? Wage Vermutung aus meinen Beobachtungen: nur beim ersten Start?
Beim testen meines workaround ist mir aufgefallen, dass bei weiteren Starts nur `charger enable` im Log auftaucht. In dem Fall kommt danach ein `charger out of sync` und danach erst `max charge current xA`. Und ab da funktioniert es problemlos.

premultiply, 2023-03-11:
Ja, die Implementierungen werden immer mit 6A initialisiert.
Danach halt der zuletzt gesetzte Wert.

VolkerK62, 2023-03-11:
Das passt zusammen. Der zuletzt gesetzte Wert ist `0`. Deshalb kommt dann im UI `Ladebereit, wartet auf Fahrzeug` und einmalig `charger out of sync` bevor es richtig losgeht.
Dann besteht das Eingangsproblem also nur beim ersten Start, weil da die "richtige" Stromvorgabe nicht umgesetzt wird.

andig, 2023-03-13:
Gibts hier noch was zu tun?

VolkerK62, 2023-03-13:
Beste Lösung wäre eine Firmwareanpassung.
Alternativ eine Anpassung der Implementierung in evcc, aber das wäre dann ein Workaround für den Workaround.

Ich habe für mich eine externe Lösung für den Ladestart gefunden.
Es kann aber jedem anderen passieren, dass längere Zeit nur mit 6A geladen wird, wenn beim ersten Ladestart gleich die volle Leistung zur Verfügung steht und dann keine Änderung kommt.

andig, 2023-03-13:
Ich sehe nichts was wir sinnvoll ändern könnten. Wir müssen uns drauf verlassen können, dass Ladebefehle ohne Fehlermeldung auch entsprechend umgesetzt werden.

closed 2023-03-13 by andig