evcc-io/evcc #6747: Allow different powers for 1-wired and 3-wired charging / charging profiles

felix-schwarz, 2023-03-10:
**Is your feature request related to a problem? Please describe.**
For some EVs, the range of supported charging powers differs between 1-wire and 3-wire charging.
Example: the Twingo Electric supports 6A-16A 1-wire but only 8A-16A 3-wire.

In addition, some charging power levels are more power efficient than others.
Example: the Twingo Electric has only about 80% efficiency at 6A (1-wire), but >90% efficiency at 7A+ (1-wire). Using 3-wired charging, the sweet spot seems to be 11A at 95% efficiency.

Right now, with 7A minimum charging power, the Twingo will charge at 1-wire, but no longer charge when the switch to 3-wired charging occurs, requiring me to manually control the charger to provide >=8A and essentially work "against" evcc.

**Describe the solution you'd like**
The minimalist solution I'd like to see is that one can set separate ranges for 1-wired and 3-wired charging per vehicle.
Even better, it should be possible to provide a list of charging power levels, too - in case a vehicle doesn't support certain levels or is inefficient at a particular level.

The bigger solution I can imagine is to have complete charging profiles, replacing the hardcoded Off/PV/Min+PV/Fast levels, with a profile allowing me to:
- define requirements before charging starts and for charging to end
- define allowed power levels for 1-wire and 3-wire charging individually

This would allow to define a "PV" profile that begins charging at 6A (1-wire) with 80% efficiency, but use at least 7A (1-wire) for "Min+PV" to not waste as much energy when it needs to paid. Also different profiles for efficient "Medium" speed charging  (11A 3-wire) and a possibly less efficient "Fast" profile charging with 16A 3-wire.

felix-schwarz, 2023-03-12:
Wo im Code müsste ich denn ansetzen, um eine Implementierung zu versuchen?

andig, 2023-03-12:
Das muss im Loadpoint passieren (loadpoint.go)- ist aber leider auch die komplexeste Ecke :(