evcc-io/evcc #6768: No vehicle soc for VW ID.5

timostr, 2023-03-11:
Since couple of days the SOC is not shown anymore.
Error message in log:
`vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'id': terms of service updated- please open app or website and confirm`

No new terms of condition in website (https://vwid.vwgroup.io/) or in WeConnect app.

jpwenzel, 2023-03-11:
Hey @timostr, I experienced the same issue last night with my Golf.
You need to log in to My Volkswagen (https://www.volkswagen.de/de/besitzer-und-nutzer/myvolkswagen.html) and accept the new terms & conditions.
That fixed it for me. :)
I did not notice any new T&Cs in the We Connect app, though. :/

timostr, 2023-03-11:
Hi,
it seems to be a different problem, because there are no new terms &
conditions displayed to me.  Neither at your mentioned webpage nor in the
App.

thebrainkafka, 2023-03-11:
Maybe you need to login on a pc or Laptop Webbrowser.

timostr, 2023-03-11:
Done all that (login PC, login App, login phone browser).

closed 2023-03-11 by andig