evcc-io/evcc #6769: prioritySoc wird nicht berücksichtigt

zenobios, 2023-03-11:
Hi zusammen,

ich habe evcc als Home Assistent Add-on eingerichtet, funktioniert an sich auch gut.

Allerdings ist mir aufgefallen, dass der eingestellte Wert von `prioritySoc` nicht berücksichtigt wird.
[image]

Laut Docu würde ich erwarten, dass dass das Fahrzeug nicht geladen wird, da die Batterie nur 27% soc hat und auch keine Energie in die Batterie fließt. Geladen wird aber trotzdem:

[image]

Hab ich vielleicht irgendwas nicht richtig konfiguriert?

thebrainkafka, 2023-03-11:
#6772 sieht auch so aus

thebrainkafka, 2023-03-11:
[image]

Bei mir ProritySoC 60. Funktioniert wie es soll.

zenobios, 2023-03-11:
Irgendwelche Logfiles die hier hilfreich wären?

closed 2023-03-11 by andig