evcc-io/evcc #6771: Charging not stated with Go-e over OCPP

thomas-hutterer-tik, 2023-03-11:
### Describe the bug

I have a Go-e charger integrated to EVCC over OCPP (as a test installation to use OCPP for non supported wall boxes like ELinta and xCharge). The connection of the Go-e to EVCC works but to start charging does not work. 
The RemoteStartTransaction gets "status":"Accepted", but the charging does not start => charger status: B
I have tested this with 55.0, 55.4 and 55.5. version of the Go-e firmware. Creating the ticket here for EVCC, because maybe this is an issue if the RemoteStartTransaction command issued by EVCC. 

### Steps to reproduce

1. configure OCPP charger in EVCC config
2. enable OCPP 1.6 on Go-e app and set URL to ws://192.168.1.132:8887/go-e-Charger-224285
3. connect a car
4. press mode=Fast on EVCC UI
5. observe that the charging does not start
6. watch the logs

### Configuration details

```yaml
---
network:
  schema: "http"
  host: "0.0.0.0"
  port: 7070
interval: "2s"
sponsortoken: "..."
log: "debug"
levels:
  ocpp: trace
meters:
- name: "eMallorca Lab"
  type: "custom"
  power:
    source: "mqtt"
    topic: "wa-pi/grid"
  currents:
  - source: "mqtt"
    topic: "wa-pi/grid/p1"
  - source: "mqtt"
    topic: "wa-pi/grid/p2"
  - source: "mqtt"
    topic: "wa-pi/grid/p3"
chargers:
- name: "OCPP-1"
  type: "template"
  template: "ocpp"
  idtag: 1
  timeout: 3m
site:
  title: "eMallorca"
  meters:
    grid: "eMallorca Lab"
    pvs: []
    batteries: []
loadpoints:
- title: "OCPP-1"
  charger: "OCPP-1"
  phases: 1
  mode: "now"
mqtt:
  broker: "192.168.1.132"
  topic: "evcc"
  user: "..."
  password: "..."
```

### Log details

```text
Mar 11 10:01:50 iwattcontroller evcc[3395]: [site  ] DEBUG 2023/03/11 10:01:50 ----
Mar 11 10:01:50 iwattcontroller evcc[3395]: [lp-1  ] DEBUG 2023/03/11 10:01:50 set charge mode: now
Mar 11 10:01:50 iwattcontroller evcc[3395]: [lp-1  ] ERROR 2023/03/11 10:01:50 charge meter: not available
Mar 11 10:01:50 iwattcontroller evcc[3395]: [site  ] DEBUG 2023/03/11 10:01:50 grid power: 553W
Mar 11 10:01:50 iwattcontroller evcc[3395]: [site  ] DEBUG 2023/03/11 10:01:50 grid currents: [1.22 0.683 0.368]A
Mar 11 10:01:50 iwattcontroller evcc[3395]: [site  ] DEBUG 2023/03/11 10:01:50 pv power: 0W
Mar 11 10:01:50 iwattcontroller evcc[3395]: [site  ] DEBUG 2023/03/11 10:01:50 site power: 553W
Mar 11 10:01:50 iwattcontroller evcc[3395]: [lp-1  ] ERROR 2023/03/11 10:01:50 meter energy: not available
Mar 11 10:01:50 iwattcontroller evcc[3395]: [lp-1  ] DEBUG 2023/03/11 10:01:50 charger status: B
Mar 11 10:01:50 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:01:50 enqueued CALL [3433559852, RemoteStartTransaction] for go-e-Charger-224285
Mar 11 10:01:50 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:01:50 dispatched request 3433559852 for go-e-Charger-224285
Mar 11 10:01:50 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:01:50 sent JSON message to go-e-Charger-224285: [2,"3433559852","RemoteStartTransaction",{"connectorId":1,"idTag":"1","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":16,"numberPhases":1}]}}}]
Mar 11 10:01:50 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:01:50 started timeout timer for go-e-Charger-224285
Mar 11 10:01:50 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:01:50 received JSON message from go-e-Charger-224285: [2,"531531555","StartTransaction",{"connectorId":1,"idTag":"1","meterStart":30142,"timestamp":"2023-03-11T09:01:50.000Z"}]
Mar 11 10:01:50 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:01:50 handling incoming CALL [531531555, StartTransaction] from go-e-Charger-224285
Mar 11 10:01:50 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:01:50 sent CALL RESULT [531531555] for go-e-Charger-224285
Mar 11 10:01:50 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:01:50 sent JSON message to go-e-Charger-224285: [3,"531531555",{"idTagInfo":{"status":"Accepted"},"transactionId":4}]
Mar 11 10:01:50 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:01:50 received JSON message from go-e-Charger-224285: [3,"3433559852",{"status":"Accepted"}]
Mar 11 10:01:50 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:01:50 handling incoming CALL RESULT [3433559852] from go-e-Charger-224285
Mar 11 10:01:50 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:01:50 completed request 3433559852 for go-e-Charger-224285
Mar 11 10:01:50 iwattcontroller evcc[3395]: [lp-1  ] DEBUG 2023/03/11 10:01:50 charger enable
Mar 11 10:01:50 iwattcontroller evcc[3395]: [lp-1  ] DEBUG 2023/03/11 10:01:50 wake-up timer: start
Mar 11 10:01:50 iwattcontroller evcc[3395]: [site  ] DEBUG 2023/03/11 10:01:50 ----
Mar 11 10:01:50 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:01:50 go-e-Charger-224285 ready to transmit again
Mar 11 10:01:50 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:01:50 timeout canceled for go-e-Charger-224285
Mar 11 10:01:51 iwattcontroller evcc[3395]: [lp-1  ] ERROR 2023/03/11 10:01:51 charge meter: not available
Mar 11 10:01:51 iwattcontroller evcc[3395]: [site  ] DEBUG 2023/03/11 10:01:51 grid power: 553W
Mar 11 10:01:51 iwattcontroller evcc[3395]: [site  ] DEBUG 2023/03/11 10:01:51 grid currents: [1.22 0.694 0.359]A
Mar 11 10:01:51 iwattcontroller evcc[3395]: [site  ] DEBUG 2023/03/11 10:01:51 pv power: 0W
Mar 11 10:01:51 iwattcontroller evcc[3395]: [site  ] DEBUG 2023/03/11 10:01:51 site power: 553W
Mar 11 10:01:51 iwattcontroller evcc[3395]: [lp-1  ] ERROR 2023/03/11 10:01:51 meter energy: not available
Mar 11 10:01:51 iwattcontroller evcc[3395]: [lp-1  ] DEBUG 2023/03/11 10:01:51 charger status: B
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.114.1

andig, 2023-03-11:
@thomas-hutterer-tik does charging actually not start or does just evcc not show that it has started?

thomas-hutterer-tik, 2023-03-12:
Charging does not start. If I in this state unplug the car and plug it in again, then charging starts