evcc-io/evcc #6774: Go-e via OCPP charge meter: not available, meter energy: not available

thomas-hutterer-tik, 2023-03-11:
### Describe the bug

I have connected a Go-e Charger via OCPP to EVCC and see error logs that charge meter and meter energy are not available

### Steps to reproduce

1. connected a Go-e Charger via OCPP to EVCC
2. restart the Go-e charger to trigger connect process
3. observe the evcc logs
...

### Configuration details

```yaml
---
network:
  schema: "http"
  host: "0.0.0.0"
  port: 7070
interval: "2s"
sponsortoken: "..."
log: "debug"
levels:
  ocpp: trace
meters:
- name: "eMallorca Lab"
  type: "custom"
  power:
    source: "mqtt"
    topic: "wa-pi/grid"
  currents:
  - source: "mqtt"
    topic: "wa-pi/grid/p1"
  - source: "mqtt"
    topic: "wa-pi/grid/p2"
  - source: "mqtt"
    topic: "wa-pi/grid/p3"
chargers:
- name: "OCPP-1"
  type: "template"
  template: "ocpp"
  host: "192.168.1.203"
  idtag: 1
  timeout: 3m
site:
  title: "eMallorca"
  meters:
    grid: "eMallorca Lab"
    pvs: []
    batteries: []
loadpoints:
- title: "OCPP-1"
  charger: "OCPP-1"
  phases: 1
  mode: "now"
mqtt:
  broker: "192.168.1.132"
  topic: "evcc"
  user: "..."
  password: "..."
```

### Log details

```text
Mar 11 10:43:51 iwattcontroller evcc[3395]: [ocpp  ] DEBUG 2023/03/11 10:43:51 chargepoint connected: go-e-Charger-224285
Mar 11 10:43:51 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:43:51 received JSON message from go-e-Charger-224285: [2,"531531531","BootNotification",{"chargePointVendor":"go-e","chargePointModel":"go-eCharger_V4","chargePointSerialNumber":"224285","firmwareVersion":"055.0"}]
Mar 11 10:43:51 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:43:51 handling incoming CALL [531531531, BootNotification] from go-e-Charger-224285
Mar 11 10:43:51 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:43:51 sent CALL RESULT [531531531] for go-e-Charger-224285
Mar 11 10:43:51 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:43:51 sent JSON message to go-e-Charger-224285: [3,"531531531",{"currentTime":"2023-03-11T09:43:51Z","interval":60,"status":"Accepted"}]
Mar 11 10:43:51 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:43:51 received JSON message from go-e-Charger-224285: [2,"531531532","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2023-03-11T09:43:51.000Z"}]
Mar 11 10:43:51 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:43:51 handling incoming CALL [531531532, StatusNotification] from go-e-Charger-224285
Mar 11 10:43:51 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:43:51 sent CALL RESULT [531531532] for go-e-Charger-224285
Mar 11 10:43:51 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:43:51 sent JSON message to go-e-Charger-224285: [3,"531531532",{}]
Mar 11 10:43:51 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:43:51 received JSON message from go-e-Charger-224285: [2,"531531533","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Finishing","timestamp":"2023-03-11T09:43:51.000Z"}]
Mar 11 10:43:51 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:43:51 handling incoming CALL [531531533, StatusNotification] from go-e-Charger-224285
Mar 11 10:43:51 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:43:51 sent CALL RESULT [531531533] for go-e-Charger-224285
Mar 11 10:43:51 iwattcontroller evcc[3395]: [ocpp  ] TRACE 2023/03/11 10:43:51 sent JSON message to go-e-Charger-224285: [3,"531531533",{}]
Mar 11 10:43:52 iwattcontroller evcc[3395]: [site  ] DEBUG 2023/03/11 10:43:52 ----
Mar 11 10:43:52 iwattcontroller evcc[3395]: [lp-1  ] ERROR 2023/03/11 10:43:52 charge meter: not available
Mar 11 10:43:52 iwattcontroller evcc[3395]: [site  ] DEBUG 2023/03/11 10:43:52 grid power: 537W
Mar 11 10:43:52 iwattcontroller evcc[3395]: [site  ] DEBUG 2023/03/11 10:43:52 grid currents: [1.18 0.622 0.411]A
Mar 11 10:43:52 iwattcontroller evcc[3395]: [site  ] DEBUG 2023/03/11 10:43:52 pv power: 0W
Mar 11 10:43:52 iwattcontroller evcc[3395]: [site  ] DEBUG 2023/03/11 10:43:52 site power: 537W
Mar 11 10:43:52 iwattcontroller evcc[3395]: [lp-1  ] ERROR 2023/03/11 10:43:52 meter energy: not available
Mar 11 10:43:52 iwattcontroller evcc[3395]: [lp-1  ] DEBUG 2023/03/11 10:43:52 charger status: B
Mar 11 10:43:54 iwattcontroller evcc[3395]: [site  ] DEBUG 2023/03/11 10:43:54 ----
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.114.1

andig, 2023-03-11:
@thomas-hutterer-tik das ist der komplette Austausch? Ich vermisse eine `GetConfiguration` Anfrage an den Charger?

knese, 2023-03-12:
Ein `GetConfiguration` kommt da bei mir auch nicht beim Start...
```
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/03/12 14:12:37 evcc 0.114.1
[main  ] INFO 2023/03/12 14:12:37 using config file: /config/evcc.yaml
[main  ] INFO 2023/03/12 14:12:37 starting ui and api at :7070
[db    ] INFO 2023/03/12 14:12:47 using sqlite database: /data/evcc.db
[go-e  ] DEBUG 2023/03/12 14:12:48 waiting for chargepoint: 1m0s
[ocpp  ] DEBUG 2023/03/12 14:12:50 chargepoint connected: go-e
[ocpp  ] TRACE 2023/03/12 14:12:50 enqueued CALL [2842196148, TriggerMessage] for go-e
[go-e  ] DEBUG 2023/03/12 14:12:50 enabling meter watchdog
[go-e  ] DEBUG 2023/03/12 14:12:50 waiting for chargepoint status: 30s
[ocpp  ] TRACE 2023/03/12 14:12:50 dispatched request 2842196148 for go-e
[ocpp  ] TRACE 2023/03/12 14:12:50 sent JSON message to go-e: [2,"2842196148","TriggerMessage",{"requestedMessage":"MeterValues"}]
[ocpp  ] TRACE 2023/03/12 14:12:50 started timeout timer for go-e
[ocpp  ] TRACE 2023/03/12 14:12:51 received JSON message from go-e: [2,"531531563","BootNotification",{"chargePointVendor":"go-e","chargePointModel":"go-eCharger_V4","chargePointSerialNumber":"204296","firmwareVersion":"055.5"}]
[ocpp  ] TRACE 2023/03/12 14:12:51 handling incoming CALL [531531563, BootNotification] from go-e
[ocpp  ] TRACE 2023/03/12 14:12:51 sent CALL RESULT [531531563] for go-e
[ocpp  ] TRACE 2023/03/12 14:12:51 sent JSON message to go-e: [3,"531531563",{"currentTime":"2023-03-12T13:12:51Z","interval":60,"status":"Accepted"}]
[ocpp  ] TRACE 2023/03/12 14:12:51 received JSON message from go-e: [2,"531531564","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2023-03-12T13:12:51.000Z"}]
[ocpp  ] TRACE 2023/03/12 14:12:51 handling incoming CALL [531531564, StatusNotification] from go-e
[ocpp  ] TRACE 2023/03/12 14:12:51 sent CALL RESULT [531531564] for go-e
[ocpp  ] TRACE 2023/03/12 14:12:51 sent JSON message to go-e: [3,"531531564",{}]
[ocpp  ] TRACE 2023/03/12 14:12:51 received JSON message from go-e: [2,"531531565","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Charging","timestamp":"2023-03-12T13:12:51.000Z"}]
[ocpp  ] TRACE 2023/03/12 14:12:51 handling incoming CALL [531531565, StatusNotification] from go-e
[ocpp  ] TRACE 2023/03/12 14:12:51 sent CALL RESULT [531531565] for go-e
[ocpp  ] TRACE 2023/03/12 14:12:51 sent JSON message to go-e: [3,"531531565",{}]
[site  ] INFO 2023/03/12 14:12:52 site config:
[site  ] INFO 2023/03/12 14:12:52   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/03/12 14:12:52     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/03/12 14:12:52     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/03/12 14:12:52   vehicles:
[site  ] INFO 2023/03/12 14:12:52     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/03/12 14:12:52 loadpoint 1:
[lp-1  ] INFO 2023/03/12 14:12:52   mode:        off
[lp-1  ] INFO 2023/03/12 14:12:52   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/03/12 14:12:52   meters:      charge ✓
[lp-1  ] INFO 2023/03/12 14:12:52     charge:    power ✓ energy ✓ currents ✗
[lp-1  ] DEBUG 2023/03/12 14:12:52 phase timer inactive
[lp-1  ] DEBUG 2023/03/12 14:12:52 pv timer inactive
[site  ] DEBUG 2023/03/12 14:12:52 ----
[lp-1  ] ERROR 2023/03/12 14:12:52 charge meter: not available
[site  ] DEBUG 2023/03/12 14:12:52 pv power: 3532W
[site  ] DEBUG 2023/03/12 14:12:53 grid power: 5783W
[site  ] DEBUG 2023/03/12 14:12:53 site power: 5783W
[lp-1  ] ERROR 2023/03/12 14:12:53 meter energy: not available
[lp-1  ] DEBUG 2023/03/12 14:12:53 charger status: C
[lp-1  ] ERROR 2023/03/12 14:12:53 charge meter: not available
[lp-1  ] INFO 2023/03/12 14:12:53 car connected
[lp-1  ] DEBUG 2023/03/12 14:12:53 vehicle api refresh
[lp-1  ] DEBUG 2023/03/12 14:12:53 pv timer elapse
[lp-1  ] DEBUG 2023/03/12 14:12:53 pv timer inactive
[lp-1  ] ERROR 2023/03/12 14:12:53 meter energy: not available
[lp-1  ] ERROR 2023/03/12 14:12:53 charge meter: not available
[lp-1  ] INFO 2023/03/12 14:12:53 start charging ->
[main  ] DEBUG 2023/03/12 14:12:53 vehicle status: C (Monas Schneeball)
[lp-1  ] INFO 2023/03/12 14:12:53 vehicle updated: unknown -> Monas Schneeball
[lp-1  ] DEBUG 2023/03/12 14:12:53 set charge mode: pv
[lp-1  ] DEBUG 2023/03/12 14:12:53 set target soc: 95
[lp-1  ] DEBUG 2023/03/12 14:12:53 vehicle status: C
[lp-1  ] DEBUG 2023/03/12 14:12:53 vehicle soc: 100%
[lp-1  ] DEBUG 2023/03/12 14:12:53 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/03/12 14:12:53 vehicle range: 432km
[lp-1  ] WARN 2023/03/12 14:12:53 charger logic error: disabled but charging
[ocpp  ] TRACE 2023/03/12 14:12:53 enqueued CALL [3236358310, RemoteStopTransaction] for go-e
[ocpp  ] TRACE 2023/03/12 14:13:20 timeout for client go-e, canceling message
[ocpp  ] TRACE 2023/03/12 14:13:20 completed request 2842196148 for go-e
[ocpp  ] TRACE 2023/03/12 14:13:20 request 2842196148 for go-e timed out
[ocpp  ] TRACE 2023/03/12 14:13:20 go-e ready to transmit again
[... 37 lines cut ...]
  - name: go-eCharger
    #type: template
    #template: go-e-v3
    #host: 192.168.178.23
    type: ocpp
    idtag: 1
    stationid: go-e
    quirks: true
    metervalues: Energy.Active.Import.Register,Power.Active.Import
site:
  - title: Daheim # display name for UI
    meters:
      grid: my_grid # grid meter reference
      pvs:
        - my_pv # pv meter reference
vehicles:
  - name: Schneeball
    type: template
    template: tesla
...
loadpoints:
- title: Parkplatz # display name for UI
  charger: go-eCharger # Wallbox Referenz
  phases: 1
```

andig, 2023-03-12:
@knese natürlich nicht, Du hast ja auch `quirks: true` womit die Configabfrage unterbunden wird. Das ist hier aber nicht der Fall.

knese, 2023-03-12:
Jo, hab ich auch grade noch gesehen: Hier ohne quirks:
```
[ocpp  ] DEBUG 2023/03/12 15:01:20 chargepoint connected: go-e
[ocpp  ] TRACE 2023/03/12 15:01:20 enqueued CALL [1241926357, GetConfiguration] for go-e
[ocpp  ] TRACE 2023/03/12 15:01:20 dispatched request 1241926357 for go-e
[ocpp  ] TRACE 2023/03/12 15:01:20 sent JSON message to go-e: [2,"1241926357","GetConfiguration",{}]
[ocpp  ] TRACE 2023/03/12 15:01:20 started timeout timer for go-e
[ocpp  ] TRACE 2023/03/12 15:01:20 received JSON message from go-e: [3,"1241926357",{"configurationKey":[{"key":"AllowOfflineTxForUnknownId","readonly":false,"value":"0"},{"key":"GetConfigurationMaxKeys","readonly":true,"value":"99"},{"key":"ClockAlignedDataInterval","readonly":false,"value":"0"},{"key":"LocalAuthorizeOffline","readonly":false,"value":"0"},{"key":"HeartbeatInterval","readonly":false,"value":"60"},{"key":"LightIntensity","readonly":false,"value":"46"},{"key":"MeterValuesSampledData","readonly":true,"value":"Current.Import,Current.Offered,Energy.Active.Import.Register,Frequency,Power.Active.Import,Power.Factor,Power.Offered,Temperature,Voltage"},{"key":"MeterValuesSampledDataMaxLength","readonly":true,"value":"10"},{"key":"MeterValueSampleInterval","readonly":false,"value":"0"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"ConnectorPhaseRotation","readonly":true,"value":"0.RST,1.RST"},{"key":"ConnectorPhaseRotationMaxLength","readonly":true,"value":"3"},{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,SmartCharging,RemoteTrigger"},{"key":"SupportedFeatureProfilesMaxLength","readonly":true,"value":"6"},{"key":"WebSocketPingInterval","readonly":false,"value":"10"},{"key":"LocalAuthListEnabled","readonly":false,"value":"0"},{"key":"LocalAuthListMaxLength","readonly":true,"value":"10"},{"key":"SendLocalListMaxLength","readonly":true,"value":"10"},{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"20"},{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current"},{"key":"ChargingScheduleMaxPeriods","readonly":true,"value":"20"},{"key":"ConnectorSwitch3to1PhaseSupported","readonly":true,"value":"true"},{"key":"MaxChargingProfilesInstalled","readonly":true,"value":"20"}]}]
[ocpp  ] TRACE 2023/03/12 15:01:20 handling incoming CALL RESULT [1241926357] from go-e
[ocpp  ] TRACE 2023/03/12 15:01:20 completed request 1241926357 for go-e
[ocpp  ] TRACE 2023/03/12 15:01:20 enqueued CALL [1821484942, ChangeConfiguration] for go-e
[ocpp  ] TRACE 2023/03/12 15:01:20 go-e ready to transmit again
[ocpp  ] TRACE 2023/03/12 15:01:20 dispatched request 1821484942 for go-e
[ocpp  ] TRACE 2023/03/12 15:01:20 sent JSON message to go-e: [2,"1821484942","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register,Power.Active.Import"}]
[ocpp  ] TRACE 2023/03/12 15:01:20 started timeout timer for go-e
[ocpp  ] TRACE 2023/03/12 15:01:20 timeout canceled for go-e
[ocpp  ] TRACE 2023/03/12 15:01:20 received JSON message from go-e: [2,"531531622","BootNotification",{"chargePointVendor":"go-e","chargePointModel":"go-eCharger_V4","chargePointSerialNumber":"204296","firmwareVersion":"055.5"}]
[ocpp  ] TRACE 2023/03/12 15:01:20 handling incoming CALL [531531622, BootNotification] from go-e
[ocpp  ] TRACE 2023/03/12 15:01:20 sent CALL RESULT [531531622] for go-e
[ocpp  ] TRACE 2023/03/12 15:01:20 sent JSON message to go-e: [3,"531531622",{"currentTime":"2023-03-12T14:01:20Z","interval":60,"status":"Accepted"}]
[ocpp  ] TRACE 2023/03/12 15:01:20 received JSON message from go-e: [3,"1821484942",{"status":"NotSupported"}]
[ocpp  ] TRACE 2023/03/12 15:01:20 handling incoming CALL RESULT [1821484942] from go-e
[ocpp  ] TRACE 2023/03/12 15:01:20 completed request 1821484942 for go-e
[ocpp  ] TRACE 2023/03/12 15:01:20 go-e ready to transmit again
[ocpp  ] TRACE 2023/03/12 15:01:20 timeout canceled for go-e
[main  ] FATAL 2023/03/12 15:01:20 cannot create charger 'go-eCharger': cannot create charger 'ocpp': ChangeConfiguration failed: NotSupported
```

andig, 2023-03-12:
@knese danke, Fehler gefunden. Go-E schickt Messwerte, das ist aber read-only. Anderes Problem, Fix folgt: https://github.com/evcc-io/evcc/issues/6807

thomas-hutterer-tik, 2023-03-12:
Ich werde noch einen Testlauf machen, und das ganze Log hochladen. 
Ich verstehe den quirks: true Parameter nicht, der wird bei mir nicht akzeptiert, als ungültige Konfig ausgewiesen.

andig, 2023-03-12:
Lass ihn doch einfach mal weg wenn es keinen Grund gibt den zu verwenden. Wir stehen hier ja noch an der Stelle wo nichtmal die OCPP Config des Chargers bekannt ist.

stumbaumr, 2023-03-12:
Wenn es der Go-e Charger mit OCPP und es nur einer ist, dann fang mal ganz einfach so damit an:
```
chargers:
- name: "OCPP-1"
  type: ocpp
```
Denn `type: ocpp` und `type: template` mit `template: ocpp` sind zwei unterschiedliche Konfigurationen...

andig, 2023-03-13:
V.a., siehe auch https://github.com/evcc-io/evcc/discussions/3806#discussioncomment-5292105, erstmal ohne das ganze Voodoo probieren. Insbesondere `meter` gibts schon lange nicht und das tut gar nichts.

Was hier in die Suppe spucken könnte ist das

> ConnectorSwitch3to1PhaseSupported (r/o): true

In dem Fall senden wir die Phasenumschaltung mit was laut https://github.com/evcc-io/evcc/discussions/3806#discussioncomment-5253836 problematisch sein kann. Das könnten wir bei Bedarf aber auch unterdrücken- 1p3p geht dann aber nicht.

Jetzt warten wir aber erstmal ab was @thomas-hutterer-tik an Logfile produziert.

andig, 2023-03-18:
@thomas-hutterer-tik im aktuellen nightly gibts diverse Fixes. Bitte neues Issue mit Logfiel falls Du danach immer noch Probleme hast.

closed 2023-03-18 by andig

thomas-hutterer-tik, 2023-03-19:
Das Problem "meter energy: not available" besteht immer noch.
Log, Version und evcc.yaml weiter unten:

[main  ] INFO 2023/03/19 18:51:03 evcc 0.114.1
[main  ] INFO 2023/03/19 18:51:03 using config file: /Users/thomas/Downloads/evcc.yaml
[main  ] INFO 2023/03/19 18:51:03 starting ui and api at :7070
[db    ] INFO 2023/03/19 18:51:03 using sqlite database: /Users/thomas/.evcc/evcc.db
[ocpp  ] DEBUG 2023/03/19 18:51:04 waiting for chargepoint: 10m0s
[ocpp  ] INFO 2023/03/19 18:51:08 chargepoint connected, registering: go-eCharger_205643
[ocpp  ] TRACE 2023/03/19 18:51:08 enqueued CALL [2806726879, GetConfiguration] for go-eCharger_205643
[ocpp  ] TRACE 2023/03/19 18:51:08 dispatched request 2806726879 for go-eCharger_205643
[ocpp  ] TRACE 2023/03/19 18:51:08 sent JSON message to go-eCharger_205643: [2,"2806726879","GetConfiguration",{}]
[ocpp  ] TRACE 2023/03/19 18:51:08 started timeout timer for go-eCharger_205643
[ocpp  ] TRACE 2023/03/19 18:51:08 received JSON message from go-eCharger_205643: [3,"2806726879",{"configurationKey":[{"key":"AllowOfflineTxForUnknownId","readonly":false,"value":"0"},{"key":"GetConfigurationMaxKeys","readonly":true,"value":"99"},{"key":"ClockAlignedDataInterval","readonly":false,"value":"0"},{"key":"LocalAuthorizeOffline","readonly":false,"value":"0"},{"key":"HeartbeatInterval","readonly":false,"value":"60"},{"key":"LightIntensity","readonly":false,"value":"100"},{"key":"MeterValuesSampledData","readonly":true,"value":"Current.Import,Current.Offered,Energy.Active.Import.Register,Frequency,Power.Active.Import,Power.Factor,Power.Offered,Temperature,Voltage"},{"key":"MeterValuesSampledDataMaxLength","readonly":true,"value":"10"},{"key":"MeterValueSampleInterval","readonly":false,"value":"0"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"ConnectorPhaseRotation","readonly":true,"value":"0.RST,1.RST"},{"key":"ConnectorPhaseRotationMaxLength","readonly":true,"value":"3"},{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,SmartCharging,RemoteTrigger"},{"key":"SupportedFeatureProfilesMaxLength","readonly":true,"value":"6"},{"key":"WebSocketPingInterval","readonly":false,"value":"10"},{"key":"LocalAuthListEnabled","readonly":false,"value":"0"},{"key":"LocalAuthListMaxLength","readonly":true,"value":"10"},{"key":"SendLocalListMaxLength","readonly":true,"value":"10"},{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"20"},{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current"},{"key":"ChargingScheduleMaxPeriods","readonly":true,"value":"20"},{"key":"ConnectorSwitch3to1PhaseSupported","readonly":true,"value":"true"},{"key":"MaxChargingProfilesInstalled","readonly":true,"value":"20"}]}]
[ocpp  ] TRACE 2023/03/19 18:51:08 handling incoming CALL RESULT [2806726879] from go-eCharger_205643
[ocpp  ] TRACE 2023/03/19 18:51:08 completed request 2806726879 for go-eCharger_205643
[ocpp  ] TRACE 2023/03/19 18:51:08 AllowOfflineTxForUnknownId (r/w): 0
[ocpp  ] TRACE 2023/03/19 18:51:08 ChargeProfileMaxStackLevel (r/o): 20
[ocpp  ] TRACE 2023/03/19 18:51:08 ChargingScheduleAllowedChargingRateUnit (r/o): Current
[ocpp  ] TRACE 2023/03/19 18:51:08 ChargingScheduleMaxPeriods (r/o): 20
[ocpp  ] TRACE 2023/03/19 18:51:08 ClockAlignedDataInterval (r/w): 0
[ocpp  ] TRACE 2023/03/19 18:51:08 ConnectorPhaseRotation (r/o): 0.RST,1.RST
[ocpp  ] TRACE 2023/03/19 18:51:08 go-eCharger_205643 ready to transmit again
[ocpp  ] TRACE 2023/03/19 18:51:08 ConnectorPhaseRotationMaxLength (r/o): 3
[ocpp  ] TRACE 2023/03/19 18:51:08 timeout canceled for go-eCharger_205643
[ocpp  ] TRACE 2023/03/19 18:51:08 ConnectorSwitch3to1PhaseSupported (r/o): true
[ocpp  ] TRACE 2023/03/19 18:51:08 GetConfigurationMaxKeys (r/o): 99
[ocpp  ] TRACE 2023/03/19 18:51:08 HeartbeatInterval (r/w): 60
[ocpp  ] TRACE 2023/03/19 18:51:08 LightIntensity (r/w): 100
[ocpp  ] TRACE 2023/03/19 18:51:08 LocalAuthListEnabled (r/w): 0
[ocpp  ] TRACE 2023/03/19 18:51:08 LocalAuthListMaxLength (r/o): 10
[ocpp  ] TRACE 2023/03/19 18:51:08 LocalAuthorizeOffline (r/w): 0
[ocpp  ] TRACE 2023/03/19 18:51:08 MaxChargingProfilesInstalled (r/o): 20
[ocpp  ] TRACE 2023/03/19 18:51:08 MeterValueSampleInterval (r/w): 0
[ocpp  ] TRACE 2023/03/19 18:51:08 MeterValuesSampledData (r/o): Current.Import,Current.Offered,Energy.Active.Import.Register,Frequency,Power.Active.Import,Power.Factor,Power.Offered,Temperature,Voltage
[ocpp  ] TRACE 2023/03/19 18:51:08 MeterValuesSampledDataMaxLength (r/o): 10
[ocpp  ] TRACE 2023/03/19 18:51:08 NumberOfConnectors (r/o): 1
[ocpp  ] TRACE 2023/03/19 18:51:08 SendLocalListMaxLength (r/o): 10
[ocpp  ] TRACE 2023/03/19 18:51:08 SupportedFeatureProfiles (r/o): Core,FirmwareManagement,SmartCharging,RemoteTrigger
[ocpp  ] TRACE 2023/03/19 18:51:08 SupportedFeatureProfilesMaxLength (r/o): 6
[ocpp  ] TRACE 2023/03/19 18:51:08 WebSocketPingInterval (r/w): 10
[ocpp  ] TRACE 2023/03/19 18:51:08 enqueued CALL [909135102, TriggerMessage] for go-eCharger_205643
[ocpp  ] DEBUG 2023/03/19 18:51:08 waiting for chargepoint status: 30s
[ocpp  ] TRACE 2023/03/19 18:51:08 dispatched request 909135102 for go-eCharger_205643
[ocpp  ] TRACE 2023/03/19 18:51:08 sent JSON message to go-eCharger_205643: [2,"909135102","TriggerMessage",{"requestedMessage":"MeterValues"}]
[ocpp  ] TRACE 2023/03/19 18:51:08 started timeout timer for go-eCharger_205643
[... cut ...]
[lp-1  ] DEBUG 2023/03/19 18:51:08 phase timer inactive
[lp-1  ] DEBUG 2023/03/19 18:51:08 pv timer inactive
[... 50 lines cut ...]
interval: "2s"
sponsortoken: "..."
log: "debug"
levels:
  ocpp: trace
meters:
- name: "Fronius Simulator"
  type: "template"
  template: "fronius-solarapi-v1"
  usage: "pv"
  host: "wa-web-back.watt-analytics.com/api"
chargers:
- name: "OCPP-1"
  type: "ocpp"
  timeout: 10m
site:
  title: "Wien"
  meters:
    pvs:
    - "Fronius Simulator"
loadpoints:
- title: "OCPP-1"
  charger: "OCPP-1"
  phases: 0
  mode: "minpv"

andig, 2023-03-19:
@thomas-hutterer-tik sieht nicht nach nightly aus?

thomas-hutterer-tik, 2023-03-20:
Hi Andreas,

wie installiere ich den nightly build?

Danke Thomas

ChristophWatt, 2023-03-22:
Das "rejected" problem scheint es beim latest nightly build noch immer zu geben:

pi@iwattcontroller:~ $ journalctl -f -u evcc.service
-- Journal begins at Thu 2022-12-22 17:26:25 CET. --
Mar 22 17:47:45 iwattcontroller evcc[1337]: [site  ] DEBUG 2023/03/22 17:47:45 grid power: 0W
Mar 22 17:47:45 iwattcontroller evcc[1337]: [mqtt  ] DEBUG 2023/03/22 17:47:45 wa-pi/grid/p1 wait for initial value
Mar 22 17:47:49 iwattcontroller systemd[1]: Stopping evcc...
Mar 22 17:47:49 iwattcontroller systemd[1]: evcc.service: Succeeded.
Mar 22 17:47:50 iwattcontroller systemd[1]: Stopped evcc.
Mar 22 17:47:50 iwattcontroller systemd[1]: evcc.service: Consumed 5.896s CPU time.
Mar 22 17:47:50 iwattcontroller systemd[1]: Started evcc.
Mar 22 17:47:51 iwattcontroller evcc[1394]: [main  ] INFO 2023/03/22 17:47:51 evcc 0.114.1 (09a05d00)
Mar 22 17:47:51 iwattcontroller evcc[1394]: [main  ] INFO 2023/03/22 17:47:51 using config file: /etc/evcc.yaml
Mar 22 17:47:51 iwattcontroller evcc[1394]: [main  ] INFO 2023/03/22 17:47:51 starting ui and api at :7070
Mar 22 17:47:53 iwattcontroller evcc[1394]: [db    ] INFO 2023/03/22 17:47:53 using sqlite database: /var/lib/evcc/evcc.db
Mar 22 17:47:53 iwattcontroller evcc[1394]: [mqtt  ] INFO 2023/03/22 17:47:53 connecting evcc-584094917 at tcp://192.168.188.43:1883
Mar 22 17:47:53 iwattcontroller evcc[1394]: [mqtt  ] DEBUG 2023/03/22 17:47:53 tcp://192.168.188.43:1883 connected
Mar 22 17:47:53 iwattcontroller evcc[1394]: [mqtt  ] DEBUG 2023/03/22 17:47:53 tcp://192.168.188.43:1883 subscribe wa-pi/grid
Mar 22 17:47:54 iwattcontroller evcc[1394]: [ocpp  ] DEBUG 2023/03/22 17:47:54 waiting for chargepoint: 2m0s
Mar 22 17:48:13 iwattcontroller evcc[1394]: [ocpp  ] INFO 2023/03/22 17:48:13 chargepoint connected, registering: go-eCharger_205643
Mar 22 17:48:13 iwattcontroller evcc[1394]: [ocpp  ] TRACE 2023/03/22 17:48:13 enqueued CALL [2763784006, GetConfiguration] for go-eCharger_205643
Mar 22 17:48:13 iwattcontroller evcc[1394]: [ocpp  ] TRACE 2023/03/22 17:48:13 dispatched request 2763784006 for go-eCharger_205643
Mar 22 17:48:13 iwattcontroller evcc[1394]: [ocpp  ] TRACE 2023/03/22 17:48:13 sent JSON message to go-eCharger_205643: [2,"2763784006","GetConfiguration",{}]
Mar 22 17:48:13 iwattcontroller evcc[1394]: [ocpp  ] TRACE 2023/03/22 17:48:13 started timeout timer for go-eCharger_205643
Mar 22 17:48:13 iwattcontroller evcc[1394]: [ocpp  ] TRACE 2023/03/22 17:48:13 received JSON message from go-eCharger_205643: [3,"2763784006",{"configurationKey":[{"key":"AllowOfflineTxForUnknownId","readonly":false,"value":"0"},{"key":"GetConfigurationMaxKeys","readonly":true,"value":"99"},{"key":"ClockAlignedDataInterval","readonly":false,"value":"0"},{"key":"LocalAuthorizeOffline","readonly":false,"value":"0"},{"key":"HeartbeatInterval","readonly":false,"value":"60"},{"key":"LightIntensity","readonly":false,"value":"100"},{"key":"MeterValuesSampledData","readonly":true,"value":"Current.Import,Current.Offered,Energy.Active.Import.Register,Frequency,Power.Active.Import,Power.Factor,Power.Offered,Temperature,Voltage"},{"key":"MeterValuesSampledDataMaxLength","readonly":true,"value":"10"},{"key":"MeterValueSampleInterval","readonly":false,"value":"0"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"ConnectorPhaseRotation","readonly":true,"value":"0.RST,1.RST"},{"key":"ConnectorPhaseRotationMaxLength","readonly":true,"value":"3"},{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,SmartCharging,RemoteTrigger"},{"key":"SupportedFeatureProfilesMaxLength","readonly":true,"value":"6"},{"key":"WebSocketPingInterval","readonly":false,"value":"10"},{"key":"LocalAuthListEnabled","readonly":false,"value":"0"},{"key":"LocalAuthListMaxLength","readonly":true,"value":"10"},{"key":"SendLocalListMaxLength","readonly":true,"value":"10"},{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"20"},{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current"},{"key":"ChargingScheduleMaxPeriods","readonly":true,"value":"20"},{"key":"ConnectorSwitch3to1PhaseSupported","readonly":true,"value":"true"},{"key":"MaxChargingProfilesInstalled","readonly":true,"value":"20"}]}]
Mar 22 17:48:13 iwattcontroller evcc[1394]: [ocpp  ] TRACE 2023/03/22 17:48:13 handling incoming CALL RESULT [2763784006] from go-eCharger_205643
Mar 22 17:48:13 iwattcontroller evcc[1394]: [ocpp  ] TRACE 2023/03/22 17:48:13 completed request 2763784006 for go-eCharger_205643
Mar 22 17:48:13 iwattcontroller evcc[1394]: [ocpp  ] TRACE 2023/03/22 17:48:13 AllowOfflineTxForUnknownId (r/w): 0
Mar 22 17:48:13 iwattcontroller evcc[1394]: [ocpp  ] TRACE 2023/03/22 17:48:13 ChargeProfileMaxStackLevel (r/o): 20
Mar 22 17:48:13 iwattcontroller evcc[1394]: [ocpp  ] TRACE 2023/03/22 17:48:13 ChargingScheduleAllowedChargingRateUnit (r/o): Current
Mar 22 17:48:13 iwattcontroller evcc[1394]: [ocpp  ] TRACE 2023/03/22 17:48:13 ChargingScheduleMaxPeriods (r/o): 20
Mar 22 17:48:13 iwattcontroller evcc[1394]: [ocpp  ] TRACE 2023/03/22 17:48:13 ClockAlignedDataInterval (r/w): 0
Mar 22 17:48:13 iwattcontroller evcc[1394]: [ocpp  ] TRACE 2023/03/22 17:48:13 ConnectorPhaseRotation (r/o): 0.RST,1.RST
Mar 22 17:48:13 iwattcontroller evcc[1394]: [ocpp  ] TRACE 2023/03/22 17:48:13 ConnectorPhaseRotationMaxLength (r/o): 3
Mar 22 17:48:13 iwattcontroller evcc[1394]: [ocpp  ] TRACE 2023/03/22 17:48:13 ConnectorSwitch3to1PhaseSupported (r/o): true
[... cut ...]
Mar 22 17:48:36 iwattcontroller evcc[1394]: [mqtt  ] DEBUG 2023/03/22 17:48:36 wa-pi/grid/p1 wait for initial value
Mar 22 17:48:46 iwattcontroller evcc[1394]: [site  ] DEBUG 2023/03/22 17:48:46 ----
Mar 22 17:48:46 iwattcontroller evcc[1394]: [lp-1  ] ERROR 2023/03/22 17:48:46 charge meter: not available
Mar 22 17:48:46 iwattcontroller evcc[1394]: [site  ] DEBUG 2023/03/22 17:48:46 grid power: 0W
Mar 22 17:48:46 iwattcontroller evcc[1394]: [mqtt  ] DEBUG 2023/03/22 17:48:46 wa-pi/grid/p1 wait for initial value
^C
pi@iwattcontroller:~ $ evcc --version
evcc version 0.114.1 (09a05d00)

stumbaumr, 2023-03-27:
Hat er gemacht:
https://github.com/goecharger/go-eCharger-API-v2/issues/126
Mal sehen, was passiert...

thomas-hutterer-tik, 2023-04-13:
The described problem is solved on Go-e Charger side, tested positively after Go-e firmware upgrade, see attached log
[evcc go-e on:off 2023-04-05.log](https://github.com/evcc-io/evcc/files/11220213/evcc.go-e.on.off.2023-04-05.log)