evcc-io/evcc #6781: [UX/UX] Listing of all PV inverters separately as an option int the UI

panzerdev, 2023-03-11:
**Is your feature request related to a problem? Please describe.**
I have installed more then one PV inverters and configured also more then one inverter in the config.

**Describe the solution you'd like**
Right now there is only a Tooltip to show the power by inverter and I would suggest a listing of the inverters and their power when there is more then one. Could be an expanding view with a plus or arrow to operate.

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

andig, 2023-03-11:
Similar to batteries, would be consistent.

StevieC121176, 2023-03-12:
So wie in dem Vorschlag oben (rote Linie) wird es eng werden in der mobilen Ansicht auf Telefonen und Tablets (Hochformat). 
Eine Anzeige im Tooltip wäre in diesem Fall aber schon hilfreich. z.Bsp. vor der Wattzahl den Namen der Anlage aus der yaml.

RTTTC, 2023-03-12:
I would like this feature added too.

naltatis, 2023-03-14:
I like the idea of being able to expand details for the production and batterie entries if there are more than one. It's much nicer than the current tooltip solution because it could stay open permanently.

Currently we don't have display names for the individual PVs and batteries. This is something we would have to introduce. Yes we could go with PV1, PV2, PV3, but I guess most users would have more meaningful names "Carport, Main roof, ..." that they would want to see there.

panzerdev, 2023-03-14:
https://github.com/evcc-io/evcc/blob/14d2ba9bb1e93575b8c1d855dfa6e97dc9695bb2/api/api.go#L49-L51

needs a Title like in Vehicle
https://github.com/evcc-io/evcc/blob/14d2ba9bb1e93575b8c1d855dfa6e97dc9695bb2/api/api.go#L141

This needs a Title property
https://github.com/evcc-io/evcc/blob/14d2ba9bb1e93575b8c1d855dfa6e97dc9695bb2/core/site.go#L34-L36

This should be all on the backend side because the frontend gets the `meterMeasurement` slice published.

andig, 2023-03-14:
We won't add additional titles at this time. We can still display multiple PVs like batteries.

panzerdev, 2023-03-14:
> We won't add additional titles at this time. We can still display multiple PVs like batteries. Seems we shouldn't have started with displaying individual devices at all ;)

Why not? Everything should have meta data. Could even be a key value object containing labels or whatever the user would like to put in a meta information. Why should this be a big deal?

andig, 2023-03-14:
It is not particularly important compared to our current priorities.

naltatis, 2023-03-14:
What do you mean?

> We can still display multiple PVs like batteries.

From my point of view the visualization of multiple pvs and batteries is quite similar right now. Only difference is, that we show power instead of kWh/%, but the mechanism is the same: tooltip, one line per device.

[image]
[image]

RTTTC, 2023-03-14:
> Currently we don't have display names for the individual PVs and batteries. This is something we would have to introduce. Yes we could go with PV1, PV2, PV3, but I guess most users would have more meaningful names "Carport, Main roof, ..." that they would want to see there.

If we name them Carport, MainRoof etc, instead of PV1, PV2 in yaml, it would already be a solution, right?

andig, 2023-03-14:
> What do you mean?

Nvm. I thought we didn‘t have the tooltip yet.

panzerdev, 2023-03-16:
[image]

Maybe a a first interrim solution we could at least add a name/label in the popup?

I can do a PR no problem.

andig, 2023-03-19:
> I can do a PR no problem.

Ich möchte keine neue can of worms aufmachen.

panzerdev, 2023-03-20:
Das ganze Thema ist durch die aktuelle Meter Implementierung schon ein ganz schönes Brett, habe aufgegeben als ich zu den dekorierten meters kam. Habe die Idee verfolgt einen generischen key value store einzuführen und da waren die Änderungen minimal.

Habe keinen PR aufs evcc repo gemacht und erstmal nur in meinem fork verbaut. Wer interesse hat kann sich den code hier gern klauen für sich selber.

https://github.com/panzerdev/evcc/pull/2

Ansonsten steht das Thema ja zur Diskussion.

andig, 2023-03-20:
> Habe die Idee verfolgt einen generischen key value store einzuführen und da waren die Änderungen minimal.

Lass uns das doch neu anschauen, wenn wir mit Ui Config durch sind. Da wird sich an den internen Strukturen ohnehin noch einiges ändern (müssen). Je weniger Ballast wir dabei mitschleppen, desto besser. Danach sind die Karten wieder neu gemischt. Es muss ja kein "nein" für immer sein sondern ein "jetzt nicht". Danke fürs Verständnis!

andig, 2023-03-22:
Sollten wir- wenns soweit ist- auch für `aux` Zähler machen.

PeterPablo, 2023-11-22:
Just to add from #10335, which is mentioned here only implicitly: Please add `title` to the MQTT topics of `meters`.

andig, 2024-02-14:
Ich sehe drei technische Möglichkeiten:

1. Alle Zähler/Charger um weitere Felder erweitern, analog `embed` bei Fahrzeugen. Das sind viele Änderungen und eigentlich brauchen die Zähler die Daten nicht. Fahrzeuge übrigens auch nicht, aber es gibt auch welche die diese Daten im Api schon mitliefern. Also vielleicht doch.
2. Alle Zähler/Charger mittels Decorators wrappen um zusätzliche Fähigkeiten/Felder zu ergänzen. Logik schön separiert, aber viel Aufwand, die Wrapper bei jedem neuen Interface ajour zu halten.
3. Separate Metadaten- das hatte @panzerdev schonmal vorgeschlagen, wäre aber eine zweite Lösung fürs gleiche Problem. Hässlich.

Nichts davon überzeugt mich so, dass ich die Lösung umsetzen wollte :(

andig, 2024-02-21:
Das ist genau der Punkt. Konsistente Lösungen wären schön. Beim Fahrzeug ist es ein Zwitter da der Title auch vom Fahrzeug-API genommen werden kann. Dann müsste das Fahrzeug also die Metadaten kennen oder (inversion of control) die Metadaten selbst anbieten. Ich hab noch nichts gefunden was mich so richtig begeistert hätte.

andig, 2024-03-31:
> Der Metadatenansatz sollten wenn dann für alle Objekte sein und auch wieder mit rausgegeben werden für Anschlussverwendung innerhalb und außerhalb von evcc.

Damit stellt sich die Frage, welche Daten Metadaten und welche eigene Daten sind?

panzerdev, 2024-03-31:
> > Der Metadatenansatz sollten wenn dann für alle Objekte sein und auch wieder mit rausgegeben werden für Anschlussverwendung innerhalb und außerhalb von evcc.
> 
> Damit stellt sich die Frage, welche Daten Metadaten und welche eigene Daten sind?

evcc muss doch da nicht opinionated sein. Ein einfaches Key Value System mit strings as keys und string als value sollte doch jeden anwendungsfall abedecken. in nen string kann ja ne zahl rein, oder auch ein json object zb.

RobBboR, 2024-04-17:
Gerade wollte ich dieses Feature vorschlagen, als ich gesehen habe, dass es ja schon ein Issue dafür gibt.

Noch ein Aspekt, der bisher nicht erwähnt wurde:
Auf dem Smartphone gibt es keinen Tooltip, so dass man die Einzelwert dort nicht einsehen kann.

Bei meinen 2 Erzeugern (SolarEdge PV und OpenDTU Balkonkraftwerk) kennt evcc als einziges System alle interessanten Werte im Haus (zusammen mit SMA EV Charger). Es wäre super, wenn ich alle momentanen Werte an ein und derselben Stellen einsehen könnte, ohne zwischen verschiedenen Apps hin und her wechseln zu müssen.

naltatis, 2024-04-18:
> Auf dem Smartphone gibt es keinen Tooltip, so dass man die Einzelwert dort nicht einsehen kann.

 Auf Touch-Geräten solllte der Tooltip "on click" funktionieren.

Tombra1889, 2024-06-30:
> Das ganze Thema ist durch die aktuelle Meter Implementierung schon ein ganz schönes Brett, habe aufgegeben als ich zu den dekorierten meters kam. Habe die Idee verfolgt einen generischen key value store einzuführen und da waren die Änderungen minimal.
> 
> Habe keinen PR aufs evcc repo gemacht und erstmal nur in meinem fork verbaut. Wer interesse hat kann sich den code hier gern klauen für sich selber.
> 
> [panzerdev#2](https://github.com/panzerdev/evcc/pull/2)
> 
> Ansonsten steht das Thema ja zur Diskussion.
> 
> [image]

Ees wäre zudem sehr cool, wenn die Anzeige pro String wäre!

SolarPowerEV, 2025-03-02:
> > Auf dem Smartphone gibt es keinen Tooltip, so dass man die Einzelwert dort nicht einsehen kann.
> 
> Auf Touch-Geräten solllte der Tooltip "on click" funktionieren.

Tut er bei mir auch.

andig, 2025-03-04:
> Lass uns das doch neu anschauen, wenn wir mit Ui Config durch sind. 

Der Zeitpunkt ist jetzt. Eine mögliche (technische) Idee wäre, dass wir die Titel vielleicht mit in `config.Named` verwursten, also außerhalb der eigentlichen Meterimplementierungen aber innerhalb der jeweiligen Gerätekonfiguration. Wenn die Zeit reicht brauchts einen PoC dafür.

andig, 2025-03-07:
Kommentar in eigener Sache: https://github.com/evcc-io/evcc/pull/19416 ist ein schönes Beispiel, warum es sich lohnt Features erst dann zu bauen wenn sich ein guter und eleganter Weg der Integration findet anstatt sie auf Gewalt "reinzudübeln". Jetzt können wir damit gleich 3 Fliegen mit einer Klappe erschlagen und perspektivisch auch weitere Sonderlocken damit ablösen. Hätten wir das früher gemacht, wäre es mindestens doppelter Aufwand gewesen...

closed 2025-03-11 by andig

cschlipf, 2025-04-26:
Ist es eigentlich geplant die 'titles' auch in die Influx DB zu schreiben?

Für meine PV habe ich noch keine Titles festgelegt (Custom ModBus Meter kann man ja über das Config UI nicht anlegen), aber falls doch, würde dann hier ein Title-Tag stehen?

```
> select * from pvPower order by time desc limit 10
name: pvPower
time                 id value
----                 -- -----
2025-04-26T09:35:45Z 2  630
2025-04-26T09:35:45Z    2910
2025-04-26T09:35:45Z 1  2280
2025-04-26T09:35:15Z    3040
2025-04-26T09:35:15Z 2  690
2025-04-26T09:35:15Z 1  2350
2025-04-26T09:34:45Z 2  740
2025-04-26T09:34:45Z 1  2400
2025-04-26T09:34:45Z    3140
2025-04-26T09:34:15Z 2  780
```

Ist natürlich redundant zur ID, würde es beim Auslesen der InfluxDB aber deutlich einfacher machen, wenn man den bereits festgelegten Title direkt mit auslesen könnte.

na-ma-ste, 2025-06-11:
Hallo,

gibt es aktuell die Möglichkeit die PV-Anlagen eigenständig zu benennen? Ich bin bisher nicht fündig geworden und würde gern in der UI statt "PV-Anlage #1" gerne einen sprechenden Namen hinterlegen.

naltatis, 2025-06-11:
@na-ma-ste diese Option gibt es. Allerdings musst du dafür deine PV-Anlage über Config UI anlegen. Über die `evcc.yaml` ist das aus technischen Gründen nicht möglich.

cschlipf, 2025-06-11:
Hi @naltatis , 

Um nochmal kurz auf meine Frage zurückzukommen: macht es Sinn den Title neben der ID in die InfluxDB zu schreiben?

Vielen Dank!

andig, 2025-06-11:
Wäre schön als Tag. PR welcome.

cschlipf, 2025-06-12:
Vielen Dank.

Mit einem PR würde ich sehr gerne dienen. Leider bekomme ich das Build Environment nicht zum laufen und zudem wären es meine ersten Schritte in Go.

Aber ich kann mit einem Feature Request dienen: #21805