evcc-io/evcc #6787: Homematic: optional credentials

andig, 2023-03-11:
Servus zusammen,

bin wieder zurück bei EVCC und freu mich sehr über die viele neuen coolen Features. 
Gehört jetzt fest zum Haushalt als Synology Docker Installation.
Mit der neuen Version 0.114.1 bekomme ich bei folgender Konfiguration jetzt eine Fehlermeldung beim Hochfahren (Charger Fehler): 
->   Error: cannot create charger 'homematic': missing credentials

In der alten Version 0.114 funktionierte es noch. Hab allerdings noch eine alte Homematic CCU2, evtl. liegt es an der geänderten API bei HMIP.
......
- type: template
  template: homematic # 
  host: 192.168.xx.xx # 
  port: 2010 # Port # Optional HMiP=2010 BidCos-RF = 2001
  device: '000218A9xxxxxx' # 
  standbypower: -600 #
  user: #  Optional
  password: # Optional 
  name: cEnergySwitchWaterPump
  meterchannel: 6 # Keine Funktion, nur Schalt HMIP Schalter ! In v0.114 noch der ENERGY_COUNTER XML Fehler.
  switchchannel: 3
  integrateddevice: true # Klappt noch nicht, führt aber auch zu keinem Fehler
  icon: waterheater

besten Dank Christian

_Originally posted by @Cruiserlive in https://github.com/evcc-io/evcc/issues/6560#issuecomment-1464981794_

andig, 2023-03-11:
/cc @thierolm sind die Credentials hier evtl. optional?

thierolm, 2023-03-12:
Man kann "Automatisches Anmeldung" aktivieren, macht aber aus Sicherheitsgründen in der Regel keiner ... und ist auch nicht die "Default"-Einstellung. 

[image]

thierolm, 2023-03-12:
jepp

PR #6798 (2023-03-12): Homematic: Enable Auto Login
Fixes #6787
closed 2023-03-12 by andig