evcc-io/evcc #6790: SMA EV Charger – PV Charging mode defaults to Stop Charge

retrostrobe, 2023-03-12:
### Describe the bug

The Fast mode works as intended. However, when the physical switch is turned to PV Charging and as soon as either Optimised or Target mode is selected, evcc sets the mode to Stop Charge. 

Ideally, evcc would ignore the charger in PV Charging mode as this is when I want Home Manager 2.0 to handle the charging.

In [smaevcharger/types.go](https://github.com/evcc-io/evcc/blob/master/charger/smaevcharger/types.go) there are references to tell which mode the switch is set to with `SwitchOeko` or `SwitchFast` and hopefully it's a straightforward solution to implement. 

### Steps to reproduce

1. Turn physical switch from Fast to PV Charging mode on charger
2. Change to either Optimised or Target mode in SMA app (or charger's web UI)
3. evcc warns that charger is out of sync and sets the mode to Stop Charge

### Configuration details

```yaml
n/a
```

### Log details

```text
WARN 2023/03/11 22:41:21 charger out of sync: expected disabled, got enabled
```

### What type of operating system are you running?

Linux

### Version

0.114.1

premultiply, 2023-03-12:
Es kann immer nur ein steuerndes System geben.
Wenn du die wirklich die SMA-Lösung nutzen möchtest brauchst du kein evcc.
Und Schnellladen funktioniert sogar mit SMA halbwegs zuverlässig.

retrostrobe, 2023-03-12:
Thanks for your reply. Nonetheless, it would be great if evcc didn't interfere with PV Charging mode. The scenario is that I want to use evcc in evenings and use the planner as it's more reliable than SMA's target charge mode. Currently, I have to disable the evcc service as it constantly turns off PV Charging. Hope it makes sense.

premultiply, 2023-03-12:
Leave the Switch in Power position and let evcc control everything.
Also the pv excess charging is much more advanced compared to the lazy SHM control.

closed 2023-03-12 by premultiply