evcc-io/evcc #6804: Growatt SPH4600 Json-Datei in EVCC einbinden

Sommerwl, 2023-03-12:
### Describe the bug

Hallo,

ich habe einen neunen Wechselrichter,  Growatt SPH4600 mit einer Batterie.

Bislang hat alles super funktioniert (Solarwatt Energiemanager im Einsatz) mit der Überschussladung und unserem eTwingo.

Nun habe ich  einen zusätzlichen Growatt WR mit einem Speicher dran.
Um an die Daten aus dem Wechselrichter zu kommen , habe ich auf den Growatt ShineWIFI-X Stick die Software 
"firmware-ShineWifiX.bin" installiert.

Mir werden jetzt die WR Daten auf einer WEB-Oberfläche und in einer Json Datei angezeigt.

Ich möchte, wenn die Batterie geladen wird, dass der eTwingo nur über PV (wenn noch Überschuss vorhanden)  oder Min+PV (minimale Ladung , in dem Fall 1,4KW) geladen wird. Ich möchte nicht, dass der eTwingo über die Batterie geladen wird. 

Ich bin jetzt überfragt welche Werte (Patameter) ich aus der Json-Datei verwenden muss und wie...

Vielleicht kann mir jemand helfen.

Danke und viele Grüße

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Hier lese ich die Daten aus meinem Energiemanager und lasse sie Anzeigen:

meters:
- name: energyManager-EVU
  type: custom
  power:
    source: calc
    add:
    - source: http
      uri: http://192.168.178.109/rest/kiwigrid/wizard/devices
      jq: .result.items[] | select(.deviceModel[].deviceClass == "com.kiwigrid.devices.location.Location") .tagValues.PowerConsumedFromGrid.value
    - source: http
      uri: http://192.168.178.109/rest/kiwigrid/wizard/devices
      jq: .result.items[] | select(.deviceModel[].deviceClass == "com.kiwigrid.devices.location.Location") .tagValues.PowerOut.value
      scale: -1
- name: energyManager-PV
  type: custom
  power:
    source: http
    uri: http://192.168.178.109/rest/kiwigrid/wizard/devices
    jq: .result.items[] | select(.deviceModel[].deviceClass == "com.kiwigrid.devices.location.Location") .tagValues.PowerProduced.value

site:
  title: Mein ZuHause
  meters:
    grid: energyManager-EVU
    pv: energyManager-PV
```

### Log details

```text
Anbei die Json Datei, wird hier lokal abgerufen :http://192.168.178.24/status

{"InverterStatus":6,"InputPower":0,"PV1Voltage":0,"PV1InputCurrent":0,"PV1InputPower":0,"PV2Voltage":0,"PV2InputCurrent":0,"PV2InputPower":0,"OutputPower":0,"GridFrequency":50.01,"L1ThreePhaseGridVoltage":227.5,"L1ThreePhaseGridOutputCurrent":1.2,"L1ThreePhaseGridOutputPower":0,"L2ThreePhaseGridVoltage":0,"L2ThreePhaseGridOutputCurrent":0,"L2ThreePhaseGridOutputPower":0,"L3ThreePhaseGridVoltage":0,"L3ThreePhaseGridOutputCurrent":0,"L3ThreePhaseGridOutputPower":0,"TodayGenerateEnergy":3.4,"TotalGenerateEnergy":87,"TWorkTimeTotal":3647874,"PV1EnergyToday":0,"PV1EnergyTotal":0.4,"PV2EnergyToday":0,"PV2EnergyTotal":0,"PVEnergyTotal":0.4,"InverterTemperature":34.9,"TemperatureInsideIPM":32.8,"BoostTemperature":32.2,"DischargePower":0,"ChargePower":150,"BatteryVoltage":53.3,"SOC":94,"ACPowerToUser":0,"ACPowerToUserTotal":0,"ACPowerToGrid":0,"ACPowerToGridTotal":0,"INVPowerToLocalLoad":0,"INVPowerToLocalLoadTotal":0,"BatteryTemperature":2.6,"BatteryState":6,"EnergyToUserToday":0.2,"EnergyToUserTotal":350.6,"EnergyToGridToday":1.1,"EnergyToGridTotal":47.7,"DischargeEnergyToday":3.8,"DischargeEnergyTotal":93.3,"ChargeEnergyToday":4.5,"ChargeEnergyTotal":111,"LocalLoadEnergyToday":2.147483647e7,"LocalLoadEnergyTotal":285.6,"Mac":"40:91:51:47:35:89","Cnt":637}
```

### What type of operating system are you running?

Linux

premultiply, 2023-03-12:
Wozu der Umweg?
Ist doch fix und fertig implementiert via Modbus:
https://docs.evcc.io/docs/devices/meters#growatt-hybrid-inverter

Ansonsten kannst du das natürlich auch gerne per Custom-Meter, HTTP-Plugin und jq auseinanderpflücken.

closed 2023-03-12 by premultiply

Sommerwl, 2023-03-12:
Ok, da stehe ich gerade etwas auf dem Schlauch...sorry

Muss ich den Teil für den ShineWIFI-X Stick nutzen?
Werden dann auch alle Notwendigen Daten aus der Batterie und dem Wechselrichter gezogen?

  # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.0.2.2 # Hostname
    port: 502 # Port

Sommerwl, 2023-03-12:
> Wozu der Umweg? Ist doch fix und fertig implementiert via Modbus: https://docs.evcc.io/docs/devices/meters#growatt-hybrid-inverter
> 
> Ansonsten kannst du das natürlich auch gerne per Custom-Meter, HTTP-Plugin und jq auseinanderpflücken.

Ich komme bei der Installation nicht weiter, habe es jetzt über eine neue Config versucht....bin planlos welche Parameter ich dort erfassen muss, habe ja nur die IP

premultiply, 2023-03-12:
You need a serial RS485 connection to the inverter.

premultiply, 2023-03-12:
Ist das die Original-Firmware des Growatt-Sticks die dieses JSON erzeugt?

Sommerwl, 2023-03-12:
> Ist das die Original-Firmware des Growatt-Sticks die dieses JSON erzeugt?

Nein, ist eine Software die ich aufgespielt habe (siehe Link)

[https://github.com/otti/Growatt_ShineWiFi-S](url)

In der nähe vom WR ist nur WLAN, ich hatte gehofft ich kann mit dem angepassten Stick den WR in EVCC einbinden.

premultiply, 2023-03-12:
Na dann pflücke das doch als Custom meters per HTTP und jq auseinander.
Hast du doch beim Solarwatt auch gemacht.
(Das gibts aber auch schon fertig: https://docs.evcc.io/docs/devices/meters#energymanager-energymanager-pro)

**pv:**
power: InputPower
energy: PVEnergyTotal

**battery:**
power: DischargePower - ChargePower
energy: DischargeEnergyTotal
soc: SOC

Die relevanten grid Daten fehlen dort leider komplett.

Sommerwl, 2023-03-13:
Ich bekomme es nicht hin....

meters:
- name: GrowattBattEndladen
  type: custom
  power:
    source: http
    uri: http://192.168.178.24/status 
    jq: .tagValues.DischargePower.value

Kannst du mir bitte Hilfestellung geben.

premultiply, 2023-03-13:
https://jqplay.org/

Sommerwl, 2023-03-13:
Ok, das war ja wirklich nicht so schwer, danke 

Jetzt ist mir nur nicht ganz klar wie ich die Parameter richtig zuweisen muss
Über den Wechselrichter wird aktuell nur eine Batterie ge- und entladen

Ich würde jetzt follgende Parameter einbinden:

SOC (81%)  -> Füllstand der Batt in %
ChargePower (1080 W)-> aktuelle Ladeleistung der Batterie
DischargePower (0W) -> aktuelle Entladeleistung der Batterie
DischargeEnergyTotal -> (98.5kWh) Brauche ich den Wert?

Welchen Parametern muss ich die Werte in EVCC zuweisen?-> usage ?

Sind das alles meter?
Was davon muss unter "site" aufgeführt werden?

Sorry, die Batterie ist neu und ich habe in den Beschreibungen nichts gefunden.
Bin echt planlos.....

Von der Logik her müssten diese als Meter hinterlegt werden.
ChargePower 
DischargePower

SOC eigentlich nur in die Anzeige.....

premultiply, 2023-03-13:
Eher so?
```yaml
meters:
- name: jq-BAT
  type: custom
  power:
    source: http
    uri: http://192.168.178.24/status
    jq: .DischargePower-.ChargePower
  soc:
    source: http
    uri: http://192.168.178.24/status
    jq: .SOC
  energy:
    source: http
    uri: http://192.168.178.24/status
    jq: .DischargeEnergyTotal
- name: jq-PV
  type: custom
  power:
    source: http
    uri: http://192.168.178.24/status
    jq: .InputPower
  energy:
    source: http
    uri: http://192.168.178.24/status
    jq: .PVEnergyTotal
```