evcc-io/evcc #6809: Homematic switch break since firmware 3.67.10

Haegar-21, 2023-03-12:
### Discussed in https://github.com/evcc-io/evcc/discussions/6750

<div type='discussions-op-text'>

<sup>Originally posted by **Haegar-21** March 10, 2023</sup>
Hallo in die Runde,

ich möchte erst einmal allen Beteiligten für Ihre Arbeit und Zeit für dieses Projekt danken.

Noch habe ich keine Wallbox und wollte mit einem einfachen Stecker aus der Homematic IP Serie die ersten Schritte machen. Leider stoße ich da auf ein paar Problem. Lt. der evcc.io Seite denke ich, dass ich alles richtig gemacht habe. Aber das denke ich wohl nur :-)

Allerdings bekomme ich im Log immer folgende Fehler:

[image]

So sieht es aus, wenn ich die Seite lade:

[image]

Und hier noch einmal meine Konfiguration:

_# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

#unique installation id
plant: 837f94df31d19915f1e798bbda35498da17f97430c916cb962d63

interval: 10s # control cycle interval
#sponsortoken: xxxx

meters:
- name: Netz
  type: template
  template: sonnenbatterie 
  usage: grid  
  host: 192.168.178.***  
  port: 8080 

- name: PV
  type: template
  template: sonnenbatterie 
  usage: pv  
  host: 192.168.178.***  
  port: 8080  

- name: Batterie
  type: template
  template: sonnenbatterie 
  usage: battery  
  host: 192.168.178.***  
  port: 8080
  
- name: HMIP-Steckdose
  type: template
  template: homematic
  usage: charge
  host: 192.168.178.***
  device: '000***CE02'  
  user: ***  
  password: ***    
  
chargers:
- name: HMIP-Steckdose
  type: template
  template: homematic
  host: 192.168.178.124
  port: 2001
  device: '000***CE02'  
  user: ***  
  password: ***
  standbypower: 1

vehicles:
- name: Fahrrad
  type: template
  template: offline 
  title: E-Bike
  capacity: 0.75
  phases: 1  
  mode: off  
  minCurrent: 0.2
  maxCurrent: 0.2

loadpoints:  
- title: Werkstatt
  charger: HMIP-Steckdose
  vehicle: Fahrrad
  mode: off
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: Zuhause
  meters:
    grid: Netz
    pv:
    - PV
    battery:
    - Batterie
    
Zudem hätte ich erwartet, dass die Schaltsteckdose sich einschaltet, wenn ich auf "Schnell" stelle.

Hat da jemand eine Idee bzw. einen Tipp für mich?</div>

thierolm, 2023-03-12:
Ich glaube ich habe den Fehler gefunden. Ich muss noch eine API Initialisierung in die evcc Schnittstelle einbauen:
[image]
Nachdem ich manuell mittels curl die Methode `system.listMethod` an meine CCU geschickt hatte, kamen auch wieder Werte ...

thierolm, 2023-03-12:
Nach einem Firmware-Update und/oder reboot der CCU scheint er das zu vergessen ...

PR #6813 (2023-03-13): Homematic: Add initialize Methods
Fixes #6809

Relates to discussion #6750
closed 2023-03-13 by andig