evcc-io/evcc #6816: Heidelberg Energy Control Value Energy seit Installation

elektronikfreak, 2023-03-13:
Hallo, 

ist es evtl. möglich den Wert "Energy since installation" aus der Heidelberg Energy Control Wallbox auszulesen und per MQTT zur Verfügung zu stellen ? Dies hätte den Vorteil den gesamten Energieverbauch der Wallbox für die Statistik einfacher auszurechnen.
Das wäre dann auch Verbraucher unabhängig.

Es handelt sich hier um die Modbusadresse 17 und 18.

Danke
Gruß

andig, 2023-03-13:
Das könntest Du per Heimautomatisierung machen. Zusätzliche Werte per evcc abzuholen ist nicht geplant.

panzerdev, 2023-03-13:
@andig könnte man dafür den https://github.com/evcc-io/evcc/discussions/5094 Modbus Proxy nutzen um beim evcc die Werte extern anzufragen? Das würde es zumindest "zentraler" machen.

@elektronikfreak falls du noch nix laufen hast für solche Sachen kann ich dir node red empfehlen als einfache möglichkeit solche zeitabhängigen Sachen zu machen.

premultiply, 2023-03-13:
Wird bereits gelesen:
https://github.com/evcc-io/evcc/blob/3c6e2ca3d635a091e2d18be91c972643ef6597a7/charger/heidelberg-ec.go#L235

andig, 2023-03-13:
@premultiply wir geben die allerdings nirgendwo aus...

andig, 2023-03-13:
> @andig könnte man dafür den https://github.com/evcc-io/evcc/discussions/5094 Modbus Proxy nutzen um beim evcc die Werte extern anzufragen? Das würde es zumindest "zentraler" machen.

Gehen tut das. Notwendig ist es aber nur wenn das Zielgerät nur eine Verbindung zulässt. Ansonsten ist da nix "zentraler" dran.

panzerdev, 2023-03-13:
> > @andig könnte man dafür den #5094 Modbus Proxy nutzen um beim evcc die Werte extern anzufragen? Das würde es zumindest "zentraler" machen.
> 
> Gehen tut das. Notwendig ist es aber nur wenn das Zielgerät nur eine Verbindung zulässt. Ansonsten ist da nix "zentraler" dran.

Wenn das Zielgerät wie hier ein RS485 Gerät ist und wenn ein USB Stick am Rechner ist und vom evcc benutzt wird um die Abfragen zu machen, könnte es sich schon lohnen. Wenn eh schon ein Modbus TCP Gateway davor ist, dann ist es egal, das stimmt.

andig, 2023-03-13:
> Wenn das Zielgerät wie hier ein RS485 Gerät ist und wenn ein USB Stick am Rechner ist und vom evcc benutzt wird um die Abfragen zu machen, könnte es sich schon lohnen. 

Yep

> Wenn eh schon ein Modbus TCP Gateway davor ist, dann ist es egal, das stimmt.

Es sei denn, das Gerät unterstützt und 1 Verbindung, das gibts auch bei TCP...

VolkerK62, 2023-03-13:
>Falls ja- als was: chargeImportEnergy oder chargeTotalImportEnergy? Achtung- wir haben auch noch chargedEnergy -> das wäre dann chargeSessionEnergy?

Sind die Bezeichnungen nicht immer gleich? 
Ich hab keine Heidelberg, aber ich sehe in MQTT den "Zählerstand" meiner Wallbox als `chargeTotalImport`

andig, 2023-03-13:
Danke @VolkerK62, Du hast recht. Tatsächlich ist das schon umgesetzt :)

closed 2023-03-13 by andig