evcc-io/evcc #6817: Terra ABB + OCPP - meter energy: not available - charger: timeout - functionality broken

OnWire, 2023-03-13:
### Describe the bug

A while after startup of evcc, error messages appear and functionality is broken, only service restart helps.

### Steps to reproduce

1. car ist connected to charger
2. start evcc
3. wait a while until error messages
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  lp-1: debug
  cache: error

# unique installation id
plant: ****

interval: 15s # control cycle interval

meters:
- type: template
  template: sonnenbatterie
  usage: grid
  host: 192.168.100.*
  port: 8080
  name: grid1
- type: template
  template: sonnenbatterie
  usage: pv
  host: 192.168.100.*
  port: 8080
  name: pv1
- type: template
  template: sonnenbatterie
  usage: battery
  host: 192.168.100.*
  port: 8080
  name: battery1
  capacity: 11

vehicles:
- type: template
  template: seat
  title: Seat Leon
  user: ****
  password: ****
  vin: ****
  capacity: 12.8
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  icon: car
  cache: 15m
  mode: PV
  targetSoc: 100
  identifiers:
  - Leon
  name: ev1

[... 220 lines cut ...]
[ocpp  ] TRACE 2023/03/13 09:26:21 sent CALL RESULT [4885882] for TACW****
[ocpp  ] TRACE 2023/03/13 09:26:21 sent JSON message to TACW****: [3,"4885882",{}]
[ocpp  ] TRACE 2023/03/13 09:26:21 received JSON message from TACW****: [2, "8881372", "StatusNotification", {"connectorId": 1, "errorCode": "NoError", "info": "null", "status": "Preparing", "vendorErrorCode": "0x0000"}]
[ocpp  ] TRACE 2023/03/13 09:26:21 handling incoming CALL [8881372, StatusNotification] from TACW****
[ocpp  ] TRACE 2023/03/13 09:26:21 sent CALL RESULT [8881372] for TACW****
[ocpp  ] TRACE 2023/03/13 09:26:21 sent JSON message to TACW****: [3,"8881372",{}]
[ocpp  ] TRACE 2023/03/13 09:26:22 received JSON message from TACW****: [3, "4239282141", {"configurationKey": [{"key": "GetConfigurationMaxKeys", "readonly": true, "value": "20"}, {"key": "HeartbeatInterval", "readonly": false, "value": "60"}, {"key": "MeterValueSampleInterval", "readonly": false, "value": "30"}, {"key": "AuthorizationKey", "readonly": false}, {"key": "LocalAuthorizeOffline", "readonly": false, "value": "true"}, {"key": "LocalPreAuthorize", "readonly": false, "value": "false"}, {"key": "LocalAuthListEnabled", "readonly": false, "value": "true"}, {"key": "LocalAuthListMaxLength", "readonly": true, "value": "16"}, {"key": "SendLocalListMaxLength", "readonly": true, "value": "8"}, {"key": "SupportedFeatureProfiles", "readonly": true, "value": "Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger"}, {"key": "ChargeProfileMaxStackLevel", "readonly": true, "value": "3"}, {"key": "ChargingScheduleAllowedChargingRateUnit", "readonly": true, "value": "Current,Power"}, {"key": "ChargingScheduleMaxPeriods", "readonly": true, "value": "25"}, {"key": "MaxChargingProfilesInstalled", "readonly": true, "value": "16"}, {"key": "NumberOfConnectors", "readonly": true, "value": "1"}, {"key": "AllowOfflineTxForUnknownId", "readonly": false, "value": "false"}, {"key": "AuthorizeRemoteTxRequests", "readonly": false, "value": "false"}, {"key": "ConnectionTimeOut", "readonly": false, "value": "120"}, {"key": "WebSocketPingInterval", "readonly": false, "value": "60"}, {"key": "StopTransactionOnInvalidId", "readonly": false, "value": "true"}, {"key": "MaxEnergyOnInvalidId", "readonly": false, "value": "1000"}, {"key": "MeterValuesAlignedData", "readonly": false, "value": "Energy.Active.Import.Register"}, {"key": "MeterValuesAlignedDataMaxLength", "readonly": true, "value": "4"}, {"key": "ClockAlignedDataInterval", "readonly": false, "value": "0"}, {"key": "MeterValuesSampledData", "readonly": false, "value": "Current.Import,Current.Offered,Energy.Active.Import.Register,Power.Active.Import,Voltage"}, {"key": "MeterValuesSampledDataMaxLength", "readonly": true, "value": "22"}, {"key": "ResetRetries", "readonly": false, "value": "0"}, {"key": "TransactionMessageAttempts", "readonly": false, "value": "10"}, {"key": "TransactionMessageRetryInterval", "readonly": false, "value": "10"}]}]
[ocpp  ] TRACE 2023/03/13 09:26:22 sent CALL ERROR [4239282141] for TACW****
[ocpp  ] TRACE 2023/03/13 09:26:22 ocpp message (4239282141): PropertyConstraintViolation - Field CallResult.Payload.ConfigurationKey[3].Value must be maximum 500, but was 0 for feature GetConfiguration
[ocpp  ] TRACE 2023/03/13 09:26:50 timeout for client TACW****, canceling message
[ocpp  ] TRACE 2023/03/13 09:26:50 completed request 4239282141 for TACW****
[ocpp  ] TRACE 2023/03/13 09:26:50 request 4239282141 for TACW**** timed out
[ocpp  ] TRACE 2023/03/13 09:26:50 TACW**** ready to transmit again
[main  ] FATAL 2023/03/13 09:26:50 cannot create charger 'wallbox1': cannot create charger 'template': cannot create charger 'ocpp': ocpp message (4239282141): GenericError - Request timed out
[main  ] FATAL 2023/03/13 09:26:50 will attempt restart in: 5m0s
[ocpp  ] DEBUG 2023/03/13 09:27:22 chargepoint disconnected: TACW****
```

### What type of operating system are you running?

Linux

### Version

0.114.1

andig, 2023-03-13:
@OnWire nimm mal bitte die ganzen Workarounds raus, dann neues Logfile am Besten mit der nightly Version:

```
    quirks: true
    metervalues: Energy.Active.Import.Register,Power.Active.Import
    meter: true
```

OnWire, 2023-03-13:
```
chargers:
  - name: wallbox1
    type: ocpp
    timeout: 20m
    stationid: TACW***
    idtag: 1
```

```
[main  ] INFO 2023/03/13 11:31:17 evcc 0.114.1 (3c6e2ca3)
[main  ] INFO 2023/03/13 11:31:17 using config file: evcc.yaml
[main  ] INFO 2023/03/13 11:31:17 starting ui and api at :7070
[db    ] INFO 2023/03/13 11:31:17 using sqlite database: /root/.evcc/evcc.db
[TACW****] DEBUG 2023/03/13 11:31:18 waiting for chargepoint: 20m0s
[ocpp  ] DEBUG 2023/03/13 11:31:18 chargepoint connected: TACW****
[ocpp  ] TRACE 2023/03/13 11:31:18 enqueued CALL [2627073400, GetConfiguration] for TACW****
[ocpp  ] TRACE 2023/03/13 11:31:18 dispatched request 2627073400 for TACW****
[ocpp  ] TRACE 2023/03/13 11:31:18 sent JSON message to TACW****: [2,"2627073400","GetConfiguration",{}]
[ocpp  ] TRACE 2023/03/13 11:31:18 started timeout timer for TACW****
[ocpp  ] TRACE 2023/03/13 11:31:18 received JSON message from TACW****: [2, "9851728", "StatusNotification", {"connectorId": 0, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x0000"}]
[ocpp  ] TRACE 2023/03/13 11:31:18 handling incoming CALL [9851728, StatusNotification] from TACW****
[ocpp  ] TRACE 2023/03/13 11:31:18 sent CALL RESULT [9851728] for TACW****
[ocpp  ] TRACE 2023/03/13 11:31:18 sent JSON message to TACW****: [3,"9851728",{}]
[ocpp  ] TRACE 2023/03/13 11:31:19 received JSON message from TACW****: [2, "3037660", "StatusNotification", {"connectorId": 1, "errorCode": "NoError", "info": "null", "status": "Preparing", "vendorErrorCode": "0x0000"}]
[ocpp  ] TRACE 2023/03/13 11:31:19 handling incoming CALL [3037660, StatusNotification] from TACW****
[ocpp  ] TRACE 2023/03/13 11:31:19 sent CALL RESULT [3037660] for TACW****
[ocpp  ] TRACE 2023/03/13 11:31:19 sent JSON message to TACW****: [3,"3037660",{}]
[ocpp  ] TRACE 2023/03/13 11:31:19 received JSON message from TACW****: [3, "2627073400", {"configurationKey": [{"key": "GetConfigurationMaxKeys", "readonly": true, "value": "20"}, {"key": "HeartbeatInterval", "readonly": false, "value": "60"}, {"key": "MeterValueSampleInterval", "readonly": false, "value": "30"}, {"key": "AuthorizationKey", "readonly": false}, {"key": "LocalAuthorizeOffline", "readonly": false, "value": "true"}, {"key": "LocalPreAuthorize", "readonly": false, "value": "false"}, {"key": "LocalAuthListEnabled", "readonly": false, "value": "true"}, {"key": "LocalAuthListMaxLength", "readonly": true, "value": "16"}, {"key": "SendLocalListMaxLength", "readonly": true, "value": "8"}, {"key": "SupportedFeatureProfiles", "readonly": true, "value": "Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger"}, {"key": "ChargeProfileMaxStackLevel", "readonly": true, "value": "3"}, {"key": "ChargingScheduleAllowedChargingRateUnit", "readonly": true, "value": "Current,Power"}, {"key": "ChargingScheduleMaxPeriods", "readonly": true, "value": "25"}, {"key": "MaxChargingProfilesInstalled", "readonly": true, "value": "16"}, {"key": "NumberOfConnectors", "readonly": true, "value": "1"}, {"key": "AllowOfflineTxForUnknownId", "readonly": false, "value": "false"}, {"key": "AuthorizeRemoteTxRequests", "readonly": false, "value": "false"}, {"key": "ConnectionTimeOut", "readonly": false, "value": "120"}, {"key": "WebSocketPingInterval", "readonly": false, "value": "60"}, {"key": "StopTransactionOnInvalidId", "readonly": false, "value": "true"}, {"key": "MaxEnergyOnInvalidId", "readonly": false, "value": "1000"}, {"key": "MeterValuesAlignedData", "readonly": false, "value": "Energy.Active.Import.Register"}, {"key": "MeterValuesAlignedDataMaxLength", "readonly": true, "value": "4"}, {"key": "ClockAlignedDataInterval", "readonly": false, "value": "0"}, {"key": "MeterValuesSampledData", "readonly": false, "value": "Current.Import,Current.Offered,Energy.Active.Import.Register,Power.Active.Import,Voltage"}, {"key": "MeterValuesSampledDataMaxLength", "readonly": true, "value": "22"}, {"key": "ResetRetries", "readonly": false, "value": "0"}, {"key": "TransactionMessageAttempts", "readonly": false, "value": "10"}, {"key": "TransactionMessageRetryInterval", "readonly": false, "value": "10"}]}]
[ocpp  ] TRACE 2023/03/13 11:31:19 sent CALL ERROR [2627073400] for TACW****
[ocpp  ] TRACE 2023/03/13 11:31:19 ocpp message (2627073400): PropertyConstraintViolation - Field CallResult.Payload.ConfigurationKey[3].Value must be maximum 500, but was 0 for feature GetConfiguration
[ocpp  ] TRACE 2023/03/13 11:31:48 timeout for client TACW****, canceling message
[ocpp  ] TRACE 2023/03/13 11:31:48 completed request 2627073400 for TACW****
[ocpp  ] TRACE 2023/03/13 11:31:48 request 2627073400 for TACW**** timed out
[ocpp  ] TRACE 2023/03/13 11:31:48 TACW**** ready to transmit again
[main  ] FATAL 2023/03/13 11:31:48 cannot create charger 'wallbox1': cannot create charger 'ocpp': ocpp message (2627073400): GenericError - Request timed out
```

andig, 2023-03-13:
@OnWire dein erstes funktioniert ja im Prinzip. Beim zweiten/dritten kennen wir die Ursache des Problems- Fix folgt. Beim ersten Log ist mir das Problem noch nicht klar. Wenn wir das finden wollen brauchts das auch in `trace`. Oder ist das `trace`? Dann ist anscheinend die Verbindung weg- warum auch immer...

OnWire, 2023-03-14:
Hallo Andreas,

korrekt, das erste funktioniert im Prinzip aber nach kurzer Zeit kommen dann die im Log ersichtlichen Fehler. Erst nach einen Service Neustart funktioniert dann wieder alles, zumindest für kurze Zeit. Es ist bereits ein Trace.. aktuell weiß ich auch nicht wo das her kommt.

andig, 2023-03-14:
> Es ist bereits ein Trace.. aktuell weiß ich auch nicht wo das her kommt.

Oh. Der Grund ist schon klar:

> chargepoint disconnected: TACW****

Das kann alles mögliche sein, anscheinend war der dann schon vorher weg und wir haben noch die alten Daten verwendet. Dann ignorieren wir das und warten hier konkret auf die Fehlerbehebung in der Ocpp Library.

OnWire, 2023-03-14:
> Das kann alles mögliche sein, anscheinend war der dann schon vorher weg und wir haben noch die alten Daten verwendet. Dann ignorieren wir das und warten hier konkret auf die Fehlerbehebung in der Ocpp Library.

Klingt nach einem Plan. ;-)

andig, 2023-03-17:
Behoben mit 592510c5d20c525bac87502f62ef1605b35a36df. In 30min könntest Du das nightly testen.

OnWire, 2023-03-17:
> Behoben mit [592510c](https://github.com/evcc-io/evcc/commit/592510c5d20c525bac87502f62ef1605b35a36df). In 30min könntest Du das nightly testen.

Wow, das ging ja schnell! Ich teste das gleich mal.

OnWire, 2023-03-17:
Nochmal ein neues Log, der "charge meter: not available, charger timeout" tritt leider immer noch nach gewisser Zeit auf:

```
[main  ] INFO 2023/03/17 11:49:22 evcc 0.114.1 (242be64f)
[main  ] INFO 2023/03/17 11:49:22 using config file: evcc.yaml
[main  ] INFO 2023/03/17 11:49:22 starting ui and api at :7070
[db    ] INFO 2023/03/17 11:49:22 using sqlite database: /root/.evcc/evcc.db
[TACW****] DEBUG 2023/03/17 11:49:23 waiting for chargepoint: 20m0s
[ocpp  ] DEBUG 2023/03/17 11:49:38 chargepoint connected: TACW****
[ocpp  ] TRACE 2023/03/17 11:49:38 enqueued CALL [3995520824, GetConfiguration] for TACW****
[ocpp  ] TRACE 2023/03/17 11:49:38 dispatched request 3995520824 for TACW****
[ocpp  ] TRACE 2023/03/17 11:49:38 sent JSON message to TACW****: [2,"3995520824","GetConfiguration",{}]
[ocpp  ] TRACE 2023/03/17 11:49:38 started timeout timer for TACW****
[ocpp  ] TRACE 2023/03/17 11:49:38 received JSON message from TACW****: [2, "5782620", "StatusNotification", {"connectorId": 0, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x0000"}]
[ocpp  ] TRACE 2023/03/17 11:49:38 handling incoming CALL [5782620, StatusNotification] from TACW****
[ocpp  ] TRACE 2023/03/17 11:49:38 sent CALL RESULT [5782620] for TACW****
[ocpp  ] TRACE 2023/03/17 11:49:38 sent JSON message to TACW****: [3,"5782620",{}]
[ocpp  ] TRACE 2023/03/17 11:49:39 received JSON message from TACW****: [2, "1411572", "StatusNotification", {"connectorId": 1, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x0000"}]
[ocpp  ] TRACE 2023/03/17 11:49:39 handling incoming CALL [1411572, StatusNotification] from TACW****
[ocpp  ] TRACE 2023/03/17 11:49:39 sent CALL RESULT [1411572] for TACW****
[ocpp  ] TRACE 2023/03/17 11:49:39 sent JSON message to TACW****: [3,"1411572",{}]
[ocpp  ] TRACE 2023/03/17 11:49:39 received JSON message from TACW****: [3, "3995520824", {"configurationKey": [{"key": "GetConfigurationMaxKeys", "readonly": true, "value": "20"}, {"key": "HeartbeatInterval", "readonly": false, "value": "60"}, {"key": "MeterValueSampleInterval", "readonly": false, "value": "30"}, {"key": "AuthorizationKey", "readonly": false}, {"key": "LocalAuthorizeOffline", "readonly": false, "value": "true"}, {"key": "LocalPreAuthorize", "readonly": false, "value": "false"}, {"key": "LocalAuthListEnabled", "readonly": false, "value": "true"}, {"key": "LocalAuthListMaxLength", "readonly": true, "value": "16"}, {"key": "SendLocalListMaxLength", "readonly": true, "value": "8"}, {"key": "SupportedFeatureProfiles", "readonly": true, "value": "Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger"}, {"key": "ChargeProfileMaxStackLevel", "readonly": true, "value": "3"}, {"key": "ChargingScheduleAllowedChargingRateUnit", "readonly": true, "value": "Current,Power"}, {"key": "ChargingScheduleMaxPeriods", "readonly": true, "value": "25"}, {"key": "MaxChargingProfilesInstalled", "readonly": true, "value": "16"}, {"key": "NumberOfConnectors", "readonly": true, "value": "1"}, {"key": "AllowOfflineTxForUnknownId", "readonly": false, "value": "false"}, {"key": "AuthorizeRemoteTxRequests", "readonly": false, "value": "false"}, {"key": "ConnectionTimeOut", "readonly": false, "value": "120"}, {"key": "WebSocketPingInterval", "readonly": false, "value": "60"}, {"key": "StopTransactionOnInvalidId", "readonly": false, "value": "true"}, {"key": "MaxEnergyOnInvalidId", "readonly": false, "value": "1000"}, {"key": "MeterValuesAlignedData", "readonly": false, "value": "Energy.Active.Import.Register"}, {"key": "MeterValuesAlignedDataMaxLength", "readonly": true, "value": "4"}, {"key": "ClockAlignedDataInterval", "readonly": false, "value": "0"}, {"key": "MeterValuesSampledData", "readonly": false, "value": "Energy.Active.Import.Register,Power.Active.Import"}, {"key": "MeterValuesSampledDataMaxLength", "readonly": true, "value": "22"}, {"key": "ResetRetries", "readonly": false, "value": "0"}, {"key": "TransactionMessageAttempts", "readonly": false, "value": "10"}, {"key": "TransactionMessageRetryInterval", "readonly": false, "value": "10"}]}]
[ocpp  ] TRACE 2023/03/17 11:49:39 handling incoming CALL RESULT [3995520824] from TACW****
[ocpp  ] TRACE 2023/03/17 11:49:39 completed request 3995520824 for TACW****
[TACW****] ERROR 2023/03/17 11:49:39 AuthorizationKey (r/w): nil
[ocpp  ] TRACE 2023/03/17 11:49:39 TACW**** ready to transmit again
[ocpp  ] TRACE 2023/03/17 11:49:39 timeout canceled for TACW****
[ocpp  ] TRACE 2023/03/17 11:49:39 enqueued CALL [3829310821, TriggerMessage] for TACW****
[TACW****] DEBUG 2023/03/17 11:49:39 waiting for chargepoint status: 30s
[ocpp  ] TRACE 2023/03/17 11:49:39 dispatched request 3829310821 for TACW****
[ocpp  ] TRACE 2023/03/17 11:49:39 sent JSON message to TACW****: [2,"3829310821","TriggerMessage",{"requestedMessage":"MeterValues"}]
[ocpp  ] TRACE 2023/03/17 11:49:39 started timeout timer for TACW****
[ocpp  ] TRACE 2023/03/17 11:49:40 received JSON message from TACW****: [3, "3829310821", {"status": "Accepted"}]
[ocpp  ] TRACE 2023/03/17 11:49:40 handling incoming CALL RESULT [3829310821] from TACW****
[ocpp  ] TRACE 2023/03/17 11:49:40 completed request 3829310821 for TACW****
[ocpp  ] TRACE 2023/03/17 11:49:40 TriggerMessage MeterValues for TACW****: Accepted
[ocpp  ] TRACE 2023/03/17 11:49:40 TACW**** ready to transmit again
[ocpp  ] TRACE 2023/03/17 11:49:40 timeout canceled for TACW****
[... cut ...]
[site  ] DEBUG 2023/03/17 12:12:57 pv power: 2339W
[site  ] DEBUG 2023/03/17 12:12:57 battery soc: 63%
[site  ] DEBUG 2023/03/17 12:12:57 battery power: -790W
[site  ] DEBUG 2023/03/17 12:12:57 grid power: 6W
[site  ] DEBUG 2023/03/17 12:12:57 site power: -784W
[lp-1  ] ERROR 2023/03/17 12:12:57 charge meter total import: not available
[lp-1  ] ERROR 2023/03/17 12:12:57 charger: timeout
```

Einen Netzwerk- bzw. Kommunikationsfehler schließe ich aus da nach den ersten Timeouts noch ein JSON von der Wallbox kommt.

andig, 2023-03-17:
Schönes Log! Irgendwas stimmt hier nicht. Wir sehen schön, dass Messwerte abgefragt werden und kommen. Nach genau 20min ist dann Schluss mit Timeout obwohl die Box zwischendurch noch antwortet. Ich hab das Gefühl, dass wir da einen kleinen Denkfehler haben. Zumindest sind wir mal einen Schritt weiter :)

andig, 2023-03-17:
Kommando zurück: es ist weiter die Verbindung zur Box, die verloren geht:

> chargepoint disconnected: TACW****

Es ist unklar, warum das passiert. Hat die Box ein Log? Was zeigt die Box an nachdem die Verbindung weg ist? 

Ich glaub wir haben noch einen Fehler bei der Meßwertverarbeitung, die Box scheint aber zusätzlich noch ein eigenes Problem zu haben...

/cc @tarelda @xantalor

andig, 2023-03-17:
>  Ich finde es bspw. interessant das die Box am Anfang zwei JSON schickt mit Connector ID 0 und 1,

Da müsste man in den Standard schauen: kann sein, dass die 0 die Summe aller Connectoren sind.

> Bist du dir sicher das es an dem "chargepoint disconnected" liegt

Sicher ist hier gar nix :O. Komisch ist allerdings, dass ich bei Dir nur `disconnected` Meldungen sehe, aber keine `connected`. Ich schau nochmal, ob das ggf ein anderer Fehler in der Library ist.

Weiteres Problem: initiales Verbindungsdelay (=10m) wird auch für die Messwerte genutzt- blöd. Und: Meßwerte fordern wir nur an wenn es eine Transaction gibt, werfen aber auch Fehler für veraltete Werte wenns keine gibt. PR folgt.

andig, 2023-03-17:
Komischerweise gibts bei mir die reconnects im Test:

```
[ocpp  ] DEBUG 2023/03/17 17:31:16 chargepoint disconnected: cp0001
[ocpp  ] DEBUG 2023/03/17 17:31:18 chargepoint connected: cp0001
[ocpp  ] DEBUG 2023/03/17 17:31:27 chargepoint disconnected: cp0001
[ocpp  ] DEBUG 2023/03/17 17:31:29 chargepoint connected: cp0001

```

andig, 2023-03-17:
Kannst in 20min testen wenn Du magst.

OnWire, 2023-03-17:
Leider keine wirkliche Änderung, den Charger Timeout habe ich zum schnelleren Testen mal auf 5 Minuten reduziert. Genau nach den 5 Minuten läuft er auch da wieder rein. 

[main  ] INFO 2023/03/17 20:18:35 evcc 0.114.1 (ec8c7631)
[main  ] INFO 2023/03/17 20:18:35 using config file: evcc.yaml
[main  ] INFO 2023/03/17 20:18:35 starting ui and api at :7070
[db    ] INFO 2023/03/17 20:18:35 using sqlite database: /root/.evcc/evcc.db
[TACW****] DEBUG 2023/03/17 20:18:36 waiting for chargepoint: 2m0s
[ocpp  ] DEBUG 2023/03/17 20:20:14 chargepoint connected: TACW****
[ocpp  ] TRACE 2023/03/17 20:20:14 enqueued CALL [3744316402, GetConfiguration] for TACW****
[ocpp  ] TRACE 2023/03/17 20:20:14 dispatched request 3744316402 for TACW****
[ocpp  ] TRACE 2023/03/17 20:20:14 sent JSON message to TACW****: [2,"3744316402","GetConfiguration",{}]
[ocpp  ] TRACE 2023/03/17 20:20:14 started timeout timer for TACW****
[ocpp  ] TRACE 2023/03/17 20:20:15 received JSON message from TACW****: [2, "6308608", "StatusNotification", {"connectorId": 0, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x0000"}]
[ocpp  ] TRACE 2023/03/17 20:20:15 handling incoming CALL [6308608, StatusNotification] from TACW****
[ocpp  ] TRACE 2023/03/17 20:20:15 sent CALL RESULT [6308608] for TACW****
[ocpp  ] TRACE 2023/03/17 20:20:15 sent JSON message to TACW****: [3,"6308608",{}]
[ocpp  ] TRACE 2023/03/17 20:20:15 received JSON message from TACW****: [2, "8096232", "StatusNotification", {"connectorId": 1, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x0000"}]
[ocpp  ] TRACE 2023/03/17 20:20:15 handling incoming CALL [8096232, StatusNotification] from TACW****
[ocpp  ] TRACE 2023/03/17 20:20:15 sent CALL RESULT [8096232] for TACW****
[ocpp  ] TRACE 2023/03/17 20:20:15 sent JSON message to TACW****: [3,"8096232",{}]
[ocpp  ] TRACE 2023/03/17 20:20:16 received JSON message from TACW****: [3, "3744316402", {"configurationKey": [{"key": "GetConfigurationMaxKeys", "readonly": true, "value": "20"}, {"key": "HeartbeatInterval", "readonly": false, "value": "60"}, {"key": "MeterValueSampleInterval", "readonly": false, "value": "30"}, {"key": "AuthorizationKey", "readonly": false}, {"key": "LocalAuthorizeOffline", "readonly": false, "value": "true"}, {"key": "LocalPreAuthorize", "readonly": false, "value": "false"}, {"key": "LocalAuthListEnabled", "readonly": false, "value": "true"}, {"key": "LocalAuthListMaxLength", "readonly": true, "value": "16"}, {"key": "SendLocalListMaxLength", "readonly": true, "value": "8"}, {"key": "SupportedFeatureProfiles", "readonly": true, "value": "Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger"}, {"key": "ChargeProfileMaxStackLevel", "readonly": true, "value": "3"}, {"key": "ChargingScheduleAllowedChargingRateUnit", "readonly": true, "value": "Current,Power"}, {"key": "ChargingScheduleMaxPeriods", "readonly": true, "value": "25"}, {"key": "MaxChargingProfilesInstalled", "readonly": true, "value": "16"}, {"key": "NumberOfConnectors", "readonly": true, "value": "1"}, {"key": "AllowOfflineTxForUnknownId", "readonly": false, "value": "false"}, {"key": "AuthorizeRemoteTxRequests", "readonly": false, "value": "false"}, {"key": "ConnectionTimeOut", "readonly": false, "value": "120"}, {"key": "WebSocketPingInterval", "readonly": false, "value": "60"}, {"key": "StopTransactionOnInvalidId", "readonly": false, "value": "true"}, {"key": "MaxEnergyOnInvalidId", "readonly": false, "value": "1000"}, {"key": "MeterValuesAlignedData", "readonly": false, "value": "Energy.Active.Import.Register"}, {"key": "MeterValuesAlignedDataMaxLength", "readonly": true, "value": "4"}, {"key": "ClockAlignedDataInterval", "readonly": false, "value": "0"}, {"key": "MeterValuesSampledData", "readonly": false, "value": "Energy.Active.Import.Register,Power.Active.Import"}, {"key": "MeterValuesSampledDataMaxLength", "readonly": true, "value": "22"}, {"key": "ResetRetries", "readonly": false, "value": "0"}, {"key": "TransactionMessageAttempts", "readonly": false, "value": "10"}, {"key": "TransactionMessageRetryInterval", "readonly": false, "value": "10"}]}]
[ocpp  ] TRACE 2023/03/17 20:20:16 handling incoming CALL RESULT [3744316402] from TACW****
[ocpp  ] TRACE 2023/03/17 20:20:16 completed request 3744316402 for TACW****
[ocpp  ] TRACE 2023/03/17 20:20:16 TACW**** ready to transmit again
[ocpp  ] TRACE 2023/03/17 20:20:16 timeout canceled for TACW****
[TACW****] ERROR 2023/03/17 20:20:16 AuthorizationKey (r/w): nil
[ocpp  ] TRACE 2023/03/17 20:20:16 enqueued CALL [578476715, TriggerMessage] for TACW****
[TACW****] DEBUG 2023/03/17 20:20:16 waiting for chargepoint status: 30s
[ocpp  ] TRACE 2023/03/17 20:20:16 dispatched request 578476715 for TACW****
[ocpp  ] TRACE 2023/03/17 20:20:16 sent JSON message to TACW****: [2,"578476715","TriggerMessage",{"requestedMessage":"MeterValues"}]
[ocpp  ] TRACE 2023/03/17 20:20:16 started timeout timer for TACW****
[ocpp  ] TRACE 2023/03/17 20:20:17 received JSON message from TACW****: [3, "578476715", {"status": "Accepted"}]
[ocpp  ] TRACE 2023/03/17 20:20:17 handling incoming CALL RESULT [578476715] from TACW****
[ocpp  ] TRACE 2023/03/17 20:20:17 completed request 578476715 for TACW****
[ocpp  ] TRACE 2023/03/17 20:20:17 TACW**** ready to transmit again
[ocpp  ] TRACE 2023/03/17 20:20:17 timeout canceled for TACW****
[ocpp  ] TRACE 2023/03/17 20:20:17 TriggerMessage MeterValues for TACW****: Accepted
[... cut ...]
[lp-1  ] DEBUG 2023/03/17 20:30:48 charge power: 458W
[site  ] DEBUG 2023/03/17 20:30:48 pv power: 0W
[site  ] DEBUG 2023/03/17 20:30:48 battery soc: 66%
[site  ] DEBUG 2023/03/17 20:30:48 battery power: 257W
[site  ] DEBUG 2023/03/17 20:30:48 grid power: 32W
[site  ] DEBUG 2023/03/17 20:30:48 site power: 289W
[lp-1  ] DEBUG 2023/03/17 20:30:48 charge meter total import: 0.425kWh
[lp-1  ] ERROR 2023/03/17 20:30:48 charger: timeout
[ocpp  ] DEBUG 2023/03/17 20:30:58 chargepoint disconnected: TACW****
[site  ] DEBUG 2023/03/17 20:31:03 ----

andig, 2023-03-17:
Hast du nochmal die Config?

OnWire, 2023-03-17:
```
chargers:
- name: wallbox1
  type: ocpp
  timeout: 5m
  stationid: TACW***
  idtag: 1
```
Der Rest entspricht der zu Anfang geposteten Config. Unterschied zum vorherigen Test wie gesagt der Timeout auf 5 Minuten reduziert.

OnWire, 2023-03-17:
Würde es vielleicht Sinn machen den Timeout Counter von vorne zu starten wenn eine Status Notification der Box kommt? So wie ich das Log interpretiere wird dieser nur am Anfang des initialen "Handshake" gecancelt bzw wenn ein Trigger an die Box geht, aber nicht mehr im weiteren Verlauf sodass er nach dem definierten Timeout dann den charger timeout meldet? Nur mal so geraten ohne den genauen Code zu kennen.

andig, 2023-03-18:
Sehr gut, wir nähern uns langsam. Ich glaub ich verstehe jetzt das Problem. Deine WB schickt von sich aus keine `StatusNotification`. D.h. die Box ist noch "da", evcc "denkt" aber irgendwann "timeout". Das sollte sich lösen lassen. PR folgt.

andig, 2023-03-18:
Gerne neuer Test in 20min

andig, 2023-03-18:
Ich mach schonmal zu, hoffe das passt. Falls nein bitte melden.

OnWire, 2023-03-19:
Ich kam leider erst heute dazu es noch einmal zu testen. Noch sind wir scheinbar nicht am Ziel, aber zumindest einen großen Schritt weiter!

```
[main  ] INFO 2023/03/19 13:27:11 evcc 0.114.1 (52d63a2e)
[main  ] INFO 2023/03/19 13:27:11 using config file: evcc.yaml
[main  ] INFO 2023/03/19 13:27:11 starting ui and api at :7070
[db    ] INFO 2023/03/19 13:27:11 using sqlite database: /root/.evcc/evcc.db
[TACW****] DEBUG 2023/03/19 13:27:12 waiting for chargepoint: 2m0s
[ocpp  ] DEBUG 2023/03/19 13:27:36 chargepoint connected: TACW****
[ocpp  ] TRACE 2023/03/19 13:27:36 enqueued CALL [1962757986, GetConfiguration] for TACW****
[ocpp  ] TRACE 2023/03/19 13:27:36 dispatched request 1962757986 for TACW****
[ocpp  ] TRACE 2023/03/19 13:27:36 sent JSON message to TACW****: [2,"1962757986","GetConfiguration",{}]
[ocpp  ] TRACE 2023/03/19 13:27:36 started timeout timer for TACW****
[ocpp  ] TRACE 2023/03/19 13:27:37 received JSON message from TACW****: [2, "3703933", "StatusNotification", {"connectorId": 0, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x0000"}]
[ocpp  ] TRACE 2023/03/19 13:27:37 handling incoming CALL [3703933, StatusNotification] from TACW****
[ocpp  ] TRACE 2023/03/19 13:27:37 sent CALL RESULT [3703933] for TACW****
[ocpp  ] TRACE 2023/03/19 13:27:37 sent JSON message to TACW****: [3,"3703933",{}]
[ocpp  ] TRACE 2023/03/19 13:27:37 received JSON message from TACW****: [2, "9234764", "StatusNotification", {"connectorId": 1, "errorCode": "NoError", "info": "null", "status": "Finishing", "vendorErrorCode": "0x0000"}]
[ocpp  ] TRACE 2023/03/19 13:27:37 handling incoming CALL [9234764, StatusNotification] from TACW****
[ocpp  ] TRACE 2023/03/19 13:27:37 sent CALL RESULT [9234764] for TACW****
[ocpp  ] TRACE 2023/03/19 13:27:37 sent JSON message to TACW****: [3,"9234764",{}]
[ocpp  ] TRACE 2023/03/19 13:27:38 received JSON message from TACW****: [3, "1962757986", {"configurationKey": [{"key": "GetConfigurationMaxKeys", "readonly": true, "value": "20"}, {"key": "HeartbeatInterval", "readonly": false, "value": "60"}, {"key": "MeterValueSampleInterval", "readonly": false, "value": "30"}, {"key": "AuthorizationKey", "readonly": false}, {"key": "LocalAuthorizeOffline", "readonly": false, "value": "true"}, {"key": "LocalPreAuthorize", "readonly": false, "value": "false"}, {"key": "LocalAuthListEnabled", "readonly": false, "value": "true"}, {"key": "LocalAuthListMaxLength", "readonly": true, "value": "16"}, {"key": "SendLocalListMaxLength", "readonly": true, "value": "8"}, {"key": "SupportedFeatureProfiles", "readonly": true, "value": "Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger"}, {"key": "ChargeProfileMaxStackLevel", "readonly": true, "value": "3"}, {"key": "ChargingScheduleAllowedChargingRateUnit", "readonly": true, "value": "Current,Power"}, {"key": "ChargingScheduleMaxPeriods", "readonly": true, "value": "25"}, {"key": "MaxChargingProfilesInstalled", "readonly": true, "value": "16"}, {"key": "NumberOfConnectors", "readonly": true, "value": "1"}, {"key": "AllowOfflineTxForUnknownId", "readonly": false, "value": "false"}, {"key": "AuthorizeRemoteTxRequests", "readonly": false, "value": "false"}, {"key": "ConnectionTimeOut", "readonly": false, "value": "120"}, {"key": "WebSocketPingInterval", "readonly": false, "value": "60"}, {"key": "StopTransactionOnInvalidId", "readonly": false, "value": "true"}, {"key": "MaxEnergyOnInvalidId", "readonly": false, "value": "1000"}, {"key": "MeterValuesAlignedData", "readonly": false, "value": "Energy.Active.Import.Register"}, {"key": "MeterValuesAlignedDataMaxLength", "readonly": true, "value": "4"}, {"key": "ClockAlignedDataInterval", "readonly": false, "value": "0"}, {"key": "MeterValuesSampledData", "readonly": false, "value": "Energy.Active.Import.Register,Power.Active.Import"}, {"key": "MeterValuesSampledDataMaxLength", "readonly": true, "value": "22"}, {"key": "ResetRetries", "readonly": false, "value": "0"}, {"key": "TransactionMessageAttempts", "readonly": false, "value": "10"}, {"key": "TransactionMessageRetryInterval", "readonly": false, "value": "10"}]}]
[ocpp  ] TRACE 2023/03/19 13:27:38 handling incoming CALL RESULT [1962757986] from TACW****
[ocpp  ] TRACE 2023/03/19 13:27:38 completed request 1962757986 for TACW****
[ocpp  ] TRACE 2023/03/19 13:27:38 TACW**** ready to transmit again
[ocpp  ] TRACE 2023/03/19 13:27:38 timeout canceled for TACW****
[ocpp  ] TRACE 2023/03/19 13:27:38 enqueued CALL [3171654433, TriggerMessage] for TACW****
[ocpp  ] TRACE 2023/03/19 13:27:38 dispatched request 3171654433 for TACW****
[ocpp  ] TRACE 2023/03/19 13:27:38 sent JSON message to TACW****: [2,"3171654433","TriggerMessage",{"requestedMessage":"MeterValues"}]
[ocpp  ] TRACE 2023/03/19 13:27:38 started timeout timer for TACW****
[ocpp  ] TRACE 2023/03/19 13:27:39 received JSON message from TACW****: [3, "3171654433", {"status": "Accepted"}]
[ocpp  ] TRACE 2023/03/19 13:27:39 handling incoming CALL RESULT [3171654433] from TACW****
[ocpp  ] TRACE 2023/03/19 13:27:39 completed request 3171654433 for TACW****
[ocpp  ] TRACE 2023/03/19 13:27:39 TACW**** ready to transmit again
[ocpp  ] TRACE 2023/03/19 13:27:39 TriggerMessage MeterValues for TACW****: Accepted
[ocpp  ] TRACE 2023/03/19 13:27:39 timeout canceled for TACW****
[... cut ...]
[lp-1  ] DEBUG 2023/03/19 13:41:41 charge meter total import: 0.139kWh
[lp-1  ] DEBUG 2023/03/19 13:41:41 charger status: B
[lp-1  ] DEBUG 2023/03/19 13:41:41 pv charge current: 1.7A = 0A + 1.7A (-391W @ 1p)
[ocpp  ] DEBUG 2023/03/19 13:41:54 chargepoint disconnected: TACW****
[site  ] DEBUG 2023/03/19 13:41:56 ----
```

Wie man sieht tritt der Timeout zwar immer noch auf, die "eingeschlafene" Verbindung kommt aber ohne Dienstneustart wieder von alleine zurück. Dann gibt es für 2-3 Minuten Daten, wieder Timeout für 2-3 Minuten und das Spiel beginnt von vorne.

andig, 2023-03-19:
> chargepoint disconnected: TACW****

Soweit ich das sehe, treten die Timeouts ab dem Moment auf, wo die Box sich disconnected, richtig? Das erscheint mir ja erstmal plausibel. Insofern würde ich sagen alles richtig.

Bleibt die Frage, warum die Verbindung abbricht. Müsste evcc hier irgendwas tun damit sie offen bleibt? Falls nein läge das Problem wohl bei der Box?

Ich muss mir nochmal das Thema Heartbeat anschauen.

andig, 2023-03-19:
Laut https://library.e.abb.com/public/0d9afc6917f84901be1d409da42c1fdf/ABB_Terra_AC_Charger_OCPP1.6_ImplementationOverview%20_v1.7.pdf:

> HeartbeatInterval: Interval of inactivity (no OCPP exchanges) with central system after which the Charge Point
should send a Heartbeat.req PDU.

Klingt für mich, als müsste die Box sich melden. Davon ist aber im Log nichts zu sehen. Könntest Du bitte zur Sicherheit auch die charger id mal auf `trace` setzen? Evtl. gibts da noch mehr. Sonst gehen mir die Ideen aus...

/cc @tarelda @xantalor maybe you guys have an idea?

OnWire, 2023-03-19:
> > chargepoint disconnected: TACW****
> 
> Soweit ich das sehe, treten die Timeouts ab dem Moment auf, wo die Box sich disconnected, richtig?

Korrekt!

Was mir gerade auffällt wo die Sonne scheint, wenn es eine scheinbar kontinuierliche Kommunikation zwischen evcc und Box gibt, tritt der Timeout nicht auf weil der Kommunikationsdialog scheinbar ständig am Leben gehalten wird.

andig, 2023-03-19:
Tja. Ich würde das für einen Bug der ABB halten, da bin ich aber auf dünnem Eis. Warum bekommen wir keine Heartbeats?

andig, 2023-03-19:
Wenn Du magst könntest Du mal gegen Steve testen obs da besser ist- ich weiss aber nicht, wie man da an Logs käme...

OnWire, 2023-03-19:
> Klingt für mich, als müsste die Box sich melden. Davon ist aber im Log nichts zu sehen. Könntest Du bitte zur Sicherheit auch die charger id mal auf `trace` setzen? Evtl. gibts da noch mehr. Sonst gehen mir die Ideen aus...

Könntest du mir kurz sagen wie?  "evcc -c evcc.yaml -l debug,ocpp:trace,charger:trace" ?

Ich hab hier aber noch einen Logschnipsel wo was von Heartbeats der Box zu erkennen ist, allerdings in einem anderen Kontext. (siehe cannot create error am Anfang)

```
[main  ] INFO 2023/03/19 14:15:41 evcc 0.114.1 (52d63a2e)
[main  ] INFO 2023/03/19 14:15:41 using config file: evcc.yaml
[main  ] INFO 2023/03/19 14:15:41 starting ui and api at :7070
[db    ] INFO 2023/03/19 14:15:41 using sqlite database: /root/.evcc/evcc.db
[TACW****] DEBUG 2023/03/19 14:15:42 waiting for chargepoint: 2m0s
[main  ] FATAL 2023/03/19 14:17:42 cannot create charger 'wallbox1': cannot create charger 'ocpp': timeout
[main  ] FATAL 2023/03/19 14:17:42 will attempt restart in: 5m0s
[ocpp  ] DEBUG 2023/03/19 14:17:58 chargepoint connected: TACW****
[ocpp  ] TRACE 2023/03/19 14:17:59 received JSON message from TACW****: [2, "3931892", "StatusNotification", {"connectorId": 0, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x0000"}]
[ocpp  ] TRACE 2023/03/19 14:17:59 handling incoming CALL [3931892, StatusNotification] from TACW****
[ocpp  ] TRACE 2023/03/19 14:17:59 sent CALL RESULT [3931892] for TACW****
[ocpp  ] TRACE 2023/03/19 14:17:59 sent JSON message to TACW****: [3,"3931892",{}]
[ocpp  ] TRACE 2023/03/19 14:17:59 received JSON message from TACW****: [2, "0139727", "StatusNotification", {"connectorId": 1, "errorCode": "NoError", "info": "null", "status": "Finishing", "vendorErrorCode": "0x0000"}]
[ocpp  ] TRACE 2023/03/19 14:17:59 handling incoming CALL [0139727, StatusNotification] from TACW****
[ocpp  ] TRACE 2023/03/19 14:17:59 sent CALL RESULT [0139727] for TACW****
[ocpp  ] TRACE 2023/03/19 14:17:59 sent JSON message to TACW****: [3,"0139727",{}]
[ocpp  ] TRACE 2023/03/19 14:19:03 received JSON message from TACW****: [2, "7290599", "Heartbeat", {}]
[ocpp  ] TRACE 2023/03/19 14:19:03 handling incoming CALL [7290599, Heartbeat] from TACW****
[ocpp  ] TRACE 2023/03/19 14:19:03 sent CALL RESULT [7290599] for TACW****
[ocpp  ] TRACE 2023/03/19 14:19:03 sent JSON message to TACW****: [3,"7290599",{"currentTime":"2023-03-19T13:19:03Z"}]
[ocpp  ] TRACE 2023/03/19 14:20:06 received JSON message from TACW****: [2, "8106258", "Heartbeat", {}]
[ocpp  ] TRACE 2023/03/19 14:20:06 handling incoming CALL [8106258, Heartbeat] from TACW****
[ocpp  ] TRACE 2023/03/19 14:20:06 sent CALL RESULT [8106258] for TACW****
[ocpp  ] TRACE 2023/03/19 14:20:06 sent JSON message to TACW****: [3,"8106258",{"currentTime":"2023-03-19T13:20:06Z"}]
[ocpp  ] TRACE 2023/03/19 14:21:10 received JSON message from TACW****: [2, "8927537", "Heartbeat", {}]
[ocpp  ] TRACE 2023/03/19 14:21:10 handling incoming CALL [8927537, Heartbeat] from TACW****
[ocpp  ] TRACE 2023/03/19 14:21:10 sent CALL RESULT [8927537] for TACW****
[ocpp  ] TRACE 2023/03/19 14:21:10 sent JSON message to TACW****: [3,"8927537",{"currentTime":"2023-03-19T13:21:10Z"}]
[ocpp  ] TRACE 2023/03/19 14:22:14 received JSON message from TACW****: [2, "2817504", "Heartbeat", {}]
[ocpp  ] TRACE 2023/03/19 14:22:14 handling incoming CALL [2817504, Heartbeat] from TACW****
[ocpp  ] TRACE 2023/03/19 14:22:14 sent CALL RESULT [2817504] for TACW****
[ocpp  ] TRACE 2023/03/19 14:22:14 sent JSON message to TACW****: [3,"2817504",{"currentTime":"2023-03-19T13:22:14Z"}]
```

andig, 2023-03-19:
> Könntest du mir kurz sagen wie

Vergiss das. Alle Nachrichten gehen ja an das "central system" und da loggen wir schon alles. Es kommt also nix an :(

OnWire, 2023-03-19:
Hier mal der Vollständigkeit halber noch ein Log wenn ein evcc und Box ständig im Gespräch sind (und es nicht einschläft):

```
Mar 19 14:33:44 raspi evcc[13827]: [main  ] INFO 2023/03/19 14:33:44 evcc 0.114.1 (52d63a2e)
Mar 19 14:33:44 raspi evcc[13827]: [main  ] INFO 2023/03/19 14:33:44 using config file: /etc/evcc.yaml
Mar 19 14:33:44 raspi evcc[13827]: [main  ] INFO 2023/03/19 14:33:44 starting ui and api at :7070
Mar 19 14:33:44 raspi evcc[13827]: [db    ] INFO 2023/03/19 14:33:44 using sqlite database: /var/lib/evcc/evcc.db
Mar 19 14:34:56 raspi evcc[13827]: [ocpp  ] DEBUG 2023/03/19 14:34:56 chargepoint connected: TACW****
Mar 19 14:34:56 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:34:56 enqueued CALL [3228930192, GetConfiguration] for TACW****
Mar 19 14:34:56 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:34:56 dispatched request 3228930192 for TACW****
Mar 19 14:34:56 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:34:56 sent JSON message to TACW****: [2,"3228930192","GetConfiguration",{}]
Mar 19 14:34:56 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:34:56 started timeout timer for TACW****
Mar 19 14:34:57 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:34:57 received JSON message from TACW****: [2, "5647073", "StatusNotification", {"connectorId": 0, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x0000"}]
Mar 19 14:34:57 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:34:57 handling incoming CALL [5647073, StatusNotification] from TACW****
Mar 19 14:34:57 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:34:57 sent CALL RESULT [5647073] for TACW****
Mar 19 14:34:57 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:34:57 sent JSON message to TACW****: [3,"5647073",{}]
Mar 19 14:34:58 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:34:58 received JSON message from TACW****: [2, "9901621", "StatusNotification", {"connectorId": 1, "errorCode": "NoError", "info": "null", "status": "Finishing", "vendorErrorCode": "0x0000"}]
Mar 19 14:34:58 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:34:58 handling incoming CALL [9901621, StatusNotification] from TACW****
Mar 19 14:34:58 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:34:58 sent CALL RESULT [9901621] for TACW****
Mar 19 14:34:58 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:34:58 sent JSON message to TACW****: [3,"9901621",{}]
Mar 19 14:34:59 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:34:59 received JSON message from TACW****: [3, "3228930192", {"configurationKey": [{"key": "GetConfigurationMaxKeys", "readonly": true, "value": "20"}, {"key": "HeartbeatInterval", "readonly": false, "value": "60"}, {"key": "MeterValueSampleInterval", "readonly": false, "value": "30"}, {"key": "AuthorizationKey", "readonly": false}, {"key": "LocalAuthorizeOffline", "readonly": false, "value": "true"}, {"key": "LocalPreAuthorize", "readonly": false, "value": "false"}, {"key": "LocalAuthListEnabled", "readonly": false, "value": "true"}, {"key": "LocalAuthListMaxLength", "readonly": true, "value": "16"}, {"key": "SendLocalListMaxLength", "readonly": true, "value": "8"}, {"key": "SupportedFeatureProfiles", "readonly": true, "value": "Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger"}, {"key": "ChargeProfileMaxStackLevel", "readonly": true, "value": "3"}, {"key": "ChargingScheduleAllowedChargingRateUnit", "readonly": true, "value": "Current,Power"}, {"key": "ChargingScheduleMaxPeriods", "readonly": true, "value": "25"}, {"key": "MaxChargingProfilesInstalled", "readonly": true, "value": "16"}, {"key": "NumberOfConnectors", "readonly": true, "value": "1"}, {"key": "AllowOfflineTxForUnknownId", "readonly": false, "value": "false"}, {"key": "AuthorizeRemoteTxRequests", "readonly": false, "value": "false"}, {"key": "ConnectionTimeOut", "readonly": false, "value": "120"}, {"key": "WebSocketPingInterval", "readonly": false, "value": "60"}, {"key": "StopTransactionOnInvalidId", "readonly": false, "value": "true"}, {"key": "MaxEnergyOnInvalidId", "readonly": false, "value": "1000"}, {"key": "MeterValuesAlignedData", "readonly": false, "value": "Energy.Active.Import.Register"}, {"key": "MeterValuesAlignedDataMaxLength", "readonly": true, "value": "4"}, {"key": "ClockAlignedDataInterval", "readonly": false, "value": "0"}, {"key": "MeterValuesSampledData", "readonly": false, "value": "Energy.Active.Import.Register,Power.Active.Import"}, {"key": "MeterValuesSampledDataMaxLength", "readonly": true, "value": "22"}, {"key": "ResetRetries", "readonly": false, "value": "0"}, {"key": "TransactionMessageAttempts", "readonly": false, "value": "10"}, {"key": "TransactionMessageRetryInterval", "readonly": false, "value": "10"}]}]
Mar 19 14:34:59 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:34:59 handling incoming CALL RESULT [3228930192] from TACW****
Mar 19 14:34:59 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:34:59 completed request 3228930192 for TACW****
Mar 19 14:34:59 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:34:59 TACW**** ready to transmit again
Mar 19 14:34:59 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:34:59 timeout canceled for TACW****
Mar 19 14:34:59 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:34:59 enqueued CALL [2279743285, TriggerMessage] for TACW****
Mar 19 14:34:59 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:34:59 dispatched request 2279743285 for TACW****
[... cut ...]
Mar 19 14:44:33 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:44:33 received JSON message from TACW****: [2, "5760688", "Heartbeat", {}]
Mar 19 14:44:33 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:44:33 handling incoming CALL [5760688, Heartbeat] from TACW****
Mar 19 14:44:33 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:44:33 sent CALL RESULT [5760688] for TACW****
Mar 19 14:44:33 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/19 14:44:33 sent JSON message to TACW****: [3,"5760688",{"currentTime":"2023-03-19T13:44:33Z"}]
```

andig, 2023-03-19:
> Hier mal der Vollständigkeit halber noch ein Log wenn evcc und Box ständig im Gespräch sind (und es nicht einschläft):

Ist klar- da wird geladen und deshalb ständig die Leistung gesteuert. Mach bitte trotzdem mal ein Ticket bei ABB: erwarten die irgendwie, dass die Box ständig mit Traffic versehen wird?

andig, 2023-03-19:
Und jetzt nochmal aus der Spec:

> The Charge Point MAY skip sending a Heartbeat.req PDU when another PDU has been sent to the Central System within the configured heartbeat interval. This implies that a Central System SHOULD assume availability of a Charge Point whenever a PDU has been received, the same way as it would have, when it received a Heartbeat.req PDU.

> With JSON over WebSocket, sending heartbeats is not mandatory. However, for time synchronization it is advised to at least send one heartbeat per 24 hour.

sreichholf, 2023-03-19:
So ganz theoretisch könnte das auch am WebSocket keepalive liegen, oder? Der wird ja der Grund sein warum ein expliziter ocpp Heartbeat mit WebSocket optional ist. Vielleicht ist der zu lang für die TerraAC?!

andig, 2023-03-19:
Keine Ahnung. Hat aber auch nix mit den fehlenden Heartbeats zu tun.

Sixten85, 2023-03-19:
Ich vermute mal die Terra Implementation Overview ist bekannt oder? https://library.e.abb.com/public/0d9afc6917f84901be1d409da42c1fdf/ABB_Terra_AC_Charger_OCPP1.6_ImplementationOverview%20_v1.7.pdf

Ich versteh den
2.3.3. WebSocket ping in relation to OCPP Heartbeat
Absatz so als wenn die Box von sich aus gar keine anstalten macht was zu senden sondern nur auf Anforderung seine Arbeit aufnimmt.

OnWire, 2023-03-20:
Das hier ist alles sehr komisch... ich schaue heute morgen noch einmal in das Log und was sehe ich?

```
Mar 20 08:52:17 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:52:16 pv charge current: 0.139A = 0A + 0.139A (-32W @ 1p)
Mar 20 08:52:27 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/20 08:52:27 received JSON message from TACW****: [2, "1008192", "Heartbeat", {}]
Mar 20 08:52:27 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/20 08:52:27 handling incoming CALL [1008192, Heartbeat] from TACW****
Mar 20 08:52:27 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/20 08:52:27 sent CALL RESULT [1008192] for TACW****
Mar 20 08:52:27 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/20 08:52:27 sent JSON message to TACW****: [3,"1008192",{"currentTime":"2023-03-20T07:52:27Z"}]
Mar 20 08:52:31 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:52:31 charge power: 1375W
Mar 20 08:52:31 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:52:31 charge meter total import: 0.461kWh
Mar 20 08:52:31 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:52:31 charger status: B
Mar 20 08:52:31 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:52:31 pv charge current: 0.3A = 0A + 0.3A (-69W @ 1p)
Mar 20 08:52:46 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:52:46 charge power: 1375W
Mar 20 08:52:46 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:52:46 charge meter total import: 0.461kWh
Mar 20 08:52:46 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:52:46 charger status: B
Mar 20 08:52:46 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:52:46 pv charge current: 0A = 0A + -0.0609A (14W @ 1p)
Mar 20 08:53:01 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:53:01 charge power: 1375W
Mar 20 08:53:02 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:53:02 charge meter total import: 0.461kWh
Mar 20 08:53:02 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:53:02 charger status: B
Mar 20 08:53:02 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:53:02 pv charge current: 0.174A = 0A + 0.174A (-40W @ 1p)
Mar 20 08:53:16 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:53:16 charge power: 1375W
Mar 20 08:53:17 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:53:17 charge meter total import: 0.461kWh
Mar 20 08:53:17 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:53:17 charger status: B
Mar 20 08:53:17 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:53:17 pv charge current: 0.13A = 0A + 0.13A (-30W @ 1p)
Mar 20 08:53:31 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/20 08:53:31 received JSON message from TACW****: [2, "4812868", "Heartbeat", {}]
Mar 20 08:53:31 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/20 08:53:31 handling incoming CALL [4812868, Heartbeat] from TACW****
Mar 20 08:53:31 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/20 08:53:31 sent CALL RESULT [4812868] for TACW****
Mar 20 08:53:31 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/20 08:53:31 sent JSON message to TACW****: [3,"4812868",{"currentTime":"2023-03-20T07:53:31Z"}]
Mar 20 08:53:31 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:53:31 charge power: 1375W
Mar 20 08:53:31 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:53:31 charge meter total import: 0.461kWh
Mar 20 08:53:31 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:53:31 charger status: B
Mar 20 08:53:31 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:53:31 pv charge current: 0A = 0A + -0.648A (149W @ 1p)
Mar 20 08:53:46 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:53:46 charge power: 1375W
Mar 20 08:53:46 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:53:46 charge meter total import: 0.461kWh
Mar 20 08:53:46 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:53:46 charger status: B
Mar 20 08:53:46 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:53:46 pv charge current: 0.23A = 0A + 0.23A (-53W @ 1p)
Mar 20 08:54:01 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:54:01 charge power: 1375W
Mar 20 08:54:02 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:54:02 charge meter total import: 0.461kWh
Mar 20 08:54:02 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:54:02 charger status: B
Mar 20 08:54:02 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:54:02 pv charge current: 0.0913A = 0A + 0.0913A (-21W @ 1p)
Mar 20 08:54:16 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:54:16 charge power: 1375W
Mar 20 08:54:17 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:54:17 charge meter total import: 0.461kWh
Mar 20 08:54:17 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:54:17 charger status: B
Mar 20 08:54:17 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:54:17 pv charge current: 0.0261A = 0A + 0.0261A (-6W @ 1p)
Mar 20 08:54:31 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:54:31 charge power: 1375W
Mar 20 08:54:31 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:54:31 charge meter total import: 0.461kWh
Mar 20 08:54:31 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:54:31 charger status: B
Mar 20 08:54:31 raspi evcc[13827]: [lp-1  ] DEBUG 2023/03/20 08:54:31 pv charge current: 0.0609A = 0A + 0.0609A (-14W @ 1p)
Mar 20 08:54:35 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/20 08:54:35 received JSON message from TACW****: [2, "8787683", "Heartbeat", {}]
Mar 20 08:54:35 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/20 08:54:35 handling incoming CALL [8787683, Heartbeat] from TACW****
Mar 20 08:54:35 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/20 08:54:35 sent CALL RESULT [8787683] for TACW****
Mar 20 08:54:35 raspi evcc[13827]: [ocpp  ] TRACE 2023/03/20 08:54:35 sent JSON message to TACW****: [3,"8787683",{"currentTime":"2023-03-20T07:54:35Z"}]
```

Schön brav jede Minute ein Heartbeat von der Box. O_o 

An der Konfiguration habe ich seit gestern nichts geändert und auch den Dienst habe ich nicht neu gestartet... irgendwie hat sich das von alleine über Nach "eingerenkt".

andig, 2023-03-20:
Na spontane Selbstteilung ist doch immer noch die beste Lösung. Ich mach hier mal wieder zu- aktuell kann ich eh nichts mehr tun...

closed 2023-03-20 by andig

andig, 2023-03-20:
Klingt nach Fehlermeldung an ABB, v.a. weil sich die Box ohne ja auch abmeldet...

OnWire, 2023-03-20:
Aber grundsätzlich läuft ja jetzt erst einmal alles was die Hauptsache ist.

Vielen Dank @andig und der Rest vom Entwicklerteam! Ich halte euch auf dem Laufenden wenn ich von ABB eine Rückmeldung habe. (dauert erfahrungsgemäß etwas... ;-) )

sreichholf, 2023-03-27:
Ich möchte hier nochmal kurz einhaken, weil mir zu den disconnects soeben etwas aufgefallen ist was mir etwas zu "wenig zufällig" erscheint.

`[ocpp  ] TRACE 2023/03/27 07:25:56 handling incoming CALL [2471772, MeterValues] from TerraAC

[ocpp  ] TRACE 2023/03/27 07:25:56 sent CALL RESULT [2471772] for TerraAC

[ocpp  ] TRACE 2023/03/27 07:25:56 sent JSON message to TerraAC: [3,"2471772",{}]

[ocpp  ] TRACE 2023/03/27 07:26:25 received JSON message from TerraAC: [2, "0880322", "Heartbeat", {}]

[ocpp  ] TRACE 2023/03/27 07:26:25 handling incoming CALL [0880322, Heartbeat] from TerraAC

[ocpp  ] TRACE 2023/03/27 07:26:25 sent CALL RESULT [0880322] for TerraAC

[ocpp  ] TRACE 2023/03/27 07:26:25 sent JSON message to TerraAC: [3,"0880322",{"currentTime":"2023-03-27T05:26:25Z"}]

[ocpp  ] DEBUG 2023/03/27 07:27:25 chargepoint disconnected: TerraAC`

Für mich sieht dass so aus, dass die TerraAC auf das schicken der currentTime glatt mit einem "disconnect" reagiert...

sreichholf, 2023-03-27:
Ich hätte ja hier noch einen verdacht.
Der Disconnect passiert hier immer eine Minute nach der letzten Nachricht.
Das Einzige was mir hier wirklich noch aufgefallen ist ist, dass evcc 54 sekunden als WebSocketPingInterval nutzt statt 60 (vermutlich aus dem occp-go Beispiel kopiert).
Ehrlich gesagt würde ich gerne mal testen wie es aussieht wenn man:

- Den Ping auf 60 stellt
- Den Ping auf 0 stellt, also deaktiviert (ob dann womöglich der Heartbeat zuverlässig kommt)

Wäre es eventuell eine Option WebSocketPingInterval zumindest vorübergehend einstellbar zu machen?
Oder ist dir/euch das einfach zu viel trara nur um die Quirks einer einzelnen Wallbox zu ermitteln?

andig, 2023-03-27:
Im Server selbst können wir nicht soviel konfigurieren:

```
// Config contains optional configuration parameters for a websocket server.
// Setting the parameter allows to define custom timeout intervals for websocket network operations.
//
// To set a custom configuration, refer to the server's SetTimeoutConfig method.
// If no configuration is passed, a default configuration is generated via the NewServerTimeoutConfig function.
type ServerTimeoutConfig struct {
	WriteWait time.Duration
	PingWait  time.Duration
}
```

Den Client könnten wir umkonfigurieren, am liebsten aber nur testhalber. Tatsächlich braucht es doch hier aber mal ein Ticket bei ABB und/oder einen parallelen Test mit Steve. Mich würde schon interessieren, ob es irgendwo eine Konstellation gibt die funktioniert?

sreichholf, 2023-03-27:
steve hab ich nicht im Angebot aber ich hab die Home-Assistant Lösung fertig eingerichtet und kann da mal gegentesten!

OnWire, 2023-03-27:
Ich hab inzwischen eine Rückmeldung von ABB:

"Im Anhang die Infos, die ich gefunden habe. Das „Heartbeat“ sollte parametriert werden können.." Dann wurde noch auf: https://www.oasis-open.org/committees/download.php/58944/ocpp-1.6.pdf verwiesen.

Ich kann jetzt nicht beurteilen ob das an der Stelle hier weiter hilft.

sreichholf, 2023-03-28:
Mit https://github.com/lbbrhzn/ocpp sehe ich die Disconnects am websocket nicht, allerdings ist der Ping dort im default auf 20s gestellt.
Allerdings ist es damit so, dass ohne das anhaken ein bestimmten Option (durch die er fest das smartcharging profil annimmt) die Wallbox gleich komplett abstürzt...

Dazu gibt's auch einen Thread in den Discussions zu der Lösung

https://github.com/lbbrhzn/ocpp/discussions/684

Ich bin ganz bei dir, dass hier eine vernünftige Antwort von ABB mehr angesagt wäre als alles andere.

Zu guter Letzt noch das zugehörge Log, das ist allerdings recht lang deswegen als gist

https://gist.github.com/sreichholf/c8701f6c53d244a884d598d8389ebf13

sreichholf, 2023-03-28:
Eine Frage habe ich da noch, würde man ein HTTP 400 auf dem websocket im Log sehen?
Wenn die HA lösung "abstürzt" sieht man davor noch einen HTTP 400.

OnWire, 2023-03-29:
Ich habe erneut bei ABB nachgehakt...

OnWire, 2023-04-17:
Leider bisher keine Antwort von ABB. :-/

Ich muss aber dazu sagen das mit 0.116.2 die Sache eigentlich ganz rund läuft. Zwar immer noch regelmäßig Disconnects der Box, aber die Kommunikation fängt sich recht schnell wieder. (auch ohne Neustart von EVCC) Leider gehen mir aufgrund dessen - so vermute ich - Ladevorgänge im Log verloren aber es gibt schlimmeres. Die Regelung selbst läuft gut.

sreichholf, 2023-04-17:
Das kann ich hier genau so bestätigen.
Auch der neuerliche MeterValues request funktioniert mit der TerraAC grundsätzlich sauber.

WurstKarate, 2023-04-26:
Guten Tag zusammen! Erst einmal vielen Dank für die detaillierte Schilderung. Das konnte mir schon viele Fragen beantworten, da ich ein identisches Verhalten in meiner Konfiguration feststellen konnte. Ich wollte an dieser Stelle noch in Erfahrung bringen, ob an der Säule "Freies Lades" aktiviert war/ist, bzw. wie eure funktionierende Konfiguration aussieht.

```
chargers:
  - name: TAC 1-2
    type: template
    template: ocpp
    stationid: TACXYZ
```

So ist das Ganze gerade bei mir konfiguriert und nach einem Ladevorgang (der dank der Überschussregelung perfekt funktioniert) legen die Logeinträge erst richtig los:

> charger out of sync: expected enabled, got disabled
charger enable: Refected

Hier hilft dann auch nur noch ein Neustart.
@OnWire dank deiner Antwort bin ich mehr gewollt diese timeouts zu ignorieren, doch fällt mir das in der GUI recht schwer.
Gibt es eine Möglichkeit die Anzeige zu blocken (der Sinn ist natürlich fragwürdig)? Habe es mal mit `log: fatal` testweise versucht, allerdings ohne Erfolg.

Vielen Dank für eure Mithilfe und noch allen einen schönen Tag!

OnWire, 2023-04-26:
Meine aktuelle Config ist folgende:

```
chargers:
- name: wallbox1
  type: ocpp
  timeout: 10m
  stationid: TACWXXXXXXX
  idtag: 1
```
Das freie laden würde ich deaktivieren, sonst fängt beim anschließen des Auto die Wallbox von alleine das laden an und als Resultat bekommst die von die geposteten Meldungen "out of sync", "charger rejected". Das liegt daran das die Ladevorgänge dann nicht vom EVCC gesteuert werden.

WurstKarate, 2023-04-26:
@OnWire Wieder einmal vielen Dank für deine schnelle Hilfe/Auskunft! gibt es eine Möglichkeit, dass die ABB ohne RFID den Ladevorgang starten kann?

OnWire, 2023-04-26:
Na wenn Sie von EVCC gesteuert wird, das ist ja der Sinn des ganzen. Ich nutze die RFID Authentifizierung auch nicht. Bei mir ist das Auto in EVCC hinterlegt und wenn angeschlossen, wird von EVCC der Ladevorgang in Eigenregie geregelt. Oder aber von Hand wenn du bspw. "schnell" bzw. "min+pv" wählst.

WurstKarate, 2023-04-26:
Vielen Dank! Das zuletzt getestete EV war nicht hinterlegt. Danke noch einmal!

OnWire, 2023-04-27:
Inzwischen mal eine Antwort von ABB, leider wieder am Thema vorbei und keine Aussage warum die Box sich disconnected. Einzige Info, der OCPP Timeout der Box ist wohl fest bei 120 Sekunden.

sreichholf, 2023-05-03:
Das würde es aber erklären.
Wenn der Timeout auf evcc seite 60 Sekunden ist, und die Box einfach harcoded 120 hat, dann wird das immer zum timeout führen.

OnWire, 2023-05-03:
> Das würde es aber erklären. Wenn der Timeout auf evcc Seite 60 Sekunden ist, und die Box einfach harcoded 120 hat, dann wird das immer zum timeout führen.

Na dann sollte es doch ausreichen auf EVCC Seite das Timeout hoch zu setzen oder den Wert parametrierbar zu machen @andig ?

andig, 2023-05-03:
Ich kenne keinen OCPP Timeout den der Server beeinflussen könnte?

andig, 2023-05-08:
Abb Support kontaktieren. Das Verhalten gibts vei anderen Boxen mit evcc nicht.

stna1981, 2023-05-08:
> Abb Support kontaktieren. Das Verhalten gibts vei anderen Boxen mit evcc nicht.

Kein Problem, wird zwar vermutlich nicht viel helfen, aber vielleicht höhlt ja steter Tropfen den Stein. Was genau soll ich denen Schreiben bzw. was soll ich "einfordern"? :-)

[14 more replies]