evcc-io/evcc #6818: MQTT Tpoic vehicleTitle wenn Gastfahrzeug ist Connected ist null -> Should be "Gastfahrzeug"

bicmac, 2023-03-13:
### Describe the bug

ich habe ein vehicle konfiguriert. Sobald ich dieses auswähle bzw es ausgwählt wird ist der Datepunkt VehicleTitle im MQTT auch richtig gesetzt.
Ist jedoch ein Gastfahrzeug connected ist der Datenpunkt null bzw. gar nicht gesetzt.
In meinen Augen sollte der DP jedoch entweder den VehicleTitle beinhalten wenn er gesetzt ist oder aber Gastfahrzeug sobald ein unbekanntes Fahrzeug connected ist (das ist ja auch der Wert den die EVCC GUI Dann anzeigt).

### Steps to reproduce

connect TESLA
check mqtt Tpoiuc-> ist tesla
Connect anderes Auto
check Mqtt Topic -> ist null :-(

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

log: info

# unique installation id
plant: 2373fb4f3fd14cc20fcc319f18cecbd88bac19075d4f9cba5957900bb17f65f3

interval: 20s # control cycle interval

sponsortoken: XXXXXX
# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
  - name: pv
    type: custom
    power: # Leistung (W)
      source: mqtt
      topic: Zaehler2/power
      scale: -1
      timeout: 30s
    energy: # Energie (kWh) - Umrechnung von Wh -> kWh
      source: mqtt
      topic: Zaehler2/total_feed
      scale: 0.001
      timeout: 30s
    currents:
      - source: mqtt
        topic: Zaehler2/power_L1
        scale: -1
      - source: mqtt
        topic: Zaehler1/power_L2
        scale: -1
      - source: mqtt
        topic: Zaehler1/power_L3
        scale: -1
  - name: grid
    type: custom
    power: # Leistung (W)
      source: mqtt
      topic: Zaehler1/power
      timeout: 30s
      #scale: -1
    energy: # Energie (kWh) - Umrechnung von Wh -> kWh
      source: mqtt
      topic: Zaehler1/total_consumption
      timeout: 30s
      scale: 0.001
    currents:
      - source: mqtt
        topic: Zaehler1/power_L1
        timeout: 30s
      - source: mqtt
        topic: Zaehler1/power_L2
        timeout: 30s
      - source: mqtt
        topic: Zaehler1/power_L3
        timeout: 30s

chargers:
  - type: template
    template: alfen
    id: 1
    host: 192.168.2.249
    port: 502
    modbus: tcpip
    name: alfen

loadpoints:
  - title: Parkplatz
    charger: alfen
    mode: off # Standardmodus zum Laden
    phases: 0 # Automatische Umschaltung zwischen 1 und 3 Phasen
    mincurrent: 6 # Mindestens 6A
    maxcurrent: 16 # Maximal 16A
    resetOnDisconnect: true
    soc:
      poll:
        mode: connected
        interval: 30m

vehicles:
  - name: Tesla
    type: custom
    title: Tesla # display name for UI
    capacity: 78 # battery capacity (kWh)
    soc:
      source: mqtt
      topic: iobrokeralias/0/Tesla/battery_level
      timeout: 24h
    status:
      source: mqtt
      topic: iobrokeralias/0/Wallbox/ConnectionStateShort
      timeout: 30s
    range:
      source: mqtt
      topic: iobrokeralias/0/Tesla/battery_range_km
      timeout: 30s

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.4477
  feedin:
    type: fixed
    price: 0.2874
  planner:
    type: grünstromindex
    zip: 91126

mqtt:
  broker: 192.168.2.37:1883
  topic: evcc
  clientid: evcc
  user: mqttuser
  password: XXXX

site:
  title: XXXX
  meters:
    grid: grid
    pvs:
      - pv
```

### Log details

```text
no log evailable
```

### What type of operating system are you running?

Linux

premultiply, 2023-03-13:
Das beisst sich mit anderen Sprachen.

Den Wert auf NULL zu setzen ist doch eigentlich richtig, oder?

bicmac, 2023-03-13:
Wieso ist das ruchtig? Es ist ja ein Fahrzeug verbunden nämlich das Gastfahrzeug. Die EVCC GUI zeig ja auch Gastfahrzeug an wenn ein unbekanntes Auto verbunden ist und nicht „NULL“

andig, 2023-03-13:
@bicmac was steht denn im Sessionlog?

andig, 2023-03-13:
Das Backend- und damit mqtt wie auch api- kennen die UI-Übersetzung von `nil` in `Gastfahrzeug` nicht. Das ist auch im Sessionlog so. Nur das UI gibt dem Ganzen einen freundlichen Namen.

@bicmac wenn Du das ändern willst müsstest Du ein Detaultfahrzeug konfigurieren dessen Name dann auch verwendet würde.

closed 2023-03-13 by andig