evcc-io/evcc #6833: Übersicht Ladevorgänge - falsche Summieren

ifeulner, 2023-03-14:
### Describe the bug

Bei der Übersicht der Ladungen wird falsch summiert. Bei Unterbrechungen im Ladevorgang wird beim Neustart anstatt dem Delta auch die vorige Ladung mitgerechnet, siehe Bild:

[image]

Der erste Ladevorgang in der Liste ist eine Fortsetzung des zweiten, da sollten also nur 1,9kWh stehen, statt 48,8 kWh. 
Der Fehler betrifft auch die weiteren in der Liste. Die Gesamtsumme stimmt dementsprechend auch nicht. 

### Steps to reproduce

1. Ladevorgang starten
2. Ladevorgang zum Zeitpunkt x beenden
3. Ladevorgang fortfahren

### Configuration details

```yaml
nicht wirklich relevant in dem Zusammenhang.
```

### Log details

```text
n/a
```

### What type of operating system are you running?

Linux

### Version

0.144.1

VolkerK62, 2023-03-14:
Also bei mir passt es. 
Version: 0.114.1 (3c6e2ca3)

[image]

ifeulner, 2023-03-15:
Ok, habe jetzt doch mal meine Config eingestellt, da es da offensichtlich Unterschiede gibt. Bei mir sind das auch durch einen Fehler / fremdverschuldeten Abbruch neu begonnene Ladevorgänge. Allerdings hatte ich nichts im Logfile gesehen, werde da nochmal drauf achten...

andig, 2023-03-15:
Bitte Logfile mit 

    --log trace

Dann sehen wir welche Werte wir haben und was in die DB geschrieben wird.

Wishmaster83, 2023-03-18:
Hallo,
ich möchte mich diesem Problem anschließen.
Bei mir ist es so, dass der Wert "sensor.evcc_charged_energy_kwh" (in Home Assistant eingebunden und über "state_topic: "evcc/loadpoints/1/chargedEnergy"" abgerufen) sehr regelmäßig Nachts gegen 4 Uhr kurz den Wert 0 annimmt und dann wieder auf den ursprünglichen Wert wechselt. Nach dem Wechsel wird der der ursprüngliche Wert jedoch als neuer Ladevorgang betrachtet und wie beim OP in voller Höhe in die Statistik hinzuaddiert:

[image]
[image]

Das zerschießt mir im HomeAssistant alle Statistiken und in evcc den Log über die Ladevorgänge. :(

closed 2023-03-18 by andig

ifeulner, 2023-03-18:
Inwiefern completed? Ich versuche noch sinnvolle traces zu bekommen, aber habe das Problem definitiv. Es gibt zwischendrin Fehler in der Kommunikation mit dem wallconnector, da sehe ich das Problem.. kann man irgendwie das ganze Log laden, bei mir läuft evcc als HA Addon.

andig, 2023-03-18:
Issues bitte mit Logfile

andig, 2023-03-19:
Ich möchte nochmal erwähnen: wir nehmen solche Fehler absolut ernst, ohne Logfile sind sie aber nicht zu beheben :(

Wishmaster83, 2023-03-19:
> Ich möchte nochmal erwähnen: wir nehmen solche Fehler absolut ernst, ohne Logfile sind sie aber nicht zu beheben :(

Ich würde das Logfile sehr gerne bereit stellen, weiß aber nicht, wie ich an dieses über HomeAssistant ran komme.
Der Zugriff per SSH Terminal funktioniert nicht, da es ja als ein Addon installiert ist.

Ich weiß, dass evcc als HA Addon nicht vom Core Team gepflegt/supported wird. Hast Du dennoch eine Idee, wie ich an das LogFile gelange? Oder wen ich diesbezüglich fragen könnte? Tante Google konnte mir bei dieser Frage bislang nicht weiterhelfen...

ifeulner, 2023-03-19:
> Ich möchte nochmal erwähnen: wir nehmen solche Fehler absolut ernst, ohne Logfile sind sie aber nicht zu beheben :(

Klar, kann ich verstehen - werde morgen hoffentlich das Problem wieder sehen und das Logfile zur Verfügung stellen - allerdings versteh ich nicht, warum das Issue nach 4 Tagen geschlossen wird, auch noch mit dem missverständlichen Status "Completed"...

andig, 2023-03-20:
> allerdings versteh ich nicht, warum das Issue nach 4 (!) Tagen geschlossen wird

Ohne Logfile gibts für uns nix zu tun, daher geschlossen. Sobald es eins gibt können wir gerne wieder auf machen. Die Diskussion hilft aber- im Gegensatz zu Logfiles- auch nicht 😉.

"Completed" bei Github ist vmtl. alles was nicht offen ist.