evcc-io/evcc #6842: Schaltbare Steckdose als Charger; `mincurrent: 0.x` verhindert `charger enable`

VolkerK62, 2023-03-15:
### Describe the bug

Wenn man bei schaltbaren Steckdosen `mincurrent` <1 einstellt, führt das beim Einschalten zu einem `max charge current: 0A` und die Steckdose wird nicht eingeschaltet.

### Steps to reproduce

1. `mincurrent: 1` einstellen; nach Ablauf von `enable.delay` wird `max charge current: 1A` und `charger enable´ gesendet
2. `mincurrent: 0.x` einstellen; nach Ablauf von `enable.delay` wird `max charge current: 0A` und KEIN `charger enable´ gesendet
3.
...

### Configuration details

```yaml
.
```

### Log details

```text
mincurrent: 0.3

Mär 15 11:54:32 raspberrypi evcc[22247]: [lp-2  ] DEBUG 2023/03/15 11:54:32 set charge mode: minpv
Mär 15 11:54:32 raspberrypi evcc[22247]: [site  ] DEBUG 2023/03/15 11:54:32 ----
Mär 15 11:54:32 raspberrypi evcc[22247]: [lp-2  ] DEBUG 2023/03/15 11:54:32 charge power: 0W
Mär 15 11:54:32 raspberrypi evcc[22247]: [site  ] DEBUG 2023/03/15 11:54:32 pv power: 612W
Mär 15 11:54:32 raspberrypi evcc[22247]: [site  ] DEBUG 2023/03/15 11:54:32 battery soc: 21%
Mär 15 11:54:32 raspberrypi evcc[22247]: [site  ] DEBUG 2023/03/15 11:54:32 battery power: -212W
Mär 15 11:54:32 raspberrypi evcc[22247]: [site  ] DEBUG 2023/03/15 11:54:32 grid power: -26W
Mär 15 11:54:32 raspberrypi evcc[22247]: [site  ] DEBUG 2023/03/15 11:54:32 grid currents: [1.06 -1.06 0.98]A
Mär 15 11:54:32 raspberrypi evcc[22247]: [site  ] DEBUG 2023/03/15 11:54:32 site power: 12W
Mär 15 11:54:33 raspberrypi evcc[22247]: [lp-2  ] DEBUG 2023/03/15 11:54:33 charger status: B
Mär 15 11:54:33 raspberrypi evcc[22247]: [lp-2  ] DEBUG 2023/03/15 11:54:33 pv charge current: 0A = 0A + -0.0511A (12W @ 1p)
Mär 15 11:54:33 raspberrypi evcc[22247]: [lp-2  ] DEBUG 2023/03/15 11:54:33 max charge current: 0A
Mär 15 11:54:39 raspberrypi evcc[22247]: [site  ] DEBUG 2023/03/15 11:54:39 ----

.
mincurrent: 1

Mär 15 11:51:27 raspberrypi evcc[22161]: [lp-2  ] DEBUG 2023/03/15 11:51:27 set charge mode: minpv
Mär 15 11:51:27 raspberrypi evcc[22161]: [site  ] DEBUG 2023/03/15 11:51:27 ----
Mär 15 11:51:27 raspberrypi evcc[22161]: [lp-2  ] DEBUG 2023/03/15 11:51:27 charge power: 0W
Mär 15 11:51:27 raspberrypi evcc[22161]: [site  ] DEBUG 2023/03/15 11:51:27 pv power: 473W
Mär 15 11:51:27 raspberrypi evcc[22161]: [site  ] DEBUG 2023/03/15 11:51:27 battery soc: 21%
Mär 15 11:51:27 raspberrypi evcc[22161]: [site  ] DEBUG 2023/03/15 11:51:27 battery power: 149W
Mär 15 11:51:27 raspberrypi evcc[22161]: [site  ] DEBUG 2023/03/15 11:51:27 grid power: -32W
Mär 15 11:51:28 raspberrypi evcc[22161]: [site  ] DEBUG 2023/03/15 11:51:28 grid currents: [-0.7 -1.06 1.03]A
Mär 15 11:51:28 raspberrypi evcc[22161]: [site  ] DEBUG 2023/03/15 11:51:28 site power: 512W
Mär 15 11:51:28 raspberrypi evcc[22161]: [lp-2  ] DEBUG 2023/03/15 11:51:28 charger status: B
Mär 15 11:51:28 raspberrypi evcc[22161]: [lp-2  ] DEBUG 2023/03/15 11:51:28 pv charge current: 0A = 0A + -2.23A (512W @ 1p)
Mär 15 11:51:28 raspberrypi evcc[22161]: [lp-2  ] DEBUG 2023/03/15 11:51:28 max charge current: 1A
Mär 15 11:51:28 raspberrypi evcc[22161]: [lp-2  ] DEBUG 2023/03/15 11:51:28 charger enable
Mär 15 11:51:28 raspberrypi evcc[22161]: [lp-2  ] DEBUG 2023/03/15 11:51:28 wake-up timer: start
Mär 15 11:51:36 raspberrypi evcc[22161]: [site  ] DEBUG 2023/03/15 11:51:36 ----
```

### What type of operating system are you running?

Linux

### Version

0.114.1 (3c6e2ca3)

andig, 2023-03-18:
Der Shelly- wie alle schaltbaren Steckdosen- kann Ströme nicht steuern. Deshalb gibt es eine fake Schnittstelle:

```go
// MaxCurrent implements the api.Charger interface
func (c *Shelly) MaxCurrent(current int64) error {
	return nil
}
``` 

Die kann nur `int64`, kein `float64`. Damit ist 0.3 äquivalent zu 0. Sollen wir den Steckdosen `float64` erlauben?

VolkerK62, 2023-03-18:
ist wahrscheinlich die einfachste Lösung

closed 2023-03-18 by andig