evcc-io/evcc #6848: Innogy Ebox always shows a consumption of 30W on standby

SamuelBrucksch, 2023-03-15:
### Describe the bug

Hi,

the eBox always has a low standby consumption of 30W according to the internal meter. that is wrong, so i guess it is some kind of measuring inaccuracy.

Is it possible to define an offset, which is not counted for the statistics? For example everything < 50W is ignored.

Thanks
Samuel

### Steps to reproduce

Just connect the eBox

### Configuration details

```yaml
chargers:
  - name: ebox
    type: template
    template: innogy-ebox
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.0.99 # Hostname
    port: 502 # Port 
```
```

### Log details

```text
none
```

### What type of operating system are you running?

Linux

### Version

0.114.1

andig, 2023-03-16:
> that is wrong

Are you sure? We can discuss not showing standby consumption in the UI if vehicle is not charging but we do need to count all powers for showing sums etc.

/cc @premultiply @naltatis

SamuelBrucksch, 2023-03-16:
Yes i'm sure that the value is wrong. I already had it in openhab before and it was the same, so not EVCC specific. Some of my colleagues have the same WB and they confirmed with an external extra meter, that it does not have a standby consumption that is as high as the reading from the WB.

premultiply, 2023-03-16:
The reason ist that the device lacks real power measurement.
The power is kind of fake by the sum of the phase currents multiplied by 230. So it is roughly the apparent power, not the active power.

SamuelBrucksch, 2023-03-16:
Is this also the case for the eBox Professional with the included meter which is eichrechtskonform? Or is that only relevant for the reported amount of charge kWh?

cseelmann, 2023-03-16:
I also have an Innogy ebox prof and noticed that issue as well. Could that be a scaling issue? Is it probably 4 Watts instead of 40 and so on?

SamuelBrucksch, 2023-03-16:
In the demo yaml you use javascript to generate random data:
https://github.com/evcc-io/evcc/blob/6eed77351b75a3e5ce0e9e82e85f7549ce147845/cmd/demo.yaml#L30-L40

Could that also be used somehow, to pre-filter the current values? Then I could use that as a workaround.

andig, 2023-03-16:
> I also have an Innogy ebox prof and noticed that issue as well. Could that be a scaling issue? Is it probably 4 Watts instead of 40 and so on?

Is the power correct when charging? That should answer it.

SamuelBrucksch, 2023-03-16:
It's really just a matter of inaccurate low values.

cseelmann, 2023-03-16:
> Is the power correct when charging? That should answer it.

Yes it is. I was thinking maybe there is some kind of scale factor involved like in sunspec. 

When I first saw the high consumption without a car connected I was thinking that this is the price for asking the box how high the current consumption is. In other words, do requests via modbus lead to the box leaving standby and therefore consume more? 

One way would then be to filter these values out, but a better solution would be to lower the intervals in that specific case to reduce power consumption which adds to large values over time. Anyone measured with external meter how the consumption with modbus and evcc enabled is and how it is without?

premultiply, 2023-03-16:
https://github.com/evcc-io/evcc/blob/6eed77351b75a3e5ce0e9e82e85f7549ce147845/charger/innogy.go#L169-L173

andig, 2023-03-17:
@premultiply wenn wir das im Loadpoint lösen machen wir es Statusabhängig.

SamuelBrucksch, 2023-03-19:
Here a sample how much it affects the statistics:
[image]

Yesterday morning it was still 99% after charging.

Please let me know if there is anything ready to test.

andig, 2023-04-14:
Closed in https://github.com/evcc-io/evcc/commit/be30fa4c1fbb00e2a414a7146a5bdd58efc88878

closed 2023-04-14 by andig