evcc-io/evcc #6855: Fiat: Fahrzeug wird nicht mehr erkannt, "unavailable"

Ham60elu, 2023-03-16:
### Describe the bug

HA Integration:
mein Fiat 500e wird seit dem Update auf 0.113.2 (und auch bei den weiteren Versionen auf die aktuelle 0.114.1) nur noch als "unavailable" angezeigt. Die Anmeldung in der Fiat-App funktioniert mit den eingetragenen Credentials.
Mit dem Update auf die 0.113.2 funtionierten bei mir auch die Einträge in der secrets.yaml (sponsortoken, user, password, vim und pin) für evcc nicht mehr (=evcc startete nicht mehr). Erst nachdem ich diese wieder orginal in die evcc.yaml eingetragen habe, funktionierte evcc wieder.

Die Fehlermeldung lautet bei mir:
"vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'fiat': login failed: Invalid LoginID: invalid loginID or password"

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
vehicles:
  - name: fiat500e
    type: template
    template: fiat
    title: Fiat 500e Cabrio  
    user: *username* #!secret fiat_user  
    password: *password* #!secret fiat_password  
    vin: *vinnumber* #!secret fiat_vin  
    capacity: 42
    pin: *pin* #!secret fiat_pin 
#    mode: pv # enable PV-charging when vehicle is identified
#    minSoC: 20 # immediately charge to 0% regardless of mode unless "off" (disabled)
#    targetSoC: 80 # limit charge to 90%
```

### Log details

```text
"vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'fiat': login failed: Invalid LoginID: invalid loginID or password"
```

### What type of operating system are you running?

Docker container

### Version

0.114.1

andig, 2023-03-16:
> login failed: Invalid LoginID: invalid loginID or password"

closed 2023-03-16 by andig

Ham60elu, 2023-03-16:
> > login failed: Invalid LoginID: invalid loginID or password"

Hallo andig,
vielleicht habe ich mich falsch ausgedrückt: 
An den Login-Daten hat sich nichts geändert und diese funktionieren auch nach wie vor, sowohl auf der Fiat-Webseite als auch in der Fiat-App. Einzige Veränderung war der Versionswechsel in evcc. 
Werden diese Daten nun anders übertragen? 
Danke für einen kurzen Hinweis.

andig, 2023-03-16:
Am Fiat Template hat sich auch nix geändert, wohl aber an der Templateverarbeitung. Falls es hier einen Fehler gibt würde ich den sehr gerne lösen, dafür bräuchte ich aber Zugangsdaten mit denen ich es reproduzieren kann. Wenn ihr helfen wollt dann bitte per Mail an info@evcc.io oder auf Slack.