evcc-io/evcc #6859: Laden Startet ohne vollständiges update im UI | Daheimladen-mb

ExecuteGK, 2023-03-16:
### Describe the bug

Das Fahrzeug ist in EVCC so konfiguriert, das die Ladung nach Fahrzeugerkennung im Modus minpv beginnen sollte.

Dies funktioniert perfect mit einer Phoenix Contact Wallbox und dem Daheimladen API model.
Nach umstellung auf Daheimladen modbus vehaellt sich das Laden anders. Das Fahrzeug wir korrekt erkannt und die Ladung beginnt mit 4KW (inkl Anzeige auf der Webseite). Allerdings wir auf der Webseite immer noch "Ladebereit. Warte auf Fahrzeug" angezeigt.

Nach einem kurzen Wechsel auf modus: sofort und zurück auf minpv, ist die Anzeige korrekt.

[image]
[image]
[image]

### Steps to reproduce

1. Fahrzeug mit modus minpv konfigurieren (nicht den LP)
2. Fahrzeug an den LP haengen
3. Anezige beobachten
4. LP auf sofort stellen
5. LP wieder auf minpv stellen

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: error
levels:
  cache: error

interval: 10s # control cycle interval

sponsortoken: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
# mqtt message broker
mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable
  # user:
  # password:

# Stromtarif
#
tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.2242 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.08 # [currency]/kWh

meters:
- type: template
  template: e3dc
  usage: grid
  host: 192.168.7.50
  port: 502
  name: grid1
- type: template
  template: e3dc
  usage: pv
  host: 192.168.7.50
  port: 502
  name: pv2
- type: template
  template: e3dc
  usage: battery
  host: 192.168.7.50
  port: 502
  name: battery3

chargers:
- type: template
  template: phoenix-ev-eth
  host: 192.168.7.8
  port: 502
  name: wallbox5
- name: daheimladen
  type: template
  template: daheimladen-mb
  host: 192.168.7.51

vehicles:
- type: template
  template: id
  title: ID.4
  user: xxxxxx
  password: xxxx
  vin: xxxx
 capacity: 77
  phases: 3
  mode: minpv
  targetSoC: 80
  minCurrent: 6
  maxCurrent: 16
  cloud: false
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox5
  #  vehicles:
  #  - ev4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

- title: Vorne
  charger: daheimladen
  #  vehicles:
  #  - ev4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Home
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery3
```

### Log details

```text
no debug log availabe yet. (car currently out)
```

### What type of operating system are you running?

Linux

### Version

v0.114.1

ExecuteGK, 2023-03-16:
Ich muss gestehen das ich ein wenig den Überblick verloren hab was denn nun gerade die richtige Firmware für mich ist.... Beobachten wir mal was nach Freitag passiert.GrüßeSebastianVon meinem/meiner Galaxy gesendet

Du hast die passende Firmware drauf?

andig, 2023-03-16:
Missing logfile

closed 2023-03-16 by andig