evcc-io/evcc #6874: Fahrzeug wird nicht erkannt, wenn der Wechselrichter aus ist

proxymus, 2023-03-16:
### Describe the bug

Ich habe folgende Komponenten:
Wechselrichter: Huawei SUN2000-15KTL-M0 mit SDongleA-05
Power Sensor: DTSU666-H
Wallbox: go-e HOMEfix 11 kW v3
Auto: Hyundai Kona-e
Kein Speicher

Mein Wechselrichter geht kurz vor Sonnenaufgang an. Ab da kann evcc die Werte des Wechselrichters und des Power Sensors abfragen. Da ich keinen Speicher habe, schaltet sich der Wechselrichter kurz nach Sonnenuntergang ab. Dieses Verhalten lässt sich nicht ändern (wurde mir auf Nachfrage von Huawei bestätigt). Tagsüber wird das Auto nach dem Anstecken erkannt. Die Lademodi funktionieren wie erwartet.

### Problem:
Das Auto wird von evcc nicht erkannt, wenn es nach Sonnenuntergang an die Wallbox angesteckt wird. Ich kann es dann nicht über evcc laden. Auch dann nicht, wenn ich den Modus auf **Schnell** umschalte. 
Frage ich die Wallbox per evcc ab, wird der Identifier korrekt ausgelesen.

Es wäre schön, wenn ich auch ohne aktiven Wechselrichter in der Nacht per **Schnell** (ggf. bis zum Ladeziel) laden könnte. Rein logisch müsste auch **PV+Min** funktionieren und dann 1-phasig/3-phasig mit 6A laden (je nach Einstellung im evcc). PV-Anteil ist in dem Fall 0.
Starte ich den Ladevorgang über die Wallbox, werden die Werte korrekt im Log von evcc angezeigt (siehe unten am Ende des Logs).

Vielen Dank für eure Unterstützung.

### Steps to reproduce

1. Auto nach Sonnenuntergang anstecken
2. evcc auf Modus **Schnell** umschalten

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

sponsortoken: <token>

telemetry: false 

log: info
levels:
  cache: error

# unique installation id
plant: <id>

interval: 5s # control cycle interval

meters:
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.178.60
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.178.60
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.178.60
  port: 502
  usage: battery
[... 255 lines cut ...]
[lp-2  ] DEBUG 2023/03/16 21:13:29 charge power: 0W
[site  ] ERROR 2023/03/16 21:13:38 pv 1 power: read failed: dial tcp 192.168.178.60:502: connect: no route to host
[site  ] DEBUG 2023/03/16 21:13:38 pv power: 0W
[site  ] ERROR 2023/03/16 21:13:47 grid meter: read failed: dial tcp 192.168.178.60:502: connect: no route to host
[site  ] DEBUG 2023/03/16 21:13:47 ----
[lp-1  ] DEBUG 2023/03/16 21:13:47 charge power: 3790W
[lp-2  ] DEBUG 2023/03/16 21:13:47 charge power: 0W
[site  ] ERROR 2023/03/16 21:13:56 pv 1 power: read failed: dial tcp 192.168.178.60:502: connect: no route to host
[site  ] DEBUG 2023/03/16 21:13:56 pv power: 0W
[site  ] ERROR 2023/03/16 21:14:05 grid meter: read failed: dial tcp 192.168.178.60:502: connect: no route to host
[site  ] DEBUG 2023/03/16 21:14:05 ----
[site  ] DEBUG 2023/03/16 21:14:05 telemetry: charge: Δ0/19Wh @ 3790W
[lp-1  ] DEBUG 2023/03/16 21:14:05 charge power: 3760W
[lp-2  ] DEBUG 2023/03/16 21:14:05 charge power: 0W
[site  ] ERROR 2023/03/16 21:14:14 pv 1 power: read failed: dial tcp 192.168.178.60:502: connect: no route to host
[site  ] DEBUG 2023/03/16 21:14:14 pv power: 0W
```

### What type of operating system are you running?

Linux

### Version

0.114.1

proxymus, 2023-03-16:
Dass kein Netzbezug ermittelt werden kann, ist klar. Das Auto müsste aber doch trotzdem angezeigt werden. Immerhin wird es im Log richtig angezeigt. Für den Lademodus **Schnell** dürfte der Netzbezug doch keine Rolle spielen (eventuell auch bei **PV+Min**, wenn ich davon ausgehe auf kleinster Stufe ohne PV zu laden) oder sehe ich das falsch?

Was das Abschalten des Wechselrichters angeht, hatte ich ein Ticket bei Huawei dazu. Bei dieser Serie lässt sich das Verhalten nicht ändern (es kann auch kein DC-Speicher nachgerüstet werden).

premultiply, 2023-03-16:
> Für den Lademodus Schnell dürfte der Netzbezug doch keine Rolle spielen (eventuell auch bei PV+Min, wenn ich davon ausgehe auf kleinster Stufe ohne PV zu laden) oder sehe ich das falsch?

Das kann man so pauschal leider nicht sagen.
Spätestens wenn eine Lastmanagement aktiv ist wäre es u.U. fatal ohne aktuelle Messwerte einfach Ladevorgänge freizugeben.

Ich weiß leider keine gute, einfache Lösung für dieses Problem.

Im Notfall mit separater Messtechnik überbauen.

andig, 2023-03-17:
> Für den Lademodus Schnell dürfte der Netzbezug doch keine Rolle spielen

Das stimmt, aktuell ist es aber so nicht umgesetzt- falls Netzzähler nicht funktioniert gehts nicht weiter.

> Spätestens wenn eine Lastmanagement aktiv ist wäre es u.U. fatal ohne aktuelle Messwerte einfach Ladevorgänge freizugeben.

Da hat @premultiply recht- unter dem Gesichtspunkt würden wir das auch nicht ändern. Du müsstest also das Problem der Netzmessung lösen. Ggf. kannst Du auch eigene Messwerte über MQTT etc zur Verfügung stellen- da könntest Du dann auch Ausfälle mit Ersatzwerten kompensieren.

An dieser Stelle `wontfix`.

proxymus, 2023-03-18:
Ok schade. Ich habe schon überlegt, ob ich die Werte vom Power Sensor parallel abgreifen kann. Dieser ist ja dauerhaft an. Ich weiß aber nicht, ob das zu Problemen führen würde und was ein sinnvoller Weg dafür wäre.
Ansonsten müsste ich vermutlich einen zweiten Zähler installieren, um alle Ladevorgänge im evcc steuern zu können (im Moment habe ich dafür 2 Stellen). Wäre es dann ein Problem, wenn Nachts keine Werte vom Wechselrichter kommen, weil dieser aus ist?
Vielen Dank für eure Unterstützung und die Entwicklung dieser Software.

andig, 2023-03-18:
Probier das mit dem Sensor doch einfach mal aus? Fehler vom WR loggen wir, die werden aber sonst ignoriert- die Steuerung erfolgt ja nur über den Netzzähler.