evcc-io/evcc #6888: Ladestand wird auf Ladeziel gesetzt, obwohl niedriger.

HotteX, 2023-03-17:
### Describe the bug

Das Problem tritt seit einigen Wochen auf, ist aber nicht "schon immer" da. 

Das System setzt den Ladestand des Autos (eventuell immer nach einem Neustart von EVCC...da bin ich nicht sicher) auf das Ladeziel. Die Restkilometer sind allerdings korrekt ausgelesen.
``
Die Kommunikation mit dem Auto funktioniert via evcc vehicle und meldet den richtigen Ladestand.

Wenn man manuell das Ladeziel erhöht, startet die Ladung, korrigiert aber nicht den Ladestand sondern rechnet ab Ladeziel dazu, sprich: bei 59%/90% ist es dann später 61%/92%

`
^Croot@wb-mgr:/home/j# evcc vehicle
[main  ] INFO 2023/03/17 12:56:19 evcc 0.114.1
[main  ] INFO 2023/03/17 12:56:19 using config file: /etc/evcc.yaml
Soc:           60%
Capacity:      82.0kWh
Charge status: B
Charged:       25.3kWh
Range:         304km
Odometer:      51715km
Finish time:   2023-03-17 12:56:00 +0100 CET
Position:      50.222189,8.880211
Target Soc:    92%
OnIdentified:  Mode:pv, MinCurrent:6, MaxCurrent:16, MinSoc:1, TargetSoc:90
Features:      []
`

### Steps to reproduce

1. Auto anschließen
2. evcc Neustarten

### Configuration details

```yaml
vehicles:
- name: ev1
  type: tesla
  title: Bistr-O-Math
  tokens:
    access: *
    refresh: *
  vin: *
  capacity: 82
  onIdentify: # set defaults when vehicle is identified
    mode: pv
    minSoC: 1
    targetSoC: 90
    mincurrent: 6
    maxcurrent: 16
  phases: 3
  cloud: false
```

### Log details

```text
[lp-1  ] DEBUG 2023/03/17 12:55:28 charge power: 0W
[site  ] DEBUG 2023/03/17 12:55:28 pv power: 8426W
[site  ] DEBUG 2023/03/17 12:55:28 battery soc: 87%
[site  ] DEBUG 2023/03/17 12:55:28 battery power: -1770W
[site  ] DEBUG 2023/03/17 12:55:28 grid power: -6050W
[site  ] DEBUG 2023/03/17 12:55:28 grid powers: [-592 -2697 -2761]W
[site  ] DEBUG 2023/03/17 12:55:28 grid currents: [-2.74 -11.4 -11.7]A
[site  ] DEBUG 2023/03/17 12:55:28 site power: -7720W
[lp-1  ] DEBUG 2023/03/17 12:55:28 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/03/17 12:55:28 charger status: B
[lp-1  ] DEBUG 2023/03/17 12:55:28 next soc poll remaining time: 59m54s
[lp-1  ] DEBUG 2023/03/17 12:55:28 targetSoc reached: 91.8% > 90%
```

### What type of operating system are you running?

Linux

### Version

114.1

andig, 2023-03-17:
> Wenn man manuell das Ladeziel erhöht, startet die Ladung, korrigiert aber nicht den Ladestand sondern rechnet ab Ladeziel dazu, sprich: bei 59% (echt) /90% (angezeigt) ist es dann später 61%/92%

Was heisst das? Wo sieht man das im Logfile?

HotteX, 2023-03-17:
Damit meinte ich: Wenn ich auf der Oberfläche den Schieber für das Ladeziel auf 100% stelle, startet die Ladung, der Ladestand wird aber nur erhöht auf Basis der vorher angezeigten 90%, also dann 91, 92... er wird nicht korrigiert auf den richtigen Wert des Autos.  

Man sieht es hier nicht im Vorgang, lediglich als Ergebnis: (Start der ersten Ladung waren 90% angezeigt und 59% realer Ladestand), dann Neustart von evcc, daher Ladeziel wieder 90%, aber Ladestand 91,8% (aber 60% real, bzw. knapp vor 61%...es ist kurz danach umgesprungen, daher oben mein Kommentar mit 61/92)

"targetSoc reached: 91.8%"
während evcc vehicle sagt: "Soc:           60%"

Es entspricht nicht genau den Werten auf der Oberfläche, vermutlich wegen Rundung.

andig, 2023-03-17:
In anderen Worten: evcc zeigt einen anderen Soc als das Fahrzeug hat? Auch nach Neustart? Bitte Logfile und in Abgrenzung dazu

    evcc vehicle