evcc-io/evcc #6893: Automatische neu Authorisierung für VW nach Status 400

ExecuteGK, 2023-03-17:
Nach einiger Laufzeit kann es passieren, das die Verbindung zur Volkswagen API nicht mehr richtig funktioniert und die API folgendes zurück meldet: 

vehicle soc: Get "https://emea.bff.cariad.digital/vehicle/v1/vehicles/VINXXXXXXXXXXX/selectivestatus?jobs=all": unexpected status: 400

Dies passiert, wenn der Auth header sich aendert, oder verloren geht. Ein Neustart von EVCC löst das Problem.

Feature request: Wenn der Status 400 nachhaltig empfangen wird, dann wird die Authentifizierung mit der Volkswagen API wie beim Neustart neu durchgeführt.

Grüße

Sebastian

andig, 2023-03-17:
> Dies passiert, wenn der Auth header sich aendert, oder verloren geht

Was heisst das? Wo geht da was verloren? Nach welcher Zeit? Wie kann ich das reproduzieren?

Ich mach das Ticket wieder auf wenn ein aussagelräftiges Logfile vorliegt.

ExecuteGK, 2023-03-17:
Das ist eine gute Frage wo das verloren geht. Ich sehe das nicht im Log. erst wenn der Wagen wieder angeschlossen wird, kommt plötzlich nur noch status 400. Beim start von evcc ist immer alles OK.

ExecuteGK, 2023-03-17:
Beispiel vom letzten start:

Mar 15 18:50:44 flowcontrol systemd[1]: Started EV Charce Control evcc.io.
Mar 15 18:50:44 flowcontrol evcc[2625]: [main  ] INFO 2023/03/15 18:50:44 evcc 0.114.1
Mar 15 18:50:44 flowcontrol evcc[2625]: [main  ] INFO 2023/03/15 18:50:44 using config file: /etc/evcc.yaml

Dann Ansluss des Wagens 18 Stunden später:

Mar 16 13:23:42 flowcontrol evcc[2625]: [main  ] ERROR 2023/03/16 13:23:42 vehicle status: Get "https://emea.bff.cariad.digital/vehicle/v1/vehicles/WVGZZZE2ZNP019742/selectivestatus?jobs=all": unexpected status: 400

andig, 2023-03-17:
> Ich mach das Ticket gerne wieder auf wenn ein aussagelräftiges Logfile vorliegt- ohne ist da nix zu machen.

zivillian, 2023-03-20:
Ich hab hier ein Log der letzten 14 Tage, aber ich bin nicht sicher ob es hilft. Der Suchbegriff wäre `unexpected status: 400`
[evcc.log.gz](https://github.com/evcc-io/evcc/files/11022034/evcc.log.gz)

andig, 2023-03-20:
Leider nicht wirklich. Dein evcc lief 7.-20. Ab dem 15. treten die Fehler auf, natürlich nur solange das Auto dran steckt. Vorher ging es. Was sich da zwischen durch geändert hat oder anders zum Neustart ist ist leider nicht zu erkennen :(

ExecuteGK, 2023-03-20:
In den Logfiles steht auch nicht wirklich viel dazu drin. Vielleicht ist das ja auch kein evcc Problem und das Backend bei VW verliert das auth Token.Auf jeden Fall würde eine automatische neu Anmeldung nach Status 400 das Problem lösen ohne das evcc von Hand neu gestartet werden muss.GrüßeSebastianVon meinem/meiner Galaxy gesendet