evcc-io/evcc #6906: OCPP auf Orbis Viaris Uni

ollyins, 2023-03-17:
Konfiguration in der WB:

 ```
URL: ws://ip_von_evcc/seriennummer
Port: 8887
``` 

Seriennummer = stationid

```
chargers:
  - name: "EVVC3Cxxxxxx"
    type: "ocpp"
    stationid: "EVVC3Cxxxxxx"
    idtag: 1
    bootnotification: true  # wahrscheinlich überflüssig
```

andig, 2023-03-17:
Sehr cool- das verpacken wir in ein Template, danke!

andig, 2023-03-18:
@ollyins könntest Du bitte nochmal probieren, ob es auch ohne `bootnotification` geht? Die Klarheit wäre wichtig.

ollyins, 2023-03-18:
Also ohne  bootnotification bekomme ich:
[image]
Also es muss wohl sein! Danach nach Neustart wieder alles ok mit bootnotification=true

closed 2023-03-18 by andig