evcc-io/evcc #6907: Shelly negative if consumption from net

tLupus1978, 2023-03-17:
### Describe the bug

I have a negative shelly value (e.g. -300W) if I consume from net.
Means my house is negative - if my house positive, I got positive values.

But I cannot specify this within the meter settings - therefore sadly evcc is not usable for me at the moment.

Is it possible to invert used values somewhere?
It is not possible within shelly. And I do not want to change the shelly sensors. 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
meters:
  - name: housemeter
    type: template
    template: shelly-3em
    usage: grid
    host: 192.168.78.60 # IP-Adresse oder Hostname
```

### Log details

```text
[site  ] DEBUG 2023/03/17 23:33:17 ----
[lp-1  ] DEBUG 2023/03/17 23:33:17 charge power: 4W
[site  ] DEBUG 2023/03/17 23:33:17 grid power: -306W
[site  ] DEBUG 2023/03/17 23:33:17 grid powers: [-146 -124 -36]W
[site  ] DEBUG 2023/03/17 23:33:18 grid currents: [-0.77 -0.83 -0.39]A
[site  ] DEBUG 2023/03/17 23:33:18 pv power: 310W
[site  ] DEBUG 2023/03/17 23:33:18 site power: -306W
```

### What type of operating system are you running?

Docker container

### Version

0.114.1

thierolm, 2023-03-17:
Are U sure U installed the shelly correctly? Its important to install the sensores in the correct direction!
See https://www.youtube.com/watch?v=OZ7dN7ccjwc&ab_channel=Technikdiebegeistert

To change the power values U can test  and use the following custom configuration, which is generated by evcc based on your template config. I added the` scale: -1` parameter fpr all of the power values:

```yaml
meters:
- name: housemeter
  type: custom
  power:
    source: http
    uri: http://192.168.78.60/status
    jq: .emeters | map(.power) | add
    scale: -1
  energy:
    source: http
    uri: http://192.168.78.60/status
    jq: .emeters | map(.total) | add
    scale: 0.001
  currents:
  - source: http
    uri: http://192.168.78.60/emeter/0
    jq: .current
  - source: http
    uri: http://192.168.78.60/emeter/1
    jq: .current
  - source: http
    uri: http://192.168.78.60/emeter/2
    jq: .current
  voltages:
  - source: http
    uri: http://192.168.78.60/emeter/0
    jq: .voltage
  - source: http
    uri: http://192.168.78.60/emeter/1
    jq: .voltage
  - source: http
    uri: http://192.168.78.60/emeter/2
    jq: .voltage
  powers:
  - source: http
    uri: http://192.168.78.60/emeter/0
    jq: .power
    scale: -1
  - source: http
    uri: http://192.168.78.60/emeter/1
    jq: .power
    scale: -1
  - source: http
    uri: http://192.168.78.60/emeter/2
    jq: .power
    scale: -1
```

tLupus1978, 2023-03-18:
thanks for your fast answer.
> Are U sure U installed the shelly correctly? Its important to install the sensores in the correct direction!
I must admit I am not - but it is giving me good results and it works for me as expected. It is just the viewpoint - from grid (your house needs 300Watts) or from the house (the house is -300 watts low of energy) - but this is a philosophical question for me.

Regarding the solution you provided: perfect, seems to work perfectly!
Thanks a lot!!

closed 2023-03-18 by tLupus1978