evcc-io/evcc #6908: Einfacher Passwortschutz für die Weboberfläche

therealmightyx, 2023-03-18:
Die Weboberfläche von evcc sollte unbedingt mit einem Passwortschutz versehen werden. Wenn evcc auf einem Server läuft, der auch über das Internet erreichbar ist (was bei vielen der Fall sein dürfte, weil diese Art Server oft auch für Home Automation usw. genutzt werden), kann aktuell jeder auf evcc zugreifen. 
Gerade bei unerfahrenen Usern sehe ich hierdurch hohe Sicherheitsrisiken.

Ich weiß, dass es schon einige Vorschläge mit VPN, Reverse Proxy usw. gibt, die sind aber  meines Erachtens für durchschnittlich versierte User nicht umsetzbar.
Ein einfacher Passwortschutz würde die Sicherheit von evcc erheblich verbessern.

premultiply, 2023-03-18:
Dafür gibt es ausreichend andere Lösungen.

closed 2023-03-18 by premultiply

therealmightyx, 2023-03-18:
Mag sein, aber nirgends in der evcc Dokumentation wird auf das Risiko hingewiesen oder eine Lösung aufgezeigt.
Das ist definitiv ein Sicherheitsrisiko!

premultiply, 2023-03-18:
An der Doku kann jede(r) mitschreiben.