evcc-io/evcc #6910: Planner: continue in PV mode when planner finishes

andig, 2023-03-18:
From Slack: we should think about splitting the planner target soc from the actual target, i.e. treat planner more as a minimum value.

/cc @naltatis

naltatis, 2023-03-18:
I get the idea, but I think we are introducing a lot of complexity where. We'll have to visualize this extra SoC limit in addition to the existing ones (min soc, target soc, vehicle limit). In addition it's not obvious to the user which soc is the limit. As an example: I use target time before longer trips. In these scenarios I want the vehicle reach a higher soc (100%) than my usual limit (80%). But you could also make the case the other way around where it's more of a minimum value as you described.

I'm not sure if it's wise to open this topic right now. I see that we have work to do here especially when we are thinking about departure schedules / recurring targets. see #5492

Da-ne-ezy, 2023-03-24:
For the visualization you could simply insert 2 bars/slider instead of currently only one bar/slider. The first bar sets the minSOC and the second the target SOC.

andig, 2023-03-24:
I think the real point here is that we should treat the planner soc as a `minsoc` with timestamp rather than as a `targetsoc`.

andromedavpn, 2023-05-15:
As a naive user who didn't read the doc, the described behavior was actually my expectation for min+PV mode. Eventually this is also my use case, preserve a minimum SoC for daily trips and in case of PV surplus charge to max.

andig, 2023-05-20:
It is unlcear what this issue should achieve. Planner uses target soc. Once target is achieved, PV mode would stop charging as well. We might revisit this behaviour once we address https://github.com/evcc-io/evcc/issues/5492.

closed 2023-05-20 by andig