evcc-io/evcc #6919: VW id - keine SOC Anzeige mehr, Fehlermeldung 400

ecfmau, 2023-03-18:
### Describe the bug

Hallo, seit gestern oder vorgestern, sehe ich das SOC meiner VW id.4 in evcc nicht mehr und die beiliegende Fehlermeldung ist angezeigt. Die WeConnect App funktioniert und zeigt den SOC richtig. Meine Tronity App ebenfalls. Ich habe nichts an meiner evcc-Konfiguration oder an meinem MyVW-Profil geändert

Gibt es ein vorübergehendes Problem mit der API oder gibt es eine Änderung, die eine neue Version von evcc erfordert?

Danke und freundliche Grüsse 
Fred

### Steps to reproduce

Auto anschliessen und Ladung starten.

### Configuration details

```yaml
vehicles:
  name: vw_id4
  title: ID.4
  type: template
  template: id
  cloud: true
  user: [...]
  password: [...]
  vin: [...]
  capacity: 77
  mode: pv
  targetSoC: 80
  minCurrent: 6
  maxCurrent: 16
```

### Log details

```text
vehicle soc: vehicle not available: cannot create vehicle 'cloud': rpc error: code = Unknown desc = token contains an invalid number of segments 

vehicle soc: rpc error: code = Unknown desc = Get \"https://emea.bff.cariad.digital/vehicle/v1/vehicles/WVGZZZxxxxxx/selectivestatus?jobs=all\": unexpected status: 400

telemetry: upload failed: unexpected status: 403
```

### What type of operating system are you running?

Docker container

### Version

0.114.1

andig, 2023-03-18:
Logfile fehlt

closed 2023-03-18 by andig

ecfmau, 2023-03-18:
Sorry, hier ist das Logfile.
[evcc.log](https://github.com/evcc-io/evcc/files/11008579/evcc.log)