evcc-io/evcc #6929: Frage: Kann targetTime über die API mit MQTT gelöscht werden?

DerDoubleD, 2023-03-18:
### Describe the bug

bei/mit http geht das ja über **DELETE /api/loadpoints/<id>/target/time**
bei/mit MQTT findet sich kein Eintrag in der Docu

### Steps to reproduce

1. publish **evcc/loadpoints/<id>/targetTime disable oder false** oder etwas in der Form wäre super
2.
3.
...

### Configuration details

```yaml
evcc.yaml
```

### Log details

```text
Log
```

### What type of operating system are you running?

Linux

### Version

V0.114.1

andig, 2023-03-19:
Gute Frage. Könntest Du mal probieren nur Nullen als Timestamp zu schicken?

andig, 2023-03-19:
@DerDoubleD Du kannst `null` verwenden, nebenbei wird das Zero Value auch gleich noch aus MQTT entfernt.

closed 2023-03-19 by andig

andig, 2023-03-19:
Die Alternative zu `null` wäre `0001-01-01T00:00:00Z`. Bitte diese verwenden, `null` wird ggf. wieder entfernt.

DerDoubleD, 2023-03-21:
Danke das funktioniert!