evcc-io/evcc #6937: SoC cache zurücksetzen bei reconnect

MathysDev, 2023-03-19:
### Describe the bug

Bei einem Reconnect sollte der Cache für den SoC gelöscht werden.
Wenn der alte SoC im Cache verbleibt startet die Ladung nicht mehr da soc > target und SoC nicht neu abgerufen wird.

### Steps to reproduce

1. Laden bis zum target
2. Abstecken, fahren, anstecken
3. Kommunikationsfehler bei einmaligem Update
4. SoC wird nicht mehr aktualisiert und Ladung startet nicht da SoC gem. Cache > target

### Configuration details

```yaml
vehicles:
- type: template
  template: hyundai
  title: IQ5
  user: XXX
  password: XXX
  capacity: 72
  cache: 60m
  phases: 3
  language: de
  mode: pv
  minSoC: 0
  targetSoC: 80
  minCurrent: 8
  maxCurrent: 20
  name: IQ5
loadpoints:
- title: Einstellhalle
  charger: Easee
  vehicle: IQ5
  mode: pv
  phases: 0
  resetOnDisconnect: true
  enable:
   threshold: 0
   delay: 2m
  guardduration: 7m
```

### Log details

```text
Mar 18 20:14:01 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:14:01 ----
Mar 18 20:14:01 raspi evcc[10593]: [lp-1  ] DEBUG 2023/03/18 20:14:01 charge power: 0W
Mar 18 20:14:02 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:14:02 pv power: 0W
Mar 18 20:14:04 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:14:04 grid power: 3276W
Mar 18 20:14:04 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:14:04 site power: 3276W
Mar 18 20:14:04 raspi evcc[10593]: [lp-1  ] DEBUG 2023/03/18 20:14:04 charge currents: [0 0 0]A
Mar 18 20:14:04 raspi evcc[10593]: [lp-1  ] DEBUG 2023/03/18 20:14:04 charger status: A
Mar 18 20:14:51 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:14:51 ----
Mar 18 20:14:51 raspi evcc[10593]: [lp-1  ] DEBUG 2023/03/18 20:14:51 charge power: 0W
Mar 18 20:14:52 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:14:52 pv power: 0W
Mar 18 20:14:54 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:14:54 grid power: 3251W
Mar 18 20:14:54 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:14:54 site power: 3251W
Mar 18 20:15:41 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:15:41 ----
Mar 18 20:15:41 raspi evcc[10593]: [lp-1  ] DEBUG 2023/03/18 20:15:41 charge power: 0W
Mar 18 20:15:42 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:15:42 pv power: 0W
Mar 18 20:15:44 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:15:44 grid power: 3251W
Mar 18 20:15:44 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:15:44 site power: 3251W
Mar 18 20:15:44 raspi evcc[10593]: [lp-1  ] DEBUG 2023/03/18 20:15:44 charge currents: [0.007 0.009 0.007]A
Mar 18 20:15:44 raspi evcc[10593]: [lp-1  ] DEBUG 2023/03/18 20:15:44 charger status: B
Mar 18 20:15:44 raspi evcc[10593]: [lp-1  ] INFO 2023/03/18 20:15:44 car connected
Mar 18 20:15:44 raspi evcc[10593]: [lp-1  ] DEBUG 2023/03/18 20:15:44 pv timer elapse
Mar 18 20:15:44 raspi evcc[10593]: [lp-1  ] DEBUG 2023/03/18 20:15:44 pv timer inactive
Mar 18 20:15:45 raspi evcc[10593]: [lp-1  ] ERROR 2023/03/18 20:15:45 vehicle soc: timeout
Mar 18 20:15:45 raspi evcc[10593]: [lp-1  ] WARN 2023/03/18 20:15:45 charger out of sync: expected disabled, got enabled
Mar 18 20:15:45 raspi evcc[10593]: [lp-1  ] DEBUG 2023/03/18 20:15:45 targetSoc reached: 90.6% > 80%
Mar 18 20:16:31 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:16:31 ----
Mar 18 20:16:31 raspi evcc[10593]: [lp-1  ] DEBUG 2023/03/18 20:16:31 charge power: 0W
Mar 18 20:16:32 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:16:32 pv power: 0W
Mar 18 20:16:34 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:16:34 grid power: 3209W
Mar 18 20:16:34 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:16:34 site power: 3209W
Mar 18 20:17:21 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:17:21 ----
Mar 18 20:17:21 raspi evcc[10593]: [lp-1  ] DEBUG 2023/03/18 20:17:21 charge power: 0W
Mar 18 20:17:22 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:17:22 pv power: 0W
Mar 18 20:17:24 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:17:24 grid power: 3211W
Mar 18 20:17:24 raspi evcc[10593]: [site  ] DEBUG 2023/03/18 20:17:24 site power: 3211W
Mar 18 20:17:25 raspi evcc[10593]: [lp-1  ] DEBUG 2023/03/18 20:17:25 vehicle odometer: 22183km
Mar 18 20:17:25 raspi evcc[10593]: [lp-1  ] DEBUG 2023/03/18 20:17:25 charge currents: [0.007 0.009 0.007]A
Mar 18 20:17:25 raspi evcc[10593]: [lp-1  ] DEBUG 2023/03/18 20:17:25 charger status: B
Mar 18 20:17:25 raspi evcc[10593]: [lp-1  ] DEBUG 2023/03/18 20:17:25 targetSoc reached: 90.6% > 80%
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.114.1

andig, 2023-03-19:
> Bei einem Reconnect sollte der Cache für den SoC gelöscht werden.

Das ist doch schon so?

MathysDev, 2023-03-19:
Bei mir wurde der SoC bereits mehrfach nicht zurückgesetzt. Beispiel gestern früh mit 91% losgefahren und Abends mit 41% angekommen. Heute morgen hatte EVCC immer noch 90.6% angezeigt (gemäss Log).

Im Log wird gestern 20:15:44 Charger Status B gesetzt (vorher A). Die 90.6% werden ohne SoC-Abfrage aus Cache gezogen.

andig, 2023-03-19:
> ERROR 2023/03/18 20:15:45 vehicle soc: timeout

Das wundert mich auch nicht wenn wir nix lesen können. Das Problem liegt hier an anderer Stelle. Welche ist aus dem Log nicht erkennbar.

closed 2023-03-19 by andig

MathysDev, 2023-03-19:
Es wird nur ein einziges Upate des SoC gemacht. Dann auf den Cache zurückzugreifen macht nach reconnect keinen Sinn.

andig, 2023-03-19:
> Welche ist aus dem Log nicht erkennbar.

Bitte Fahrzeug mit loggen.

MathysDev, 2023-03-19:
Loglevel waren:
log: info
levels:
  cache: error
  site: debug
  lp-1: trace
  hyundai: info

Sammle weiter Daten...

andig, 2023-03-19:
hyundai: trace

MathysDev, 2023-03-26:
Hier ein neues Log mit Trace. Woher die 85.0% SoC kommen ist weiterhin unbekannt, auf jeden Fall sind sie nicht aktuell.

`Mar 23 22:07:59 raspi evcc[3972173]: [lp-1  ] DEBUG 2023/03/23 22:07:59 charger status: A
Mar 23 22:08:46 raspi evcc[3972173]: [site  ] DEBUG 2023/03/23 22:08:46 ----
Mar 23 22:08:46 raspi evcc[3972173]: [lp-1  ] DEBUG 2023/03/23 22:08:46 charge power: 0W
Mar 23 22:08:48 raspi evcc[3972173]: [site  ] DEBUG 2023/03/23 22:08:48 pv power: 0W
Mar 23 22:08:49 raspi evcc[3972173]: [site  ] DEBUG 2023/03/23 22:08:49 grid power: 13804W
Mar 23 22:08:49 raspi evcc[3972173]: [site  ] DEBUG 2023/03/23 22:08:49 site power: 13804W
Mar 23 22:09:36 raspi evcc[3972173]: [site  ] DEBUG 2023/03/23 22:09:36 ----
Mar 23 22:09:36 raspi evcc[3972173]: [lp-1  ] DEBUG 2023/03/23 22:09:36 charge power: 2710W
Mar 23 22:09:38 raspi evcc[3972173]: [site  ] DEBUG 2023/03/23 22:09:38 pv power: 0W
Mar 23 22:09:39 raspi evcc[3972173]: [site  ] DEBUG 2023/03/23 22:09:39 grid power: 14124W
Mar 23 22:09:39 raspi evcc[3972173]: [site  ] DEBUG 2023/03/23 22:09:39 site power: 14124W
Mar 23 22:09:39 raspi evcc[3972173]: [lp-1  ] DEBUG 2023/03/23 22:09:39 charge currents: [15.6 0.01 0.008]A
Mar 23 22:09:39 raspi evcc[3972173]: [lp-1  ] DEBUG 2023/03/23 22:09:39 charger status: C
Mar 23 22:09:39 raspi evcc[3972173]: [lp-1  ] INFO 2023/03/23 22:09:39 car connected
Mar 23 22:09:39 raspi evcc[3972173]: [lp-1  ] DEBUG 2023/03/23 22:09:39 pv timer elapse
Mar 23 22:09:39 raspi evcc[3972173]: [lp-1  ] DEBUG 2023/03/23 22:09:39 pv timer inactive
Mar 23 22:09:39 raspi evcc[3972173]: [lp-1  ] INFO 2023/03/23 22:09:39 start charging ->
Mar 23 22:09:39 raspi evcc[3972173]: [hyundai] TRACE 2023/03/23 22:09:39 POST https://prd.eu-ccapi.hyundai.com:8080/api/v1/user/oauth2/token
Mar 23 22:09:41 raspi evcc[3972173]: [hyundai] TRACE 2023/03/23 22:09:41 grant_type=refresh_token&redirect_uri=https%3A%2F%2Fwww.getpostman.com%2Foauth2%2Fcallback&refresh_token=[RTOKEN]
Mar 23 22:09:41 raspi evcc[3972173]: --
Mar 23 22:09:41 raspi evcc[3972173]: {"access_token":"[TOKEN]","token_type":"Bearer","expires_in":86400}
Mar 23 22:09:41 raspi evcc[3972173]: [hyundai] TRACE 2023/03/23 22:09:41 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/[VEHICLE_ID]/status/latest
Mar 23 22:09:41 raspi evcc[3972173]: [hyundai] TRACE 2023/03/23 22:09:41 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{"lat":46.928153,"lon":7.576,"type":0},"head":160,"speed":{"value":0,"unit":0},"time":"20230323220836"},"vehicleStatus":{"airCtrlOn":true,"engine":true,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"0DH","unit":0,"hvacTempType":1},"defrost":false,"acc":true,"evStatus":{"batteryCharge":false,"batteryStatus":60,"batteryPlugin":0,"remainTime2":{"etc1":{"value":39,"unit":1},"etc2":{"value":740,"unit":1},"etc3":{"value":175,"unit":1},"atc":{"value":620,"unit":1}},"drvDistance":[{"rangeByFuel":{"evModeRange":{"value":242,"unit":1},"totalAvailableRange":{"value":242,"unit":1}},"type":2}],"reservChargeInfos":{"reservChargeInfo":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"00H","unit":0,"hvacTempType":0},"airCtrl":0,"heating1":0}}},"offpeakPowerInfo":{"offPeakPowerTime1":{"starttime":{"time":"1200","timeSection":0},"endtime":{"time":"1200","timeSection":0}},"offPeakPowerFlag":0},"reserveChargeInfo2":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"00H","unit":0,"hvacTempType":0},"airCtrl":0,"heating1":0}}},"reservFlag":0,"ect":{"start":{"day":0,"time":{"time":"0000","timeSection":0}},"end":{"day":0,"time":{"time":"0000","timeSection":0}}},"targetSOClist":[{"targetSOClevel":0,"dte":{"rangeByFuel":{"evModeRange":{"value":439,"unit":1},"totalAvailableRange":{"value":439,"unit":1}},"type":2},"plugType":0},{"targetSOClevel":0,"dte":{"rangeByFuel":{"evModeRange":{"value":439,"unit":1},"totalAvailableRange":{"value":439,"unit":1}},"type":2},"plugType":1}]},"chargePortDoorOpenStatus":2},"ign3":true,"hoodOpen":false,"transCond":false,"steerWheelHeat":0,"sideBackWindowHeat":0,"tirePressureLamp":{"tirePressureLampAll":0},"battery":{"batSoc":98,"batState":0,"sjbDeliveryMode":1,"batSignalReferenceValue":{"batWarning":65},"powerAutoCutMode":2},"lampWireStatus":{"stopLamp":{"stopLampStatus":false,"leftLamp":false,"rightLamp":false},"headLamp":{"headLampStatus":false,"leftLowLamp":false,"rightLowLamp":false,"leftHighLamp":false,"rightHighLamp":false},"turnSignalLamp":{"turnSignalLampStatus":false,"leftFrontLamp":false,"rightFrontLamp":false,"leftRearLamp":false,"rightRearLamp":false}},"smartKeyBatteryWarning":false,"washerFluidStatus":false,"breakOilStatus":false,"sleepModeCheck":false,"time":"20230323220100","remoteWaitingTimeAlert":{"remoteControlAvailable":1,"remoteControlWaitingTime":168,"elapsedTime":"00:00:00"},"systemCutOffAlert":0,"tailLampStatus":0,"hazardStatus":0},"odometer":{"value":22389.3,"unit":1}}},"msgId":"2f453df3-4e30-4a38-8a8a-15757aba4e89"}
Mar 23 22:09:41 raspi evcc[3972173]: [hyundai] TRACE 2023/03/23 22:09:41 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/[VEHICLE_ID]/status
[... cut ...] }}},"msgId":"6f7dc935-7704-4771-8441-c1aac7e91174"}
Mar 23 22:12:09 raspi evcc[3972173]: [lp-1  ] ERROR 2023/03/23 22:12:09 vehicle soc: timeout
Mar 23 22:12:09 raspi evcc[3972173]: [lp-1  ] DEBUG 2023/03/23 22:12:09 targetSoc reached: 85.0% > 80%
Mar 23 22:12:56 raspi evcc[3972173]: [site  ] DEBUG 2023/03/23 22:12:56 ----
Mar 23 22:12:56 raspi evcc[3972173]: [lp-1  ] DEBUG 2023/03/23 22:12:56 charge power: 0W
Mar 23 22:12:58 raspi evcc[3972173]: [site  ] DEBUG 2023/03/23 22:12:58 pv power: 0W
`

andig, 2023-03-26:
> Woher die 85.0% SoC kommen ist weiterhin unbekannt, auf jeden Fall sind sie nicht aktuell.

Das ist nicht zu erkennen bei dem Logschnipsel.

andig, 2023-03-26:
@panzerdev @nename0 ihr habt beide mehr Ahnung als ich von Bluelink. Könntet ihr vllt auch mal ein Auge drauf werfen?

nename0, 2023-03-26:
> > Woher die 85.0% SoC kommen ist weiterhin unbekannt, auf jeden Fall sind sie nicht aktuell.
> 
> Das ist nicht zu erkennen bei dem Logschnipsel.

Naja das ist offensichtlich: 
https://github.com/evcc-io/evcc/blob/b93a7029738ca8a15319d6d3f395a04f59a08cc7/core/loadpoint.go#L1522

`lp.vehicleSoc` wird nie geresetet.

 @MathysDev Hast du mal probiert in der App das Auto zu updaten und hat sich dann der graue Zeitstempel aktualisiert?
Sieht für mich jetzt ehr so aus als hätte das Auto am Ladepunkt kein Mobilfunk.

Könnte auch sein, dass da ein 2G Netz abgeschaltet wurde. Da scheint es ja in der Schweiz andere Zeitpläne als in D zu geben. (PS: Vielleicht Koordinaten auch anonymisieren 😉)

andig, 2023-03-26:
Na 403 ist forbidden. Mit Empfang sollte das nix zu tun haben.

nename0, 2023-03-26:
403? Ich sehe da nur 408er

MathysDev, 2023-03-26:
@nename0 Die Verbindung scheint nicht stabil zu sein, Teile der Aufrufe schlagen auch auf dem Mobile fehl, aber nicht alle. Mit dem Smartphone ist der Empfang zuverlässig... Ich hatte bisher vermutet dass einfach bluelink schlechte Software ist...

Habe nochmals durchs Log gelesen und habe es so verstanden:  (Logs von Hyundai gibt es im Zeitraum nicht mehr. Andere Quellen habe ich nur die relevanten Zeitpunkte).

- 22.03. 22:24: batteryStatus: 85% 
- 23.03. 07:22: car disconnected
- 23.03. 22:09: car connected, SoC timeout
- 23.03. 22:15: Letzter bluelink-Aufruf, timeout
---- hier keine Aufrufe von bluelink mehr gefunden
- 24.03. 10:54: SoC-Target hochgesetzt, Ladung startet und bluelink wird wieder abgerufen

Idealerweise würde EVCC aus meiner Sicht auch /status/latest berücksichtigen

Hier die genannten Logs:

```
Mar 22 21:21:22 raspi evcc[3972173]: [hyundai] TRACE 2023/03/22 21:21:22 POST https://prd.eu-ccapi.hyundai.com:8080/api/v1/user/oauth2/token
Mar 22 21:21:23 raspi evcc[3972173]: [hyundai] TRACE 2023/03/22 21:21:23 grant_type=refresh_token&redirect_uri=https%3A%2F%2Fwww.getpostman.com%2Foauth2%2Fcallback&refresh_token=[RTOKEN]
Mar 22 21:21:23 raspi evcc[3972173]: [hyundai] TRACE 2023/03/22 21:21:23 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/[VEHICLE]/status/latest
Mar 22 21:21:23 raspi evcc[3972173]: [hyundai] TRACE 2023/03/22 21:21:23 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{[COORD]},"head":160,"speed":{"value":0,"unit":0},"accuracy":{"hdop":10,"pdop":10},"time":"20230321150559"},"vehicleStatus":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"02H","unit":0,"hvacTempType":1},"defrost":false,"acc":false,"evStatus":{"batteryCharge":true,"batteryStatus":70,"batteryPlugin":2,"remainTime2":{"etc1":{"value":54,"unit":1},"etc2":{"value":560,"unit":1},"etc3":{"value":135,"unit":1},"atc":{"value":140,"unit":1}},"drvDistance":[{"rangeByFuel":{"evModeRange":{"value":308,"unit":1},"totalAvailableRange":{"value":308,"unit":1}},"type":2}],"reservChargeInfos":{"reservChargeInfo":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"00H","unit":0,"hvacTempType":0},"airCtrl":0,"heating1":0}}},"offpeakPowerInfo":{"offPeakPowerTime1":{"starttime":{"time":"1200","timeSection":0},"endtime":{"time":"1200","timeSection":0}},"offPeakPowerFlag":0},"reserveChargeInfo2":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"00H","unit":0,"hvacTempType":0},"airCtrl":0,"heating1":0}}},"reservFlag":0,"ect":{"start":{"day":0,"time":{"time":"0000","timeSection":0}},"end":{"day":0,"time":{"time":"0000","timeSection":0}}},"targetSOClist":[{"targetSOClevel":0,"dte":{"rangeByFuel":{"evModeRange":{"value":457,"unit":1},"totalAvailableRange":{"value":457,"unit":1}},"type":2},"plugType":0},{"targetSOClevel":0,"dte":{"rangeByFuel":{"evModeRange":{"value":457,"unit":1},"totalAvailableRange":{"value":457,"unit":1}},"type":2},"plugType":1}]},"chargePortDoorOpenStatus":1},"ign3":true,"hoodOpen":false,"transCond":true,"steerWheelHeat":0,"sideBackWindowHeat":0,"tirePressureLamp":{"tirePressureLampAll":0},"battery":{"batState":1,"sjbDeliveryMode":0,"batSignalReferenceValue":{"batWarning":65},"powerAutoCutMode":2},"lampWireStatus":{"stopLamp":{"stopLampStatus":false,"leftLamp":false,"rightLamp":false},"headLamp":{"headLampStatus":false,"leftLowLamp":false,"rightLowLamp":false,"leftHighLamp":false,"rightHighLamp":false},"turnSignalLamp":{"turnSignalLampStatus":false,"leftFrontLamp":false,"rightFrontLamp":false,"leftRearLamp":false,"rightRearLamp":false}},"smartKeyBatteryWarning":false,"washerFluidStatus":false,"breakOilStatus":false,"sleepModeCheck":false,"time":"20230321150603","remoteWaitingTimeAlert":{"remoteControlAvailable":1,"remoteControlWaitingTime":168,"elapsedTime":"15:37:42"},"systemCutOffAlert":0,"tailLampStatus":0,"hazardStatus":0},"odometer":{"value":22283.3,"unit":1}}},"msgId":"af67b284-3cbd-4d84-87f3-7495a732e1ea"}
Mar 22 21:23:00 raspi evcc[3972173]: [hyundai] TRACE 2023/03/22 21:23:00 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/[VEHICLE]/status/latest
Mar 22 21:23:01 raspi evcc[3972173]: [hyundai] TRACE 2023/03/22 21:23:01 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{[COORD]},"head":160,"speed":{"value":0,"unit":0},"accuracy":{"hdop":10,"pdop":10},"time":"20230321150559"},"vehicleStatus":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"02H","unit":0,"hvacTempType":1},"defrost":false,"acc":false,"evStatus":{"batteryCharge":true,"batteryStatus":70,"batteryPlugin":2,"remainTime2":{"etc1":{"value":54,"unit":1},"etc2":{"value":560,"unit":1},"etc3":{"value":135,"unit":1},"atc":{"value":140,"unit":1}},"drvDistance":[{"rangeByFuel":{"evModeRange":{"value":308,"unit":1},"totalAvailableRange":{"value":308,"unit":1}},"type":2}],"reservChargeInfos":{"reservChargeInfo":{"reservChargeInfoDetail":{"reservInfo":{"day":[9],"time":{"time":"1200","timeSection":0}},"reservChargeSet":false,"reservFatcSet":{"defrost":false,"airTemp":{"value":"00H","unit":0,"hvacTempType":0 [... cut ...] {"remoteControlAvailable":1,"remoteControlWaitingTime":168,"elapsedTime":"12:49:37"},"systemCutOffAlert":0,"tailLampStatus":0,"hazardStatus":0},"msgId":"b34e6014-443a-49a6-aa8e-54b6c50c1218"}
Mar 24 10:58:15 raspi evcc[3972173]: [lp-1  ] DEBUG 2023/03/24 10:58:15 vehicle status: C
Mar 24 10:58:15 raspi evcc[3972173]: [lp-1  ] DEBUG 2023/03/24 10:58:15 vehicle soc: 56%
Mar 24 10:58:15 raspi evcc[3972173]: [lp-1  ] DEBUG 2023/03/24 10:58:15 vehicle soc limit: 100%
```

PS: Danke für den Hinweis :)

nename0, 2023-03-26:
Also erst mal der Unterschied zwischen den beiden API-Calls:

  -   `.../status/latest` damit kann man den letzten Zustand den das Auto an den Bluelink Server gesendet hat holen. Dabei wird das Auto nicht involviert, wird also auch nicht aufgeweckt. Das Auto sendet aber auch von sich aus Zustände z.B. wenn man das Auto abstellt oder eine Ladung beendet wird. Dann hat man nach dem Abstellen des Autos direkt den letzten SOC in der App. In den App sieht man das immer als aktuellen Zustand inklusive dem Zeitstempel von dem der Zustand stammt.
  -  `.../status` damit zwingt man das Auto einen neuen Zustand zu liefern. Dabei wird das Auto aufgeweckt. In der App geht das über den Refresh-Knopf oben rechts.

> Idealerweise würde EVCC aus meiner Sicht auch /status/latest berücksichtigen

Ja das könntest du teilweise erreichen, indem du die `expiry` im bluelink config erhöhst, damit werden dann auch ältere `../status/latest` akzeptiert.

Die eine Stunde zwischen 22:12 und 23:13 kommt durch dein `cache: 60m`.
So ganz verstehe ich dass jetzt auch nicht warum er dannach aufhört die SOC zu pollen. Denke mal `socUpdated` wird gesetzt: 
https://github.com/evcc-io/evcc/blob/b93a7029738ca8a15319d6d3f395a04f59a08cc7/core/loadpoint.go#L1364

Und dann gibt `socPollAllowed` nie wieder true zurück: 
https://github.com/evcc-io/evcc/blob/b93a7029738ca8a15319d6d3f395a04f59a08cc7/core/loadpoint.go#L1316-L1344

Verstehe jetzt aber nicht wie das dann bei anderen Vehicle Providern funktioniert, dass weiterhin gepolled wird trotz Fehlern von der API? @andig

andig, 2023-03-27:
> Verstehe jetzt aber nicht wie das dann bei anderen Vehicle Providern funktioniert, dass weiterhin gepolled wird trotz Fehlern von der API?

Ich bin mir aktuell nicht sicher, ob/ dass das wirklich der Fall ist :/

nename0, 2023-03-27:
> Auch bei Disconnect: `evVehicleDisconnectHandler` -> `setActiveVehicle` -> `unpublishVehicle`. Damit ist er bei Connect immer noch 0.

Aber nur wenn das nicht das defaultVehicle war: https://github.com/evcc-io/evcc/blob/7e7fff48b64c3a82060f8caedca0c92ac4fb72f8/core/loadpoint_vehicle.go#L101-L107

> In dem Fehlerfall sollte allerdings egtl. (hab nicht nochmal nachgeschaut) auch `api.ErrMustRetry` der Fehler sein, wäre also kein Widerspruch.

Meinst du der `ResettableCached` oder Bluelink gibt `api.ErrMustRetry` zurück?
Bluelink gibt manchmal `api.ErrMustRetry` zurück manchmal nicht. Nicht so ganz konsistent.