evcc-io/evcc #6943: Wallbox elliconnect (ohne meter) stürzt ab bzw. disconnected, wenn Fahrzeug eingesteckt wird

TobiasSt78, 2023-03-19:
### Describe the bug

Ich habe nun seit mehreren Wochen und seit mehreren Versionen mit der Installation meins Vaters das Problem, dass die Wallbox sich beim Verbinden mit einem ID.3 im PV Modus disconnected.
Auffällig ist, dass das Koppeln oft lang dauert bzw. es sind mehrere Versuche erforderlich bzw. die Wallbox muss neu gebootet werden, damit das Koppeln wieder gelingt. 
Im Debug Modus gelingt das Koppeln auf Anhieb - im ganz normalen "Systemdienstmodus" ist es Glückssache. 

Die Wallbox ist per sehr stabilem WLAN (3 m Luftlinie) verbunden. LTE und LAN sind deaktiviert.
Die Box bleibt über die Website die ganze Zeit erreichbar

Bei meiner eigenen Installation mit Elli Wallboxen (mit Meter) funktioniert EVCC absolut einwandfrei und die YAML Dateien habe ich mehrfach abgeglichen.

Auch tritt der Fehler mit wie ohne diesem Code-Schnipsel auf:
```
  interfaces:
  - wlan0
  shipid: EVCC-6531383766356435
```

Ein Kabel ist bestellt und als einzige Option in meiner Macht steht der Switch von WLAN zu LAN. Meine Elli-Wallbox funktioniert jedoch auch per WLAN gut, weshalb ich das eigentlich nicht als DEN Fehler ansehe.

Der Fehler tritt auch auf, wenn testweise "ellipro" für den gleichen Charger (ohne Meter) definiert wird.

### Steps to reproduce

1. EVCC starten und Wallbox koppeln -> alles prima bis
2. Auto verbinden im "Schnell Modus" -> alles funktioniert
3. Auto abdocken und Modus auf "PV wechseln"
4. Auto anstopseln -> rote Fehlermedlungen in der EVCC Übersicht, Wallbox-Konfig Seite zeigt "nicht verbunden"
5. es hilft nun nur Auto Abstöpseln, Entkoppeln, Wallbox neu starten, EVCC neu starten, neu koppeln...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: XXXXXX

interval: 10s # control cycle interval

meters:
- name: grid_solar_edge
  type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.1.93  
  port: 1502  
  usage: grid  
  modbus: tcpip  
  timeout: 10s  
  
- name: pv_solar_edge
  type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.1.93  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  timeout: 10s  

- name: battery_byd
[... 477 lines cut ...]
[site  ] DEBUG 2023/03/19 14:51:24 grid power: -2W
[site  ] DEBUG 2023/03/19 14:51:24 site power: -2177W
[lp-1  ] DEBUG 2023/03/19 14:51:24 charger status: A
[lp-1  ] DEBUG 2023/03/19 14:51:24 next soc poll remaining time: 59m40s
[lp-1  ] WARN 2023/03/19 14:51:24 charger out of sync: expected enabled, got disabled
[lp-1  ] ERROR 2023/03/19 14:51:24 charger: can not enable charging as ev is unplugged
[lp-1  ] DEBUG 2023/03/19 14:51:24 charger disable: contactor delay 12m34s
[eebus ] DEBUG 2023/03/19 14:51:24 Send: d:_i:47859_Elli-Wallbox-2045A11IB1 notify 589 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2023/03/19 14:51:25 Send: d:_i:47859_Elli-Wallbox-2045A11IB1 notify 590 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2023/03/19 14:51:26 Send: d:_i:47859_Elli-Wallbox-2045A11IB1 notify 591 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2023/03/19 14:51:27 Send: d:_i:47859_Elli-Wallbox-2045A11IB1 notify 592 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2023/03/19 14:51:27 Send: d:_i:47859_Elli-Wallbox-2045A11IB1 notify 593 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2023/03/19 14:51:28 Send: d:_i:47859_Elli-Wallbox-2045A11IB1 notify 594 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2023/03/19 14:51:29 Send: d:_i:47859_Elli-Wallbox-2045A11IB1 notify 595 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2023/03/19 14:51:30 Send: d:_i:47859_Elli-Wallbox-2045A11IB1 notify 596 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2023/03/19 14:51:31 Send: d:_i:47859_Elli-Wallbox-2045A11IB1 notify 597 DeviceDiagnosisHeartbeatData
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.114.1

andig, 2023-03-19:
/cc @DerAndereAndi

DerAndereAndi, 2023-03-19:
Die Wallbox meldet das EV scheinbar als nicht verbunden:

```
[eebus ] DEBUG 2023/03/19 14:50:59 Recv: d:_i:47859_Elli-Wallbox-2045A11IB1: to NodeManagement notify 679 NodeManagementUseCaseData
[eebus ] DEBUG 2023/03/19 14:50:59 Recv: d:_i:47859_Elli-Wallbox-2045A11IB1: to NodeManagement notify 680 NodeManagementDetailedDiscoveryData
[eebus ] INFO 2023/03/19 14:50:59 ev disconnected
```

Bitte mal den Log Level von `eebus` auf `trace` stellen. Dann sieht man auch die Verbindungsversuche von evcc aus und ob die Verbindungsversuche von der Wallbox aus überhaupt kommen.

Eine WLAN Verbindung wird bei EEBUS immer zu Problemen führen können, wenn die Verbindung auch nur kurz weg ist, ist auch die Kontrolle von evcc weg und muss neu aufgebaut werden. Oft stürzt dabei der EEBUS Server auf der Wallbox ab, so dass das nicht immer funktioniert. Diese Probleme haben mehrere Personen gemeldet, doch in allen Logs zeigt das bisher auf ein Problem der Wallbox hin und nicht auf evcc und damit haben wir hier keinen Einfluss. Die besten Ergebnisse werden mit einer LAN Verbindung erzielt, so wie ich es hier bei mir installiert habe und damit überhaupt keine Verbindungsprobleme habe.

TobiasSt78, 2023-03-19:
Danke!

Hier kommt der erste Trace - von „EVCC Starten nicht möglich“, da die Kopplung fehlschlägt - und ohne etwas zu tun - nach 30 min funktioniert es dann. Das ist schon auffallend lang, oder? 

10 m lan Kabel und 40 cm langer Bohrer sind bestellt… die Umstellung auf lan ist etwas aufwändiger. Werde berichten…

`pi@PI-NEU:~ $ sudo journalctl -fau evcc
-- Journal begins at Fri 2023-03-10 21:55:13 CET. --
Mar 19 17:54:47 PI-NEU evcc[2404811]: [main  ] INFO 2023/03/19 17:54:47 starting ui and api at :7070
Mar 19 17:54:47 PI-NEU evcc[2404811]: [db    ] INFO 2023/03/19 17:54:47 using sqlite database: /root/.evcc/evcc.db
Mar 19 17:54:47 PI-NEU evcc[2404811]: [eebus ] INFO 2023/03/19 17:54:47 Local SKI:  f40ae61b52537cfe39ce285ed0ece43a2253be4d
Mar 19 17:54:47 PI-NEU evcc[2404811]: [eebus ] DEBUG 2023/03/19 17:54:47 mdns: announce
Mar 19 17:54:47 PI-NEU evcc[2404811]: [eebus ] DEBUG 2023/03/19 17:54:47 mdns: using avahi
Mar 19 17:54:47 PI-NEU evcc[2404811]: [eebus ] DEBUG 2023/03/19 17:54:47 starting websocket server on :4712
Mar 19 17:54:48 PI-NEU evcc[2404811]: [eebus ] TRACE 2023/03/19 17:54:48 registering ski: 9c8e73d2eb540792384d0edcba40a53537402735
Mar 19 17:54:48 PI-NEU evcc[2404811]: [eebus ] DEBUG 2023/03/19 17:54:48 mdns: start search
Mar 19 17:55:18 PI-NEU evcc[2404811]: [main  ] FATAL 2023/03/19 17:55:18 cannot create charger 'wallbox_garage': cannot create charger 'template': cannot create charger 'eebus': i/o timeout
Mar 19 17:55:18 PI-NEU evcc[2404811]: [main  ] FATAL 2023/03/19 17:55:18 will attempt restart in: 5m0s
Mar 19 18:00:18 PI-NEU evcc[2404811]: [eebus ] DEBUG 2023/03/19 18:00:18 mdns: stop announcement
Mar 19 18:00:18 PI-NEU systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Mar 19 18:00:18 PI-NEU systemd[1]: evcc.service: Failed with result 'exit-code'.
Mar 19 18:00:23 PI-NEU systemd[1]: evcc.service: Scheduled restart job, restart counter is at 10.
Mar 19 18:00:23 PI-NEU systemd[1]: Stopped evcc.
Mar 19 18:00:23 PI-NEU systemd[1]: Started evcc.
Mar 19 18:00:24 PI-NEU evcc[2406346]: [main  ] INFO 2023/03/19 18:00:24 evcc 0.114.1
Mar 19 18:00:24 PI-NEU evcc[2406346]: [main  ] INFO 2023/03/19 18:00:24 using config file: /etc/evcc.yaml
Mar 19 18:00:24 PI-NEU evcc[2406346]: [main  ] INFO 2023/03/19 18:00:24 starting ui and api at :7070
Mar 19 18:00:24 PI-NEU evcc[2406346]: [db    ] INFO 2023/03/19 18:00:24 using sqlite database: /root/.evcc/evcc.db
Mar 19 18:00:24 PI-NEU evcc[2406346]: [eebus ] INFO 2023/03/19 18:00:24 Local SKI:  f40ae61b52537cfe39ce285ed0ece43a2253be4d
Mar 19 18:00:24 PI-NEU evcc[2406346]: [eebus ] DEBUG 2023/03/19 18:00:24 mdns: announce
Mar 19 18:00:24 PI-NEU evcc[2406346]: [eebus ] DEBUG 2023/03/19 18:00:24 mdns: using avahi
Mar 19 18:00:24 PI-NEU evcc[2406346]: [eebus ] DEBUG 2023/03/19 18:00:24 starting websocket server on :4712
Mar 19 18:00:26 PI-NEU evcc[2406346]: [eebus ] DEBUG 2023/03/19 18:00:26 incoming connection request from 9c8e73d2eb540792384d0edcba40a53537402735
Mar 19 18:00:26 PI-NEU evcc[2406346]: [eebus ] DEBUG 2023/03/19 18:00:26 ski 9c8e73d2eb540792384d0edcba40a53537402735 is not paired, closing the connection
Mar 19 18:00:26 PI-NEU evcc[2406346]: [eebus ] TRACE 2023/03/19 18:00:26 registering ski: 9c8e73d2eb540792384d0edcba40a53537402735
Mar 19 18:00:26 PI-NEU evcc[2406346]: [eebus ] DEBUG 2023/03/19 18:00:26 mdns: start search
Mar 19 18:00:56 PI-NEU evcc[2406346]: [main  ] FATAL 2023/03/19 18:00:56 cannot create charger 'wallbox_garage': cannot create charger 'template': cannot create charger 'eebus': i/o timeout
Mar 19 18:00:56 PI-NEU evcc[2406346]: [main  ] FATAL 2023/03/19 18:00:56 will attempt restart in: 5m0s
Mar 19 18:04:14 PI-NEU evcc[2406346]: [eebus ] DEBUG 2023/03/19 18:04:14 incoming connection request from 9c8e73d2eb540792384d0edcba40a53537402735
Mar 19 18:04:14 PI-NEU evcc[2406346]: [eebus ] TRACE 2023/03/19 18:04:14 Send: 9c8e73d2eb540792384d0edcba40a53537402735 {"connectionHello":[{"phase":"ready"},{"waiting":60000}]}
Mar 19 18:04:14 PI-NEU evcc[2406346]: [eebus ] TRACE 2023/03/19 18:04:14 Recv: 9c8e73d2eb540792384d0edcba40a53537402735 {"connectionHello":[{"phase":"ready"},{"waiting":60000}]}
Mar 19 18:04:14 PI-NEU evcc[2406346]: [eebus ] TRACE 2023/03/19 18:04:14 Recv: 9c8e73d2eb540792384d0edcba40a53537402735 {"messageProtocolHandshake":[{"handshakeType":"announceMax"},{"version":[{"major":1},{"minor":0}]},{"formats":[{"format":["JSON-UTF8"]}]}]}
Mar 19 18:04:14 PI-NEU evcc[2406346]: [eebus ] TRACE 2023/03/19 18:04:14 Send: 9c8e73d2eb540792384d0edcba40a53537402735 {"messageProtocolHandshake":[{"handshakeType":"select"},{"version":[{"major":1},{"minor":0}]},{"formats":[{"format":["JSON-UTF8"]}]}]}
Mar 19 18:04:14 PI-NEU evcc[2406346]: [eebus ] TRACE 2023/03/19 18:04:14 Recv: 9c8e73d2eb540792384d0edcba40a53537402735 {"messageProtocolHandshake":[{"handshakeType":"select"},{"version":[{"major":1},{"minor":0}]},{"formats":[{"format":["JSON-UTF8"]}]}]}
Mar 19 18:04:14 PI-NEU evcc[2406346]: [eebus ] TRACE 2023/03/19 18:04:14 Send: 9c8e73d2eb540792384d0edcba40a53537402735 {"connectionPinState":[{"pinState":"none"}]}
Mar 19 18:04:14 PI-NEU evcc[2406346]: [eebus ] TRACE 2023/03/19 18:04:14 Recv: 9c8e73d2eb540792384d0edcba40a53537402735 {"connectionPinState":[{"pinState":"none"}]}
Mar 19 18:04:14 PI-NEU evcc[2406346]: [eebus ] TRACE 2023/03/19 18:04:14 Send: 9c8e73d2eb540792384d0edcba40a53537402735 {"accessMethodsRequest":[]}
[... cut ...]
Mar 19 18:23:39 PI-NEU evcc[2412558]: [lp-1  ] INFO 2023/03/19 18:23:39   mode:        pv
Mar 19 18:23:39 PI-NEU evcc[2412558]: [lp-1  ] INFO 2023/03/19 18:23:39   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
Mar 19 18:23:39 PI-NEU evcc[2412558]: [lp-1  ] INFO 2023/03/19 18:23:39   meters:      charge ✗
Mar 19 18:23:39 PI-NEU evcc[2412558]: [lp-1  ] INFO 2023/03/19 18:23:39 vehicle updated: unknown -> ID.3
Mar 19 18:23:40 PI-NEU evcc[2412558]: [lp-1  ] INFO 2023/03/19 18:23:40 car disconnected
        
`

andig, 2023-03-19:
Bitte formatieren

DerAndereAndi, 2023-03-19:
Der springende Punkt ist das hier, welches sich wiederholt:

```
[main ] FATAL 2023/03/19 18:00:56 cannot create charger 'wallbox_garage': cannot create charger 'template': cannot create charger 'eebus': i/o timeout
[main ] FATAL 2023/03/19 18:00:56 will attempt restart in: 5m0s
```

evcc erwartet dass die Wallbox innerhalb von 30 Sekunden korrekt reagiert. Aber die Wallbox reagiert eben nicht, der Fehler zeigt dass die Wallbox gar nicht reagiert, weil evtl. der EEBUS DIenst dort  startet und ne Weile braucht. Und dann startet evcc nach 5 Minuten automatisch neu um es nochmal zu versuchen.

Und dann kommt das hier:

```
[eebus ] DEBUG 2023/03/19 18:04:15 9c8e73d2eb540792384d0edcba40a53537402735 websocket read error: websocket: close 1006 (abnormal closure): unexpected EOF
```

Der EEBUS Dienst auf der Wallbox ist abgestürzt und damit wiederholt sich das Spiel und es funktioniert letztendlich.

Wie schon Anfangs beschrieben: Das Problem ist 100% auf der Wallbox Seite und nicht in evcc und in meinem EEBUS Code.

andig, 2023-03-19:
Dann machen wir hier mal zu.... Danke @DerAndereAndi fürs rein schauen.

closed 2023-03-19 by andig

TobiasSt78, 2023-03-20:
Ok, vielen Dank fürs prüfen

TobiasSt78, 2023-04-15:
Ich habe nun seit gut 2 Wochen die ELLI-Wallbox per LAN direkt an der Fritzbox und nicht mehr per WLAN dran. Seither sind die Abstürze des eeBus tatsächlich komplett weg und EVCC läuft nun stabil. 

Vielen Dank!

Was ich parallel noch geändert habe, und was einen positiven Einfluss haben könnte: Das SWAP File war mit 100 MB ständig am Anschlag. Nach Vergrößerung auf 300 MB sind nun "nur" noch 2/3 ständig belegt. Auf dem RPI3 läuft in Docker Raspberrymatic parallel und entweder Docker oder Raspberrymatic braucht offenbar viel RAM - ist also nicht direkt ein EVCC Problem ;-)