evcc-io/evcc #6950: KIA EV6 Ladezustand

FSTNDonald, 2023-03-19:
### Describe the bug

Moin,

Konfiguration:
(1) Victron Mulitplus-II -> regelt PV Zufuhr, d.h. Prio-1: in Speicher und Prio-2: Überschuss ins Netz
(2) go-e Wallbox GEMINI -> lädt 1- oder 3-phasig den PKW aus dem Netz
(3) PKW KIA EV6 -> soll *ausschliesslich* mit PV-Überschuss geladen werden, d.h. Prio-3

Der EVCC Server ist Ubuntu, aktuelle distri und EVCC kit.

Problem:
EVCC regelt perfekt (1) und (2) - nur bei (3) wird nicht der aktuelle Ladezustand angezeigt und für die Lastenverteilung verwendet.
Erläuterung:
Der EV6 ist über Fahrzeug -> KIA -> KIA Bluelink angemeldet (für den EV6 gibt es keine andere  Auswahlmöglichkeit). --> hier wird die Fehlerursache liegen !!

Die "KIA Bluelink" Anmeldung erfolgt mit den Anmeldedaten für "KIA Connect", d.h. dem Portal für den EV6.
In EVCC wird der Ladezustand mit dem gerade bei der Einrichtung von EVCC aktuellen Ladezustand als "100 Prozent" gesetzt (in der Realität nur 70%), während in "KIA Connect" der Ladezustand korrekt mit 70% angezeigt wird.

Da der EV6 seine Aufladung selber regelt, möchte ich eine Ladesteuerung konfigurieren, die nur aus:
(1) Prio-2: PV Überschuss ins Netz erkennen
(2) go-e Wallbox GEMINI so ansteuern, dass genau maximal dieser Überschuss über die Wallbox in den EV6 geladen wird.
Der EV6 sagt der Wallbox, wenn er voll ist und bricht den Ladevorgang ab.
Eine Verarbeitung der anscheinend nicht korrekt empfangenen Daten aus dem EV6 ist meiner Meinung nach eigentlich nicht notwendig ...?

### Steps to reproduce

siehe oben

### Configuration details

```yaml
root@evcc:~# cat /etc/evcc.yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: 9edeaa52cbe6db1fa558e4b778bxxx8b414df4713ca7740f6972ec5b64d27519

interval: 10s # control cycle interval

sponsortoken: haha
# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: victron-energy
  usage: grid
  host: 192.xxx.0.177
  port: 502
  name: grid1
- type: template
  template: victron-energy
  usage: pv
  host: 192.xxx.0.177
  port: 502
  name: pv2
- type: template
  template: victron-energy
  usage: battery
  host: 192.xxx.0.177
  port: 502
  name: battery3

chargers:
- type: template
  template: go-e-v3
  host: 192.xxx.0.183
  name: wallbox5

vehicles:
- type: template
  template: kia
  title: EV6
  user: xxx
  password: 'xxx'
  vin: xxx
  capacity: 77
  language: de
  name: ev4

loadpoints:
- title: go-e
  charger: wallbox5
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: xxx
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - xxx@evcc:~#
```

### Log details

```text
root@evcc:~# evcc --log debug
[main  ] INFO 2023/03/19 17:19:34 evcc 0.114.1
[main  ] INFO 2023/03/19 17:19:34 using config file: /etc/evcc.yaml
[main  ] INFO 2023/03/19 17:19:34 starting ui and api at :7070
[db    ] INFO 2023/03/19 17:19:34 using sqlite database: /root/.evcc/evcc.db
[lp-1  ] WARN 2023/03/19 17:19:36 locking phase config to 3p for switchable charger
[site  ] INFO 2023/03/19 17:19:36 site config:
[site  ] INFO 2023/03/19 17:19:36   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/03/19 17:19:36     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/03/19 17:19:36     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/03/19 17:19:36     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/03/19 17:19:36   vehicles:
[site  ] INFO 2023/03/19 17:19:36     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/03/19 17:19:36 loadpoint 1:
[lp-1  ] INFO 2023/03/19 17:19:36   mode:        pv
[lp-1  ] INFO 2023/03/19 17:19:36   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/03/19 17:19:36   meters:      charge ✓
[lp-1  ] INFO 2023/03/19 17:19:36     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/03/19 17:19:36 phase timer inactive
[lp-1  ] DEBUG 2023/03/19 17:19:36 pv timer inactive
[site  ] DEBUG 2023/03/19 17:19:36 ----
[main  ] FATAL 2023/03/19 17:19:36 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
root@evcc:~#
```

### What type of operating system are you running?

Linux

### Version

Ubuntu LTS 20.02 / evcc version 0.114.1

VolkerK62, 2023-03-19:
Wenn du dein Auto immer bis 100% laden willst, ist der Soc des Fahrzeug ja völlig egal.
Dann brauchst du auch gar keine Auto oder nur ein offline-vehicle konfigurieren. Da wird nix abgefragt, sondern geladen, bis das Auto "voll" sagt.

andig, 2023-03-19:
> wird nicht der aktuelle Ladezustand angezeigt und für die Lastenverteilung verwendet.

Was soll das bedeuten? Was für eine Lastenverteilung?