evcc-io/evcc #6953: Add template for newer SofarSolar hybrid inverters

cschlipf, 2023-03-19:
**Is your feature request related to a problem? Please describe.**
Missing support for newer SofarSolar Invertes. In my case HYD 10K-3PH, but protocoll applies to:
- HYD 5 … 20K-3PH and HYD3 … 6K-EP
- SOFAR 80 … 136KTL
- SOFAR 5 … 24KTL-G3

**Describe the solution you'd like**
Out of the box template and configuration for these inverters via RS485

**Describe alternatives you've considered**
Support via Solarman API, but there is a very long delay of more than 5 minutes.

**Additional context**
SOFAR Modbus Protocol - G3: [211116_SOFAR.HYD-3PH.and.SOFAR.-G3.Modbus.Protocol.2021-10-14_Client (1).xlsx](https://github.com/evcc-io/evcc/files/11012055/211116_SOFAR.HYD-3PH.and.SOFAR.-G3.Modbus.Protocol.2021-10-14_Client.1.xlsx)

Partially working meters config:
```
meters:
  - name: sofar-inverter
    type: custom
    power:
      source: modbus
      id: 1
      device: /dev/ttyUSB0   
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x0488 # Total PCC active power. Selling electricity is positive, buying electricity is negative
        type: holding
        decode: int16
      scale: 10
    energy:
      source: modbus
      id: 1
      device: /dev/ttyUSB0
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x021C
        type: holding
        decode: uint32
    soc:
      source: modbus
      id: 1
      device: /dev/ttyUSB0
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x0608 # No. 1 battery pack SOC
        type: holding
        decode: uint16
```

I will be happy to test any other registers you need me to test, but right now I am a bit lost, which registers you need for a template.

cschlipf, 2023-03-19:
Obige config gibt bei mir folgendes aus:
```
$ evcc meter
[main  ] INFO 2023/03/19 20:12:22 evcc 0.114.1 (e7e12a90)
[main  ] INFO 2023/03/19 20:12:22 using config file: /home/cschlipf/evcc.yaml
[db    ] INFO 2023/03/19 20:12:23 using sqlite database: /home/cschlipf/.evcc/evcc.db
Power:  -10W
Energy: read failed: serial: timeout
Soc:    57%
```
Die Werte habe ich überprüft und stimmen aktuell. Für Energy gibt es einen timeout. Lasse ich nun aber den Energy block weg, dann erhalte ich den Timeout für den SOC:
```
$ evcc meter
[main  ] INFO 2023/03/19 20:15:18 evcc 0.114.1 (e7e12a90)
[main  ] INFO 2023/03/19 20:15:18 using config file: /home/cschlipf/evcc.yaml
[db    ] INFO 2023/03/19 20:15:19 using sqlite database: /home/cschlipf/.evcc/evcc.db
Power: 0W
.Soc:  read failed: serial: timeout
```
Verstehe noch nicht was hier los ist. Sieht aus als würde immer der zweite Wert nicht auslesbar sein.

cschlipf, 2023-03-19:
Für die Currents müsste diese Konfiguration gelten:
```
    currents:
      - source: modbus
        id: 1
        device: /dev/ttyUSB0
        baudrate: 9600
        comset: "8N1"
        register:
          address: 0x0492 # R phase PCC current
          type: holding
          decode: uint16
        scale: 0.01
      - source: modbus
        id: 1
        device: /dev/ttyUSB0
        baudrate: 9600
        comset: "8N1"
        register:
          address: 0x049D # S phase PCC current
          type: holding
          decode: uint16
        scale: 0.01
      - source: modbus
        id: 1
        device: /dev/ttyUSB0
        baudrate: 9600
        comset: "8N1"
        register:
          address: 0x04A8 # T phase PCC current
          type: holding
          decode: uint16
        scale: 0.01
```
Leider kann ich das nun nicht validieren, weil ich hier jetzt wieder das read failed: serial : timeout erhalte wie zuvor für Energy:
```
evcc -c ~/evcc.yaml meter
[main  ] INFO 2023/03/19 22:29:52 evcc 0.114.1 (e7e12a90)
[main  ] INFO 2023/03/19 22:29:52 using config file: /home/cschlipf/evcc.yaml
[db    ] INFO 2023/03/19 22:29:53 using sqlite database: /home/cschlipf/.evcc/evcc.db
Power:          -600W
Current L1..L3: read failed: serial: timeout
Soc:            30%
Capacity:       5.0kWh
```

cschlipf, 2023-03-20:
Ich bin etwas weitergekommen. Aktueller Stand:

```
site:
  title: Pirolweg
  meters:
    grid: sofargrid
    pv: sofarpv
    battery: sofarbattery
  # prioritySoc: 95

meters:
  - name: sofargrid
    type: custom
    power:
      source: modbus
      id: 1
      device: /dev/ttyUSB0   
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x0488 # Total PCC active power. Selling electricity is positive, buying electricity is negative
        type: holding
        decode: int16
      scale: -10
    currents:
      - source: modbus
        id: 1
        device: /dev/ttyUSB0
        baudrate: 9600
        comset: "8N1"
        register:
          address: 0x0492 # R phase PCC current
          type: holding
          decode: uint16
        scale: 0.01
        timeout: 10s
      - source: modbus
        id: 1
        device: /dev/ttyUSB0
        baudrate: 9600
        comset: "8N1"
        register:
          address: 0x049D # S phase PCC current
          type: holding
          decode: uint16
        scale: 0.01
        timeout: 10s
      - source: modbus
        id: 1
        device: /dev/ttyUSB0
        baudrate: 9600
        comset: "8N1"
        register:
          address: 0x04A8 # T phase PCC current
          type: holding
          decode: uint16
        scale: 0.01
        timeout: 10s
  - name: sofarpv
    type: custom
    power:
      source: modbus
      id: 1
      device: /dev/ttyUSB0   
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x05C4 # Total PV power
        type: holding
        decode: uint16
      scale: 100
  - name: sofarbattery
    type: custom
    power:
      source: modbus
      id: 1
      device: /dev/ttyUSB0   
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x0606 # The charging and discharging power of the first battery pack. Charge is positive, discharge is negative
        type: holding
        decode: int16
      scale: -10
    soc:
      source: modbus
      id: 1
      device: /dev/ttyUSB0
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x0608 # No. 1 battery pack SOC
        type: holding
        decode: uint16
      timeout: 2s
    capacity: 5
```

cschlipf, 2023-03-20:
Und desto mehr Werte ich auslese, desto öfters bekomme ich diese "read failed: serial: timeout" - jetzt sporadischer bei verschiedenen Werten. Ich habe mir mal noch den von @premultiply empfohlenen RS484-USB Adapter bestellt. Meine Vermutung/Hoffnung ins blaue, dass der Adapter, den ich habe nicht sonderlich zuverlässig ist.

cschlipf, 2023-03-21:
@premultiply - auch mit dem  QITA USB RS485 Converter habe ich leider keine Verbesserung der Auslesequalität. Eher im Gegenteil. Die Currents kann ich eigentlich nie messen, da hier 3 Einzelwerte gelesen werden und eigentlich immer einer von ihnen fehlschlägt. Ich bekomme es aber hin, wenn ich die Register als Power auslese.

Irgendwelche Ideen woran das liegen könnte? Brauche ich einen abgeschirmten Draht? Aktuell nutze ich einfach nur einen Klingeldraht, der etwa 40cm lang ist.

premultiply, 2023-03-21:
Tut mir leid, ich kann dir nicht folgen.

Wo sieht man das? Trace-Log?

cschlipf, 2023-03-21:
Beim ausführen von `evcc meter`. Das ist jetzt meine manuelle config, aber bei der generierten Config ist es nicht anders.

1. Aufruf:
```
$ evcc -c evcc.yaml meter
[main  ] INFO 2023/03/22 00:05:31 evcc 0.114.1 (a216a867)
[main  ] INFO 2023/03/22 00:05:31 using config file: evcc.yaml
[db    ] INFO 2023/03/22 00:05:41 using sqlite database: /home/cschlipf/.evcc/evcc.db
sofar-grid
----------
Power:          260W
Energy:         read failed: serial: timeout
Current L1..L3: read failed: serial: timeout

sofar-pv
--------
Power:  0W
Energy: 488.1kWh

sofar-battery-1
---------------
Power:    -10W
Energy:   read failed: serial: timeout
Soc:      29%
Capacity: 5.0kWh
```
2. Aufruf:
```
$ evcc -c evcc.yaml meter
[main  ] INFO 2023/03/22 00:06:59 evcc 0.114.1 (a216a867)
[main  ] INFO 2023/03/22 00:06:59 using config file: evcc.yaml
[db    ] INFO 2023/03/22 00:06:59 using sqlite database: /home/cschlipf/.evcc/evcc.db
sofar-grid
----------
Power:          1760W
Energy:         panic: runtime error: index out of range [3] with length 2
Current L1..L3: read failed: serial: timeout

sofar-pv
--------
Power:  0W
Energy: read failed: failed to read from serial port within deadline

sofar-battery-1
---------------
Power:    read failed: serial: timeout
Energy:   read failed: unexpected EOF
.Soc:     read failed: serial: timeout
Capacity: 5.0kWh
```
3. Aufruf
```
$ evcc -c evcc.yaml meter
[main  ] INFO 2023/03/22 00:08:17 evcc 0.114.1 (a216a867)
[main  ] INFO 2023/03/22 00:08:17 using config file: evcc.yaml
[db    ] INFO 2023/03/22 00:08:17 using sqlite database: /home/cschlipf/.evcc/evcc.db
sofar-grid
----------
Power:          260W
Energy:         read failed: serial: timeout
Current L1..L3: read failed: serial: timeout

sofar-pv
--------
Power:  0W
Energy: 488.1kWh

sofar-battery-1
---------------
Power:    -10W
Energy:   read failed: serial: timeout
Soc:      29%
Capacity: 5.0kWh
```

Jedes Mal sind andere Werte betroffen und es gibt auch immer wieder andere Fehlermeldungen.

premultiply, 2023-03-21:
Mmmmpf.
Braucht der WR vielleicht eine Denkpause (delay) zwischen den einzelnen Abfragen?

premultiply, 2023-03-21:
@cschlipf Probier mal bitte bei jeder manuellen Abfrage noch `delay: 1s` einzufügen.

cschlipf, 2023-03-22:
Hi @premultiply ,

Vielen Dank!

Ich habe jetzt zwei Dinge getan:

1. Schritt: Ich habe einen qualitativ besseren RS485 Adapter genommen: Waveshare Industrial USB to RS485 Converter with original FT232RL (https://www.amazon.de/dp/B082D5PXJ4) - das hat schon gleich mal eine Verbesserung gebracht.
2. Schritt: Die delays wie Du sagtest eingebaut. 
    - Mit 1s wurde das schon deutlich besser. 
    - Runter auf 0.5s: wurde es wieder schlechter. 
    - Hoch auf 2s: Es wird besser
    - Hoch auf 3s: Es wird deutlich zuverlässiger. Zum ersten Mal bekomme ich komplette Auslesungen, aber immer noch hin und wieder Fehler. 
    - Hoch auf 5s: Die Auslesung wird zäh - aber es gibt immer noch Fehler. Keine spürbare Verbesserung gegenüber 3s.
 
Ich denke evcc ignoriert Fehler beim Einlesen und nimmt dann den nächsten Wert? Daher scheinen mir 3s delay der Sweetspot zu sein.

Meter config ist jetzt:

```
meters:
  - name: sofar-grid
    type: custom
    power:
      source: modbus
      id: 1
      device: /dev/ttyUSB0   
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x0488 # Total PCC active power. Selling electricity is positive, buying electricity is negative
        type: holding
        decode: int16
      scale: -10
      delay: 3s
    energy:
      source: modbus
      id: 1
      device: /dev/ttyUSB0
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x068E # Energy_Purchase_Total
        type: holding
        decode: uint32
      scale: 0.1
      delay: 3s
    currents:
      - source: modbus
        id: 1
        device: /dev/ttyUSB0
        baudrate: 9600
        comset: "8N1"
        register:
          address: 0x0492 # R phase PCC current
          type: holding
          decode: uint16
        scale: 0.01
        delay: 3s
      - source: modbus
        id: 1
        device: /dev/ttyUSB0
        baudrate: 9600
        comset: "8N1"
        register:
          address: 0x049D # S phase PCC current
          type: holding
          decode: uint16
        scale: 0.01
        delay: 3s
      - source: modbus
        id: 1
        device: /dev/ttyUSB0
        baudrate: 9600
        comset: "8N1"
        register:
          address: 0x04A8 # T phase PCC current
          type: holding
          decode: uint16
        scale: 0.01
        delay: 3s
  - name: sofar-pv
    type: custom
    power:
      source: modbus
      id: 1
      device: /dev/ttyUSB0   
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x05C4 # Total PV power
        type: holding
        decode: uint16
      scale: 100
      delay: 3s
    energy:
      source: modbus
      device: /dev/ttyUSB0   
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x0686 # PV_Generation_Total
        type: holding
        decode: uint32
      scale: 0.1
      delay: 3s
  - name: sofar-battery-1
    type: custom
    power:
      source: modbus
      id: 1
      device: /dev/ttyUSB0   
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x0606 # The charging and discharging power of the first battery pack. Charge is positive, discharge is negative
        type: holding
        decode: int16
      scale: -10
      delay: 3s
    energy:
      source: modbus
      id: 1
      device: /dev/ttyUSB0   
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x069A # Bat_Discharge_Total
        type: holding
        decode: uint32
      scale: 0.1
      delay: 3s
    soc:
      source: modbus
      id: 1
      device: /dev/ttyUSB0
      baudrate: 9600
      comset: "8N1"
      register:
        address: 0x0608 # No. 1 battery pack SOC
        type: holding
        decode: uint16
      delay: 3s
    capacity: 5
```

Baust Du den Delay in das Template ein? Dann würde ich morgen nochmal den nightly build testen.

cschlipf, 2023-03-22:
Und damit läuft evcc jetzt... Hurra!

premultiply, 2023-03-22:
Kannst du mal probieren was passiert wenn du die `id: 200` nutzt?

cschlipf, 2023-03-23:
> Kannst du mal probieren was passiert wenn du die `id: 200` nutzt?

Habe das delay rausgemacht, und die id auf 200 gesetzt bei allen metern. Ergebnis: kein Unterschied zu `id: 1`.

andig, 2023-03-24:
Fehlt hier noch was?

premultiply, 2023-03-24:
Nein.

closed 2023-03-24 by premultiply

cschlipf, 2023-03-24:
Vielen Dank Euch beiden. Ihr macht hier einen super Job. Mir ist erst heute aufgefallen, wie sehr die Liste der supporteten Geräte explodiert ist.

Ich werde morgen den nightly testen und Feedback geben.

cschlipf, 2023-03-25:
Habe es gerade mit der Nightly getestet.

Smooth!

Vielen Dank!

cschlipf, 2023-05-01:
@premultiply @andig 

Haben EVCC nun auch bei einem Freund zum laufen gebracht. Nur zur Bestätigung: Auch hier geht ohne das Delay von 1s nichts.

andig, 2023-05-01:
@cschlipf d.h. wir sollten ein delay fix mit ins Template einbauen?

SyrusTS4, 2023-07-14:
Ich hatte auch immer Timeout, zwar mit eine ESP, jedoch hat bei mir die Verwendung eines 120 Ohm Widerstandes eine deutliche Verbesserung gebracht, um nicht zu sagen, die Timeouts eliminiert.

PR #6966 (2023-03-20): Add SofarSolar G3 inverters
Fixes #6953

/cc @cschlipf Bitte testen.