evcc-io/evcc #6956: Add Solarwatt Battery Flex

premultiply, 2023-03-19:
Für das Solarwatt myReserve gibt es eine EVCC-Einbindung. Der wurde aber duruch Das Battery Flex abgelöst.

Gibt es jemanden, der eine Solarawtt Battery Flex einbinden kann? Die Abfrage der Werte kann über den Energy Manager erfolgen:
http://192.168.178.61/rest/kiwigrid/wizard/devices
Im resultierenden Json sind die folgenden Werte relevant:
ROOT.result.items[3].tagValues.StateOfCharge
ROOT.result.items[3].tagValues.PowerACOut
ROOT.result.items[3].tagValues.PowerACIn

                   ...
                    "StateOfCharge": {
                        "guid": "9746a5da-f482-4fe0-96ba-aaa71c841d30",
                        "tagName": "StateOfCharge",
                        "value": 56
                    },
                    "PowerACOut": {
                        "guid": "9746a5da-f482-4fe0-96ba-aaa71c841d30",
                        "tagName": "PowerACOut",
                        "value": 503.65899999999993
                    },
                    "PowerACIn": {
                        "guid": "9746a5da-f482-4fe0-96ba-aaa71c841d30",
                        "tagName": "PowerACIn",
                        "value": 0
                    },
                    ...

Eventuell kann man auch mit einem EVCC-Plugin arbeiten.

_Originally posted by @govgel in https://github.com/evcc-io/evcc/discussions/6954_

premultiply, 2023-03-20:
@govgel Bitte einmal das gesamte JSON.

govgel, 2023-03-20:
Das Template zeigt offenbar Zu- und Abfluss- aus der Batterie richtig an. Allerdings gibt es bezüglich des SOC den Fehler:
"ERROR 2023/03/20 15:43:58 battery 1 soc: jq: empty result". Das komplette JSON liegt bei.
[solarwatt_json.txt](https://github.com/evcc-io/evcc/files/11019626/solarwatt_json.txt)

premultiply, 2023-03-21:
`com.kiwigrid.devices.solarwatt.BatteryFlex.BatteryFlex` statt `com.kiwigrid.devices.solarwatt.MyReserve` muss es in diesem Fall heissen.

PR #6978 (2023-03-21): Solarwatt: Generalize Battery SoC
Fixes #6956
closed 2023-03-21 by premultiply