evcc-io/evcc #6968: Charging is not detected sometimes

thorstenwagner, 2023-03-20:
### Describe the bug

Ich hab es schon ein paar mal beobachtet, aber es passiert auch gerade jetzt. Das Auto ist meine Wallbox (Easee) angeschlossen, aber der Ladevorgang wird nicht von evcc detektiert.

In der log gibt es verdächtige Einträge:

```
Mar 20 17:17:06 raspberrypi evcc[1039]: [lp-1  ] INFO 2023/03/20 17:17:06 vehicle updated: unknown -> Skoda
Mar 20 17:17:19 raspberrypi evcc[1039]: [lp-1  ] INFO 2023/03/20 17:17:19 vehicle updated: Skoda -> unknown
Mar 20 17:17:40 raspberrypi evcc[1039]: [lp-1  ] INFO 2023/03/20 17:17:40 vehicle updated: unknown -> Skoda
Mar 20 17:31:15 raspberrypi evcc[1039]: [lp-1  ] INFO 2023/03/20 17:31:15 vehicle updated: Skoda -> unknown
Mar 20 17:31:21 raspberrypi evcc[1039]: [lp-1  ] INFO 2023/03/20 17:31:21 vehicle updated: unknown -> Skoda
Mar 20 17:31:51 raspberrypi evcc[1039]: [lp-1  ] INFO 2023/03/20 17:31:51 vehicle updated: Skoda -> unknown
```

`evcc health` zeigt mir Folgendes an:
```
[main  ] INFO 2023/03/20 17:36:45 evcc 0.114.1
[main  ] ERROR 2023/03/20 17:36:45 health check failed
```

### Steps to reproduce

-

### Configuration details

```yaml
chargers:
- type: template
  template: easee
  user: **
  password: **
  charger: **
  name: wallbox2

loadpoints:
- title: Garage
  charger: wallbox2
  mode: now
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

vehicles:
- name: skoda
  type: template
  template: enyaq
  capacity: 60
  user: **
```

### Log details

```text
[main  ] INFO 2023/03/20 17:40:56 evcc 0.114.1
[main  ] INFO 2023/03/20 17:40:56 using config file: /etc/evcc.yaml
[main  ] INFO 2023/03/20 17:40:57 starting ui and api at :7070
[db    ] INFO 2023/03/20 17:40:57 using sqlite database: /home/twagner/.evcc/evcc.db
[lp-1  ] WARN 2023/03/20 17:41:07 locking phase config to 3p for switchable charger
[site  ] INFO 2023/03/20 17:41:07 site config:
[site  ] INFO 2023/03/20 17:41:07   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/03/20 17:41:07     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/03/20 17:41:07     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/03/20 17:41:07   vehicles:
[site  ] INFO 2023/03/20 17:41:07     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/03/20 17:41:07 loadpoint 1:
[lp-1  ] INFO 2023/03/20 17:41:07   mode:        now
[lp-1  ] INFO 2023/03/20 17:41:07   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/03/20 17:41:07   meters:      charge ✓
[lp-1  ] INFO 2023/03/20 17:41:07     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/03/20 17:41:07 phase timer inactive
[lp-1  ] DEBUG 2023/03/20 17:41:07 pv timer inactive
[lp-1  ] DEBUG 2023/03/20 17:41:08 max charge current: 6A
[site  ] DEBUG 2023/03/20 17:41:08 ----
[server] FATAL 2023/03/20 17:41:08 unlinkat /tmp/evcc: operation not permitted
[lp-1  ] DEBUG 2023/03/20 17:41:08 charge power: 4124W
```
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.114.1

closed 2023-03-20 by andig

andig, 2023-03-20:
Das wär gut.

andig, 2023-03-20:
Das sind 10s Logfile. Zeigen die den falschen Status? Welchen?

thorstenwagner, 2023-03-21:
Gibt's denn ne Moeglichkeit mehr als 10s zu bekommen? 

Nach einem Reboot von evcc, wurde der Ladevorgang direkt erkannt. Ein Indiz das Easee eher nicht das Problem ist?