evcc-io/evcc #6980: Readouts from Victron System wrong

grasshide, 2023-03-21:
### Describe the bug

[image]

Cerbo GX Firmware v2.93
The Modbus TCP Service is enabled.

The readouts from my Victron System are not correlating to what I see in the UI.
I'm getting values (also in the log), but they don't make sense.

Two more points, not related to this:
- It took me ages to figure out how to configure a custom vehicle with mqtt. The docs are quite thin there. It was not clear to me in what format the mqtt messages have to be. I ended up with separate topics for soc and range and messages with raw values. But this is in none of the examples.
- The UI has a bug and will only show this vehicle values gathered from mqtt if I switch once between the configured vehicle and the guest vehicle and back.

### Steps to reproduce

1. install the docker compose version of evcc
2. connect to victron with the given config
3. see the problem written above

### Configuration details

```yaml
meters:
  - name: my_grid
    type: template
    template: victron-energy
    usage: grid
    host: 192.168.x.x
  - name: my_pv
    type: template
    template: victron-energy
    usage: pv
    host: 192.168.x.x
  - name: my_battery
    type: template
    template: victron-energy
    usage: battery
    host: 192.168.x.x
```
```

### Log details

```text
[lp-1  ] DEBUG 2023/03/21 11:04:47 charge power: 0W
[site  ] DEBUG 2023/03/21 11:04:47 grid power: -1W
[site  ] DEBUG 2023/03/21 11:04:47 pv power: 1W
[site  ] DEBUG 2023/03/21 11:04:47 site power: -1W
[lp-1  ] DEBUG 2023/03/21 11:04:47 charger status: B
```
```

### What type of operating system are you running?

Docker container

### Version

0.114.1

Hofyyy, 2023-03-21:
ich hab mir evcc damals auf dem mac entpackt. solange getestet bis alles super lief und es dann in den docker geschoben. nur falls du die denkst wie soll das im docker nur gehen. aus meiner sicht würde ich da nicht gross testen.

grasshide, 2023-03-21:
[image]

Ohne Änderungen an der Config sehe ich die Batterie auf der Kommandozeile.

Hofyyy, 2023-03-21:
Ok, das sieht doch aber schon mal gut aus, oder?

Hofyyy, 2023-03-21:
Wichtig ist das EVCC nur alle 10 Sekunden ausliest und Victron ja mehr oder minder live ist. Von daher siehst du auch nicht die 1:1 Werte.

grasshide, 2023-03-21:
> Ok, das sieht doch aber schon mal gut aus, oder?

Theoretisch ja, die UI sieht aber aus wie oben. Der PV Eingang fehlt dort komplett. Der Ladestrom in die Batterie ebenfalls.

Hofyyy, 2023-03-21:
Dann offtopic zu deinem Fahrzeug guck mal hier: https://github.com/evcc-io/evcc/discussions/5909#discussioncomment-5013790

Hofyyy, 2023-03-21:
Hmm, ok dir fehlt also die gleichstromseite wenn ich das richtig sehe.

grasshide, 2023-03-21:
> Also nur das ich es richtig verstehe, in der Commandozeile hast du alle Werte die du brauchst? PV Erzeugung und Ladestrom der Batterie?

Genau, die Anzeige der UI ist nur komplett falsch:
[image]

premultiply, 2023-03-21:
Falsche Konfig-Datei? evcc nicht neugestartet?

Hofyyy, 2023-03-21:
Genau! Der zieht die falsche Config.

grasshide, 2023-03-21:
Wie kann das sein? Zieht die Commandline eine andere Config als die Web UI?!

grasshide, 2023-03-21:
> Falsche Konfig-Datei? evcc nicht neugestartet?

Config stimmt. Container ist neu gestartet. Hab alles noch einmal geprüft.

Hofyyy, 2023-03-21:
Das kann schon sein, abhängig wie du den Docker eingestellt hast. Ich würde es einmal auf irgendeinem PC lokal starten. Sieht zu 99,9% nach einer falschen config aus.

grasshide, 2023-03-21:
Ich hab's gefunden.

Die Beschreibung der Site benötigt in der config auch noch die PV Anlage und die Batterie.
Ganz schwierig das mit den Beispielen so hin zu bekommen :(
Dennoch vielen Dank für eure Unterstützung!

closed 2023-03-21 by grasshide