evcc-io/evcc #6986: Don't signal "charge power must not be negative" when charging starts

Diavolezza, 2023-03-21:
I get the error message "charge power must not be negative:" almost every time I start charging. It's only a few watts (I currently have -4 to -17 in the log) and only for a very short time (< 1s). The wallbox is a go-e gemini. I have an SMA Home Manager.

A threshold (e. g. -20 W) or a grace period (5 s) would be great, before the warning is thrown.

premultiply, 2023-03-21:
Please ask goE to fix their firmware.

andig, 2023-03-21:
Wir wollen ja falsch angeschlossene Zähler identifizieren. Ein kleines Fenster macht den Zweck nicht kaputt; das können wir heilen?

closed 2023-03-21 by andig

tdxchris, 2023-08-25:
> I get the error message "charge power must not be negative:" almost every time I start charging. It's only a few watts (I currently have -4 to -17 in the log) and only for a very short time (< 1s). [...]

Do you resolve the problem in the meantime?

premultiply, 2023-08-25:
Calibrating your metering device may help?

andig, 2023-08-25:
Siehe https://github.com/evcc-io/evcc/pull/6990/files. 20W sind erlaubt- anscheinend hast Du ein anderes Problem. Bitte neues Issue!