evcc-io/evcc #6988: Add Solis / Ginglong inverters

premultiply, 2023-03-21:
Ich habe meine Konfiguration geändert und rufe inzwischen den Datenstick lokal im Netzwerk ab (dieser ist jedoch in regelmäßigen Abständen kurzzeitig nicht erreichbar, stört aber nicht weiter):
```
- type: custom
  name: pv2
  power:
    source: http
    uri: http://admin:admin@192.168.1.39/status.html
    method: GET
    regex: webdata_now_p\s*=\s*\"(\d+)\"
    headers:
    - content-type: application/json
    timeout: 60s
```
IP Adresse und Login entsprechend ändern.

_Originally posted by @Jan2412 in https://github.com/evcc-io/evcc/discussions/2773#discussioncomment-5382623_

premultiply, 2023-03-21:
@Jan2412 Könntest du mir zum testen mal den vollständigen Inhalt der status.html bei aktiver Erzeugung anhängen?

premultiply, 2023-03-21:
Können wohl auch einfach Modbus?
https://ginlongsolis.freshdesk.com/support/solutions/articles/36000100225-sunspec-starting-sn
https://ginlongsolis.freshdesk.com/support/solutions/articles/36000340158-modbus-communication-for-solis-inverters
https://ginlongsolis.freshdesk.com/support/solutions/articles/36000339385-modbus-communications-for-rhi-1p-5-10-k-hves-5g-us

Jan2412, 2023-03-22:
> @Jan2412 Könntest du mir zum testen mal den vollständigen Inhalt der `status.html` bei aktiver Erzeugung anhängen?
Anbei die gewünschte Datei.
In der Variablen "webdata_now_p" steht die aktuelle Erzeugung/Leistung des Wechselrichters

[status.zip](https://github.com/evcc-io/evcc/files/11040736/status.zip)

andig, 2023-04-01:
Ping ;)

andig, 2023-04-21:
@Jan2412 Currently nobody from the core team has time to work on this. A PR for a template would be great. See volkszaehler for example.

closed 2023-05-09 by premultiply