evcc-io/evcc #6991: Zielladen nach Neustart: "warte auf Fahrzeug" fehlt

micw, 2023-03-21:
### Describe the bug

Hallo zusammen,

ich mache gerade meine ersten Gehversuche mit EVCC. Da ich die Konfiguration noch oft ändere, starte ich den Container auch immer wieder neu. Dabei ist mir folgendes aufgefallen.

Wenn ich ein Fahrzeug (Nissan Leaf) anschließe und eine Zielzeit vorgebe, steht "Warte auf Fahrzeug", bis der Ladestand des Autos da ist. Wenn ich EVCC neu starte, steht "nicht verbunden" und die Ladezeit ist weg.

Ich warte jetzt, bis dort wieder "verbunden" steht und stelle Ladeziel und Ladezeit neu ein. Als Ladestand steht "--". 

Erwartet hätte ich, dass nun wieder "warte auf Fahrzeug" dasteht. Stattdessen bekomme ich manchmal "verbunden", manchmal die aktuelle Zeit als Startzeit (obwohl es nicht mit dem Laden startet).

[image]

Nach vielleicht einer Minute kommt dann die richtige Zeitplanung:

[image]

Ich vermute mal, dass es sich um einen Darstellungsfehler handelt.

### Steps to reproduce

Siehe oben

### Configuration details

```yaml
Welcher Teil ist hier relevant?
```

### Log details

```text
evcc    | [lp-1  ] INFO 2023/03/21 22:01:33 car connected
evcc    | [lp-1  ] DEBUG 2023/03/21 22:01:33 pv timer elapse
evcc    | [lp-1  ] DEBUG 2023/03/21 22:01:33 pv timer inactive
evcc    | [lp-1  ] DEBUG 2023/03/21 22:01:33 charger: guard elapse
evcc    | [lp-1  ] DEBUG 2023/03/21 22:01:33 guard timer inactive
evcc    | [lp-1  ] DEBUG 2023/03/21 22:02:11 set target soc: 55
evcc    | [lp-1  ] DEBUG 2023/03/21 22:02:13 set target soc: 60
<--- hier habe ich die Zielzeit eingestellt -->
evcc    | [lp-1  ] DEBUG 2023/03/21 22:02:19 switched phases: 3p
evcc    | [site  ] DEBUG 2023/03/21 22:02:19 ----
evcc    | [lp-1  ] DEBUG 2023/03/21 22:02:19 charge power: 0W
evcc    | [lp-2  ] DEBUG 2023/03/21 22:02:19 charge power: 0W
evcc    | [mqtt  ] DEBUG 2023/03/21 22:02:19 solar/116182009632/0/power wait for initial value
evcc    | [mqtt  ] DEBUG 2023/03/21 22:02:29 solar/116182009632/0/power wait for initial value
evcc    | [mqtt  ] DEBUG 2023/03/21 22:02:39 solar/116182009632/0/power wait for initial value
evcc    | [site  ] ERROR 2023/03/21 22:02:49 pv 1 power: solar/116182009632/0/power outdated: 10s
evcc    | [mqtt  ] DEBUG 2023/03/21 22:02:49 solar/114183090265/0/power wait for initial value
evcc    | [mqtt  ] DEBUG 2023/03/21 22:02:59 solar/114183090265/0/power wait for initial value
evcc    | [mqtt  ] DEBUG 2023/03/21 22:03:09 solar/114183090265/0/power wait for initial value
evcc    | [site  ] ERROR 2023/03/21 22:03:19 pv 2 power: solar/114183090265/0/power outdated: 10s
evcc    | [site  ] DEBUG 2023/03/21 22:03:19 pv power: 0W
evcc    | [site  ] DEBUG 2023/03/21 22:03:19 grid power: 771W
evcc    | [site  ] DEBUG 2023/03/21 22:03:19 site power: 771W
evcc    | [lp-1  ] DEBUG 2023/03/21 22:03:19 charge currents: [0 0 0]A
evcc    | [lp-1  ] DEBUG 2023/03/21 22:03:19 charger status: B
evcc    | [lp-1  ] DEBUG 2023/03/21 22:03:20 vehicle status: B
evcc    | [lp-1  ] DEBUG 2023/03/21 22:03:20 vehicle soc: 51%
evcc    | [lp-1  ] DEBUG 2023/03/21 22:03:20 vehicle range: 130km
evcc    | [lp-1  ] DEBUG 2023/03/21 22:03:20 planned 1h22m49s until 2023-03-22 05:30:00 +0100 CET at 3220W: total plan duration: 1h22m49s, avg cost: 0.233
evcc    | [lp-1  ] DEBUG 2023/03/21 22:03:20 pv charge current: 0A = 0A + -3.35A (771W @ 1p)
evcc    | [lp-1  ] DEBUG 2023/03/21 22:03:20 pv enable timer reset
evcc    | [lp-1  ] DEBUG 2023/03/21 22:03:20 pv timer inactive
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.114.1 (a216a867)

andig, 2023-03-21:
Um das zu verstehen bräuchten wir zu jedem der Screenshots den Output von /api/state und dazu noch Deine Config.

micw, 2023-03-21:
Alles klar. Die gesamte Config minus die Passworte, nehme ich an?

Ich teste es morgen Abend nochmal. Im Moment muss das Auto laden, damit meine Frau morgen auf Arbeit (und wieder heim ;-) ) kommt.

naltatis, 2023-04-05:
Ich glaube der Grund hier ist relativ einfach. Auf dem zweiten Screenshot sieht man, dass evcc den aktuellen SoC des Leafs nach dem Neustart noch nicht sofort kennt. Daher gibts auch noch keine geplante Startzeit weil nicht klar ist ob/welche Energiemenge noch geladen werden muss.

Vielleicht machst später Sinn noch einen Hinweis beim noch fehlenden Fahrzeug SoC einen Hinweis/Animation/.. anzuzeigen um zu symbolisieren, dass da noch was im Hintergrund läuft. Wenn ich das richtig einschätze betrifft das aber wirklich nur das Neustart-Szenario. Im laufenden Betrieb ist der SoC ggf. alt aber eigentlich immer gesetzt.

andig, 2023-04-10:
@naltatis gibts hier aktuell was zu tun?

naltatis, 2023-04-11:
Solange das Problem nur einmalig beim evcc (Neu)Start auftritt und sich dann selbst behebt (Fahrzeug erkannt) würde ich erstmal nichts ändern. Über den Fahrzeug-Update-Status ("Zuletzt aktualisiert vor ..." oder "Daten werden aktualisiert") sollten wir uns generell mal Gedanken machen. Aber würde ich erstmal nicht in diesem Issue sehen.

closed 2023-04-11 by naltatis