evcc-io/evcc #6994: Add Tibber Pulse local access

premultiply, 2023-03-21:
### Discussed in https://github.com/evcc-io/evcc/discussions/6993

<div type='discussions-op-text'>

<sup>Originally posted by **micw** March 21, 2023</sup>
Hallo zusammen,

kürzlich habe ich herausgefunden, dass man beim Tibber Pulse Bridge den Webserver dauerhaft aktivieren kann. Ein User hat angemerkt, dass man dort auch per HTTP den letzten Wert vom Pulse IR sehen kann.

Das habe ich zu einem Projekt zusammengewürfelt: https://github.com/micw/tibber-pulse-reader
Damit bekomme ich die Daten live aus dem Pulse ins MQTT, ohne den Umweg über die Cloud.

Vielleicht ist das für Euch interessant. Bedient Euch auch gerne an Code und Konzept, wenn ihr diese Funktion direkt in evcc einbauen wollt.</div>

andig, 2023-03-21:
Hilft es uns, reverse-engineerte Schnittstellen nachzubauen wenn es offizielle gibt? Ich find das aus Wartungsgesichtspunkten keine gute Idee.

premultiply, 2023-03-21:
Gibt wohl einfach das rohe SML des Zählers aus.

andig, 2023-03-24:
Das ist super, aber ich würde keine zwei Implementierungen pflegen wollen.

closed 2023-03-24 by andig