evcc-io/evcc #6999: Fronius Wattpilot: support authenticated charging

micw, 2023-03-22:
**Is your feature request related to a problem? Please describe.**

Currently I think about how to fix the following issue. When I connect a car to the charger, it immediately starts charging. After a while evcc sees a differenct between desired state and current state and stops charging. I'd rather have the car not charging on connect.

The only way I found is to enable authorization of the car. That means the charge can only be started either by a RFID or in the App.

**Describe the solution you'd like**

If the implementation in evcc would support to start charging when car auth is required, the charge would only start after it is allowed by evcc.
In addition, evcc could check if the auth was granted externally (e.g. by the wattpolit app or by RFID) and use this as trigger to switch from "PV" to "NOW". This way, I could simply enforce charging by RFID without the need to set a charge goal in evcc.

**Describe alternatives you've considered**

EVCC could poll the charger more often, so that charging is stopped immediately after a car is plugged. That would work but the user experience is not that good.

**Additional context**

I have 2 wattpilot and would be able to test an implementation.

andig, 2023-03-23:
Closing here since we fully depend on upstream. Happy to take a PR.

closed 2023-03-23 by andig

micw, 2023-03-23:
Da sich "upstream" dem Thema schon angenommen hat (https://github.com/evcc-io/evcc/discussions/6992, danke @mabunixda ), fände ich es git, den Request solange offen zu lassen, bis da ein Update oder ein "geht nicht" vorliegt.